espressif/vscode-esp-idf-extension #289: [Question]: Minimum OpenOCD version error message when trying to flash with JTAG

ihouses, 2021-01-11:
I am using the last version of the tool: v0.6. I have downloaded the tool chain and everything with the visual studio extension.
I can compile and flash with no problem with the UART but if I try to flash with the JTAG I get this meesage:

Minimum OpenOCD version v0.10.0-esp32-20201125 is required while you have v0.10.0-esp32-20200709 version installed

The version that the tool installed is the v0.10.0-esp32-20200709 stored in C:\esp\tools\.espressif\tools\openocd-esp32

How can I change or update the OpenOCD version??

brianignacio5, 2021-01-12:
Yes the thing is, some changes were implemented on openOCD to allow Visual Studio Code to perform JTAG flashing from this version.

If you want to replace openOCD, just get one of the latest [openOCD releases](https://github.com/espressif/openocd-esp32/releases) and replace `c:\\esp\\tools\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20200709\\openocd-esp32\\bin` in your `idf.customExtraPaths` for the bin directory of your chosen release version, for example `c:\\esp\\tools\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20201202\\openocd-esp32\\bin` if you extract this version on your `.espressif/tools` directory. 
Also update `idf.customExtraVars` OPENOCD_SCRIPTS to new OpenOCD Scripts folder path.

brianignacio5, 2021-01-15:
I'm closing this issue since answer to question is provided. Please comment here if this solution doesn't fix your issue.

closed 2021-01-15 by brianignacio5

jmporter34, 2021-07-23:
Newb question regarding "replace XXX with YYY in your idf.customExtraPaths" - what is the file name that has to be edited ?
Same question for "Also update idf.customExtraVars OPENOCD_SCRIPTS to new OpenOCD Scripts folder path.".
Thanks

brianignacio5, 2021-07-27:
@jmporter34 

Command Palette (F1 or View Menu -> Command Palette) and type **Preferences: Open Settings (UI)** or Command Palette (F1 or View Menu -> Command Palette) and type **Preferences: Open Settings (JSON)** or in your project `.vscode/settings.json`.

This is described in [SETUP JSON Manual Configuration](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/SETUP.md#json-manual-configuration) documentation.

sriccardi-invn, 2021-09-28:
Hi,
I had the same issue. I updated OpenOCD to v0.10.0-esp32-20210902 and I still have the issue: 
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32: Target halted, PC=0x4008FBA3, debug_reason=00000001
Info : Listening on port 3333 for gdb connections
Info : accepting 'tcl' connection on tcp/6666
Info : dropped 'tcl' connection
Info : accepting 'tcl' connection on tcp/6666
invalid command name "program_esp_bins"
Info : accepting 'tcl' connection on tcp/6666
Info : dropped 'tcl' connection
Info : accepting 'tcl' connection on tcp/6666
invalid command name "program_esp_bins"