espressif/vscode-esp-idf-extension #295: [Bug Report]: Flashing through WSL1 environment not working since v0.6.0

risinek, 2021-01-16:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [ ] Make sure you have searched for existing bugs and features request before you post an issue.
- [ ] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [ ] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
Since updating to v0.6.0 the extension (that is installed in WSL debian and runs in VSC installed on Win 10) is fetching serial ports from Windows 10 instead of WSL1. .
On version v.0.5.1 it shows `/dev/ttySx` ports correctly from WSL environment. Since v0.6.0 I get `COMx` from Windows environment.

When I try to flash the project, it runs the flash process via `powershell.exe` which does not exist in WSL Linux environment. (I have also tried manually overriding `idf.port` in `settings.json` to `"idf.port": "/dev/ttyS4"` instead of `COM4` that was provided via UI serial port selector with the same result)
```
Executing task: powershell.exe -Command "python \\\\wsl$\\Debian\\home\\username\\esp\\esp-idf\\components\\esptool_py\\esptool\\esptool.py -p /dev/ttyS4 -b 460800 --after hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size detect 0x8000 partition_table\partition-table.bin 0x1000 bootloader\bootloader.bin 0x10000 app.bin"
```
As I don't have python in Windows environment installed, this will end up in message 
```
Python was not found; run without arguments to install from the Microsoft Store,
 or disable this shortcut from Settings > Manage App Execution Aliases.
```

I'm able to build project via WSL environment, so this applies only for selecting serial port and flashing.

**To Reproduce**

1. Have WSL1 installed
2. Run VSC project in Remote-WSL environment
3. Build project
4. Select serial port
**Expected behavior:** it will give you Linux `/dev/ttySx` serial ports to choose from
**Actual behavior:** it will give Windows COMs instead
5. Flash project
**Expected behavior:** it will run python directly from WSL environment
**Actual behavior:** it will run powershell.exe in WSL environment

**Expected behavior**
ESP-IDF extension should use Linux environment only tools if it's running inside Remote-WSL environment in VSC.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**

- OS Name & Version: Windows 10 1909
- VSCode Version: 1.52.1 (installed on WSL Debian)
- ESP-IDF Version: 4.1
- Python Version: 3.5
- WSL - WSL1 Debian
- Remote-WSL - remote-wsl-0.52.0

**Logs**
If applicable, please share the log file which can be obtained from

```
{"message":"Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:15)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:15)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n\n\tat /home/username/.vscode-server/extensions/espressif.esp-idf-extension-0.6.0/dist/extension.js:1:7658\n\tat ChildProcess.exithandler (child_process.js:286:7)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2021-01-16T10:31:22.025Z"}
{"message":"ESP-IDF Terminal process ended with exit code 1.","stack":"Error: ESP-IDF Terminal process ended with exit code 1.\n\tat ChildProcess.<anonymous> (/home/username/.vscode-server/extensions/espressif.esp-idf-extension-0.6.0/dist/extension.js:1:258814)\n\tat ChildProcess.emit (events.js:223:5)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:272:12)","level":"error","timestamp":"2021-01-16T10:31:47.799Z"}
{"message":"Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:15)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:15)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n\n\tat /home/username/.vscode-server/extensions/espressif.esp-idf-extension-0.6.0/dist/extension.js:1:7658\n\tat ChildProcess.exithandler (child_process.js:286:7)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Socket.<anonymous> (internal/child_process.js:430:11)\n\tat Socket.emit (events.js:223:5)\n\tat Pipe.<anonymous> (net.js:664:12)","level":"error","timestamp":"2021-01-16T10:32:24.142Z"}
{"message":"Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:15)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:15)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n\n\tat /home/username/.vscode-server/extensions/espressif.esp-idf-extension-0.6.0/dist/extension.js:1:7658\n\tat ChildProcess.exithandler (child_process.js:286:7)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Socket.<anonymous> (internal/child_process.js:430:11)\n\tat Socket.emit (events.js:223:5)\n\tat Pipe.<anonymous> (net.js:664:12)","level":"error","timestamp":"2021-01-16T10:37:59.130Z"}
{"message":"Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:15)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:15)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n\n\tat /home/username/.vscode-server/extensions/espressif.esp-idf-extension-0.6.0/dist/extension.js:1:7658\n\tat ChildProcess.exithandler (child_process.js:286:7)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2021-01-16T10:40:41.486Z"}
{"message":"Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:15)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:15)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n\n\tat /home/username/.vscode-server/extensions/espressif.esp-idf-extension-0.6.0/dist/extension.js:1:7658\n\tat ChildProcess.exithandler (child_process.js:286:7)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2021-01-16T10:40:42.552Z"}
{"message":"Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:15)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:15)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n\n\tat /home/username/.vscode-server/extensions/espressif.esp-idf-extension-0.6.0/dist/extension.js:1:7658\n\tat ChildProcess.exithandler (child_process.js:286:7)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Socket.<anonymous> (internal/child_process.js:430:11)\n\tat Socket.emit (events.js:223:5)\n\tat Pipe.<anonymous> (net.js:664:12)","level":"error","timestamp":"2021-01-16T10:49:34.094Z"}
{"message":"Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:15)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:15)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n\n\tat /home/username/.vscode-server/extensions/espressif.esp-idf-extension-0.6.0/dist/extension.js:1:7658\n\tat ChildProcess.exithandler (child_process.js:286:7)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2021-01-16T10:49:35.695Z"}
[... cut ...] /esp32ulp-elf-binutils/bin:/home/username/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/home/username/.espressif/tools/openocd-esp32/v0.10.0-esp32-20191114/openocd-esp32/bin",
    "idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"/home/username/.espressif/tools/openocd-esp32/v0.10.0-esp32-20191114/openocd-esp32/share/openocd/scripts\"}",
    "idf.port": "COM4",
    "idf.flashType": "UART"
}
```

**Additional context**
Add any other context about the problem here.

brianignacio5, 2021-01-18:
Thanks for reporting this. We tried to fix the issue in WSL 2 of serial ports are not available by using windows host python instead of Linux tools as explained in [WSL doc](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/WSL.md). I can see how this affect WSL 1 behavior. Will fix soon to avoid using this method on WSL 1.

closed 2021-01-19 by brianignacio5

risinek, 2021-02-06:
I just tried to update to v0.6.1, but unfortunately I still run into the same issue - if I try to flash via WSL1, it tries to run powershell.exe and fails due to python missing in Windows (wsl host). 
If there is anything I can help you with in term of testing/providing more info about my setup, let me know.

risinek, 2021-02-06:
I've just checked your PR again, and I see the condition whether WSL1 or WSL2 should be expected depends on OS release right? But in fact, you can have WSL1 and WSL2 side by side on newest Windows and switch between them as you wish. 
My idea is to introduce some setting/config property and set WSL2 approach as default, and if user wants to use WSL1 on purpose, he would just override it in the settings.json file or somewhere. Do you think this is feasible @brianignacio5?

brianignacio5, 2021-02-07:
Are you sure you are calling the right distribution in Visual Studio Code ? Make sure to use `Remote-WSL: New Window using distro...` instead of `Remote-WSL: New Window...` since the latter will use default WSL distribution. I've tested this with both WSL 1 and WSL 2 and they work properly, as in WSL 1 uses ports /dev/ttyS and WSL 2 calls Powershell.exe

risinek, 2021-02-20:
I have never updated to WSL2 nor enabled anything related to it, but still 0.6.1 and now 1.0.0 also still try to flash using powershell.exe and it's looking for python on Windows. Do you have some hints what info I should gather to debug why the condition for WSL1/WSL2 still resolves my WSL env as WSL2?

brianignacio5, 2021-02-22:
In the WSL terminal, type `uname -r` and let me know the result.

Also in Windows do `wsl -l -v` from Powershell terminal.

We use the fact that kernel > 4.19 should be WSL 2 as stated in https://askubuntu.com/a/1177730

risinek, 2021-02-25:
Alright here are the outputs from those commands.
```
$ uname -r
4.4.0-18362-Microsoft
```

```
> wsl -l -v
  NAME      STATE           VERSION
* Debian    Running         1
 ```
So clearly I have WSL1 but the extension still tries to go with the workaround for WSL2 while flashing.

risinek, 2021-02-25:
I think I found the problem - in your PR #298 you didn't updated the condition in `flash/flashTask.ts` [(block between lines 82 and 91) ](https://github.com/brianignacio5/vscode-esp-idf-extension/blob/6c1d1017e261665ccdd7c1cd6be7d7b6d90ca13d/src/flash/flashTask.ts#L82..L91)
Should I fix it and open PR or will you do it?

brianignacio5, 2021-02-26:
> I think I found the problem - in your PR #298 with the fix you didn't include the condition in `flash/flashTask.ts` [(block between lines 82 and 91) ](https://github.com/brianignacio5/vscode-esp-idf-extension/blob/6c1d1017e261665ccdd7c1cd6be7d7b6d90ca13d/src/flash/flashTask.ts#L82..L91)
> I've opened PR with a fix #331 331

Oh wow ! Thanks for looking into it @risinek and your PR 😄

PR #298 (2021-01-19): WSL1 bugfix
On WSL 1, serial ports are available. The previous fix for WSL2 in #224 is not required for WSL 1.

add semver version compare method in `src/utils.ts`. Could be used in other methods.

Fix #295
PR #331 (2021-02-26): Fix flashing on WSL1  (VSC-580)
This was originally caused by  #224. Then it was partially fixed by #298 but `flash/flashTask.ts` was missed.

In this PR I just copypasted fix by @brianignacio5 to a missing spot.

Fix #295