espressif/vscode-esp-idf-extension #297: [Bug Report]:  Failed to flash the device (JTag)

JosefrBeltran, 2021-01-18:
**Describe the bug**
When I try to flash via "Flash (with JTag)" task, I get an error:

`Failed to flash the device (JTag), please try again [got response: 'C:/Users/josefr/.espressif/tools/openocd-esp32/v0.10.0-esp32-20201202/openocd-esp32/bin/../share/openocd/scripts/target/esp_common.cfg:130: Error: c:Usersjosefrespworkspacepotentiostatuild/flasher_args.json: Invalid argument in procedure 'program_esp_bins' at file "C:/Users/josefr/.espressif/tools/openocd-esp32/v0.10.0-esp32-20201202/openocd-esp32/bin/../share/openocd/scripts/target/esp_common.cfg", line 130', expecting: '0']` 

I have the last version of openocd-esp32 ( 20201202) and have changed it in` idf.customExtraPaths` and in the user settings of `idf.customExtraVars`.

Openocd-esp32 starts correctly. I can even flash using the terminal, but not using the vscode extension.

**Environment:**

- OS Name & Version: Windows 10 Pro 20H2
- VSCode Version: 1.52.1
- ESP-IDF Version: v4.1
- Python Version: 3.7.3

**Logs**

`{"user":true,"message":"Failed to flash the device (JTag), please try again [got response: 'C:/Users/josefr/.espressif/tools/openocd-esp32/v0.10.0-esp32-20201202/openocd-esp32/bin/../share/openocd/scripts/target/esp_common.cfg:130: Error: c:Usersjosefrespworkspacepotentiostat\build/flasher_args.json: Invalid argument\nin procedure 'program_esp_bins' \nat file \"C:/Users/josefr/.espressif/tools/openocd-esp32/v0.10.0-esp32-20201202/openocd-esp32/bin/../share/openocd/scripts/target/esp_common.cfg\", line 130', expecting: '0']","stack":"Error: JTAG_FLASH_FAILED\n\tat c:\\Users\\josefr\\.vscode\\extensions\\espressif.esp-idf-extension-0.6.0\\dist\\extension.js:8:291926\n\tat Generator.throw (<anonymous>)\n\tat a (c:\\Users\\josefr\\.vscode\\extensions\\espressif.esp-idf-extension-0.6.0\\dist\\extension.js:8:263921)\n\tat runMicrotasks (<anonymous>)\n\tat processTicksAndRejections (internal/process/task_queues.js:94:5)","level":"error","timestamp":"2021-01-18T09:22:57.255Z"}
`

**Extension Configuration settings**
-  `"idf.espIdfPathWin":` `"c:\\Users\\josefr\\esp\\esp-idf"`
- `"idf.customExtraPaths":` `"C:\\Users\\josefr\\.espressif\\tools\\xtensa-esp32-elf\\esp-2020r2-8.2.0\\xtensa-esp32-elf\\bin;C:\\Users\\josefr\\.espressif\\tools\\xtensa-esp32s2-elf\\esp-2020r2-8.2.0\\xtensa-esp32s2-elf\\bin;C:\\Users\\josefr\\.espressif\\tools\\esp32s2ulp-elf\\2.28.51-esp-20191205\\esp32s2ulp-elf-binutils\\bin;C:\\Users\\josefr\\.espressif\\tools\\esp32ulp-elf\\2.28.51-esp-20191205\\esp32ulp-elf-binutils\\bin;C:\\Users\\josefr\\.espressif\\tools\\cmake\\3.13.4\\bin;C:\\Users\\josefr\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20201202\\openocd-esp32\\bin;C:\\Users\\josefr\\.espressif\\tools\\ninja\\1.9.0;C:\\Users\\josefr\\.espressif\\tools\\idf-exe\\1.0.1;C:\\Users\\josefr\\.espressif\\tools\\ccache\\3.7"`
- `"idf.customExtraVars":` `"{\"OPENOCD_SCRIPTS\":\"${TOOL_PATH}/openocd-esp32/share/openocd/scripts\"}"`
- `"idf.adapterTargetName":` `"esp32"`
- `"idf.openOcdConfigs":` `[
         "interface/ftdi/c232hm.cfg",
         "board/esp-wroom-32.cfg"
  ]`

pwmb, 2021-01-18:
>  Error: c:Usersjosefrespworkspacepotentiostat�uild/flasher_args.json: Invalid argument in procedure

Seems like an issue with the path, will try to replicate the scenario.

JosefrBeltran, 2021-01-18:
Hi pwmb,

No, the full path is C:\Users\josefr\esp\workspace\potentiostat\build

jankrejci, 2021-01-18:
I have the same issue

meltdown03, 2021-01-19:
It works in linux. I think the TCL language (What OpenOCD uses in the cfg files) is having trouble parsing Windows file structure with its backwards slashes. I hard-coded the the directory into the `extension.ts` file, on line: https://github.com/espressif/vscode-esp-idf-extension/blob/f543fffcc76d5ad34506237382e28f14ed902bb5/src/extension.ts#L1845

in place of `${buildFolder}` but changed it to `c:\\\\Users\\\\User\\\\esp_projects\\\\simple_sniffer\\\\build` (with 4 slashes instead of 2).
It got past the error in the OP and echoed `*** Programming Started ***`, _**but**_ then couldn't properly parse the `flasher_args.json` file. It was unable to find `c:Usersuseresp_projects�uild\/partition_table/partition-table.bin` (Note the 2 different slashes.) and failed with the response `error -1`.

Maybe the best solution would be to have the ESP-IDF Extension parse the `flasher_args.json` file (file name and addr) and use the `program_esp` OpenOCD command to flash.
These are just my thoughts on this, I am in no way an expert

PR #301 (2021-01-21): Fix JTAG Flashing issue (on windows)
Use forced `/` instead of `\\`.

One can disable this by changing the configuration `openocd.jtag.command.force_unix_path_separator` (default `true`)

Closes #297
closed 2021-01-21 by pwmb