espressif/vscode-esp-idf-extension #305: Espressif IDF extention issue flsah / monitor

versamodule, 2021-01-28:
I am using Espressif IDF extention v0.6.1.
After flashing, i open the monitor which works just fine.
However, if I recompile and try to flash again I get an error saying "SerialException: could not open port..."
I know this is because it did not close the com port before it started flashing.
Can you guys fix this please, this was an issue before and was fixed but now its back again.

brianignacio5, 2021-02-01:
We need more information about your setup.

Did you make sure you closed the ESP-IDF Monitor terminal before running flash again ?

**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [ ] Make sure you have searched for existing bugs and features request before you post an issue.
- [ ] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [ ] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
A clear and concise description of what the bug is (current behavior).

**To Reproduce**
Steps to reproduce the behavior (including the full log).

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**

- OS Name & Version: [e.g. MacOS X 10.14.5]
- VSCode Version: [e.g. 1.35.1]
- ESP-IDF Version: [e.g. v3.2.2]
- Python Version: [e.g. 2.7.10]

> **NOTE**: You can use the `ESP-IDF: Doctor command` to generate a report of your configuration.

**Logs**
If applicable, please share the log file which can be obtained from

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

**Extension Configuration settings**
Open Command Palette (F1) and type `Preferences: Open Settings (JSON)` or use `{CURRENT_PROJECT}/.vscode/settings.json` to get the following extension configuration settings value:

- `idf.espIdfPath` (MacOS or Linux) or `idf.espIdfPathWin` (Windows) also called `Path to locate ESP-IDF framework (IDF_PATH)` in `Preferences: Open Settings (UI)`.
- `idf.pythonBinPath` (MacOS or Linux) or `idf.pythonBinPathWin` (Windows) also called `Python absolute binary path used to execute ESP-IDF Python Scripts` in `Preferences: Open Settings (UI)`.
- `idf.openOcdConfigs` also called `List of configuration files inside OpenOCD Scripts directory` in `Preferences: Open Settings (UI)`.
- `idf.customExtraPaths` also called `Paths to be appended to PATH` in `Preferences: Open Settings (UI)`.
- `idf.customExtraVars` also called `Variables to be added to system environment variables` in `Preferences: Open Settings (UI)`.
- `idf.adapterTargetName` also called `Target name for ESP-IDF Debug Adapter` in `Preferences: Open Settings (UI)`.

**Additional context**
Add any other context about the problem here.

versamodule, 2021-02-01:
I have a video showing it.
http://versamodule.com/ESP32/ESP%20Monitor_player.html

I am off work today, so i am unable to add the other things you wanted.
Watch the video, and then if you need more feel free to ask.

James

brianignacio5, 2021-02-03:
Thanks @versamodule will cook a fix soon

PR #314 (2021-02-18): kill monitor process, update debug adapter
Kill ESP-IDF Monitor process when flash is called.

Fix #305 

Update esp-dap-adapter to latest commit.

Can you give it a try @versamodule ?
closed 2021-02-18 by brianignacio5