espressif/vscode-esp-idf-extension #306: [Feature Request]: Install last version of OpenOCD automatically

ihouses, 2021-01-28:
Hi,

I have installed the last version of VSCODE ESP-IDF extension. 
I install the tool chain as normal but when I want to use the JTAG flash I get the following message:

**Minimum OpenOCD version v0.10.0-esp32-20201125 is required while you have v0.10.0-esp32-20200709 version installed**

I deleted manually the OpenOCD from the tools folder in order to force the VSCODE extension to install OpenOCD but looks like is installing a fixed version? 

There is some instructions around about how to upgrade manually the OpenOCD files and set new path variables that I couldn't really manage to follow.

Is there any easy solution for that?

brianignacio5, 2021-02-01:
Hi @ihouses the thing is that each ESP-IDF includes a tools.json that describe each ESP-IDF tool version that has been tested for that given version. Some features (like JTAG flashing for IDEs) where added later on OpenOCD than the version used for testing in that release for ESP-IDF.

TL;DR Yes, it is fixed on ESP-IDF to install specific version of Tools.

What is wrong with [this solution](https://github.com/espressif/vscode-esp-idf-extension/issues/289#issuecomment-758372220) ?

ihouses, 2021-02-01:
Hi,

Yes I managed to install manually the OpenOCD and now I can debug by first time with the esp-prog board :)

Then as I understand looks like the OpenOCD was not still mature enough when was included in the Visual Studio.  Sothere is nothing else to do that waiting longer that gets more functionalities.

Thanks!

closed 2021-02-02 by brianignacio5