espressif/vscode-esp-idf-extension #307: suggest install/upgrade python requirements automatically

phatpaul, 2021-01-28:
I'm still using ESP-IDF 3.3 with the Legacy Make (msys2).

I've been excitedly watching this VScode extension for a while and trying it out each time there's an update.  But I could never get it to work (until now).  It always fails to build after installing with python errors.  I understand there are countless situations to consider, here's mine.
Windows 7 x64 and Windows 10.  I chose to use the express setup with IDV v4.1 and chose the default python at C:\Users\pabbott\AppData\Local\Programs\Python\Python36-32\python.exe

The first build failed with an error that some required python packages are missing.

I see a warning about a new version of PIP, so I run the suggested command and upgraded pip successfully.  But the build still fails with: 
```
The following Python requirements are not satisfied:
pyserial>=3.0
future>=0.15.2
cryptography>=2.1.4
pyparsing>=2.0.3,<2.4.0
pyelftools>=0.22
esp-windows-curses; sys_platform == 'win32'
To install the missing packages, please run "C:\Users\pabbott\esp\esp-idf\install.bat"
```

The message is telling me to run install.bat.  So I run it and it reports success.  But the IDF build still fails.  I suspect the install.bat needs to run in the python virtual env, but the message doesn't mention how to do that.

Then I searched around and found there is a IDF extension command "ESP-IDF: Install ESP-IDF extension Python Packages".  Ran that and now I'm able to build.  Yay!

**Describe the solution you'd like**
I suggest that the tool should automatically upgrade PIP and then install the required python modules into the virtualenv when using the express install.  This will improve the experience for newcomers to this otherwise excellent extension.  
Or at least make it easier to find out how to fix this issue (i.e. a better error message, tool-tip, or troubleshooting guide).

brianignacio5, 2021-02-01:
Hi @phatpaul The extension setup should be installing python packages on the virtual environment.

There is probably some error happening in your setup that is preventing the packages to be installed. 

Since this feature is already implemented, can you please provide the information for bug report ?

**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [ ] Make sure you have searched for existing bugs and features request before you post an issue.
- [ ] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [ ] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
A clear and concise description of what the bug is (current behavior).

**To Reproduce**
Steps to reproduce the behavior (including the full log).

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**

- OS Name & Version: [e.g. MacOS X 10.14.5]
- VSCode Version: [e.g. 1.35.1]
- ESP-IDF Version: [e.g. v3.2.2]
- Python Version: [e.g. 2.7.10]

> **NOTE**: You can use the `ESP-IDF: Doctor command` to generate a report of your configuration.

**Logs**
If applicable, please share the log file which can be obtained from

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

**Extension Configuration settings**
Open Command Palette (F1) and type `Preferences: Open Settings (JSON)` or use `{CURRENT_PROJECT}/.vscode/settings.json` to get the following extension configuration settings value:

- `idf.espIdfPath` (MacOS or Linux) or `idf.espIdfPathWin` (Windows) also called `Path to locate ESP-IDF framework (IDF_PATH)` in `Preferences: Open Settings (UI)`.
- `idf.pythonBinPath` (MacOS or Linux) or `idf.pythonBinPathWin` (Windows) also called `Python absolute binary path used to execute ESP-IDF Python Scripts` in `Preferences: Open Settings (UI)`.
- `idf.openOcdConfigs` also called `List of configuration files inside OpenOCD Scripts directory` in `Preferences: Open Settings (UI)`.
- `idf.customExtraPaths` also called `Paths to be appended to PATH` in `Preferences: Open Settings (UI)`.
- `idf.customExtraVars` also called `Variables to be added to system environment variables` in `Preferences: Open Settings (UI)`.
- `idf.adapterTargetName` also called `Target name for ESP-IDF Debug Adapter` in `Preferences: Open Settings (UI)`.

**Additional context**
Add any other context about the problem here.

freetoair, 2021-02-22:
Completely useless installation instructions. I struggle with the ESP-IDF extension by installing it for 2 days and eventually I went back to Notepad ++ NppExec again.

brianignacio5, 2021-02-23:
I'm sorry to hear that @freetoair What instructions are you referring to ? Maybe you can provide us some feedback to see how we can improve it.

Here is the link to the [setup documentation](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/SETUP.md)

closed 2021-02-26 by brianignacio5