espressif/vscode-esp-idf-extension #316: [Bug Report]: Failed reading CMakeLists.txt file when trying to use CMakeLists.txt Editor

meltdown03, 2021-02-18:
Latest VSCode-Extension built from source. (Issue was present in previous builds as well)
ArchLinux OS
IDF 4.4-dev

On any project, attempting to open the CMakeLists.txt Editor fails. The error is that the CMakeLists.txt file cannot be found even though it does exist. Both the Project and Component files do the same thing. If I step through the extension code by using the debug feature of VSCode Extension Host, it works fine. I think it may be a timing issue or some Async problem. I'm not that familiar with Type/JavaScript.
Here is what is logged:
``` bash
{"user":true,"message":"Failed reading /home/neal/esp/esp-adf/examples/get-started/play_mp3_control/CMakeLists.txt","stack":"TypeError: t is not iterable\n\tat /home/neal/.vscode/extensions/espressif.esp-idf-extension-0.6.1/dist/extension.js:8:242241\n\tat /home/neal/.vscode/extensions/espressif.esp-idf-extension-0.6.1/dist/extension.js:8:242496\n\tat Generator.next (<anonymous>)\n\tat s (/home/neal/.vscode/extensions/espressif.esp-idf-extension-0.6.1/dist/extension.js:8:241938)","level":"error","timestamp":"2021-02-18T23:08:40.465Z"}
```
[image]

brianignacio5, 2021-02-19:
Thanks for reporting @meltdown03 Will take a look and work on a fix asap

meltdown03, 2021-02-19:
``` bash
# (Automatically converted from project Makefile by convert_to_cmake.py.)

# The following lines of boilerplate have to be in your project's CMakeLists
# in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)

include($ENV{ADF_PATH}/CMakeLists.txt)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(play_mp3)
```

I tried removing the comments and whitespace, but same thing.

closed 2021-02-20 by brianignacio5

meltdown03, 2021-02-20:
The editor works now but has an issue with the specific CMakeLists.txt file I posted above. Since there are 2 include lines, it removes the default line and won't work. Maybe in the next release you could add a + button for the includes.

PR #319 (2021-02-20): Relocate CMakeLists Schema to root directory
Fix #316

Move CMakeListsSchema.json to root directory since it is needed on every CMakeLists.txt editor webview call.

Rename gulp_clean task to clean for clarity.