espressif/vscode-esp-idf-extension #323: [Feature Request]: Provide Compiler Path Vartiable

boarchuz, 2021-02-19:
**Is your feature request related to a problem? Please describe.**
There are VS Code variables such as ${config:idf.espIdfPath} and ${workspaceFolder} which are useful in VS Code configuration (settings.json, tasks.json, etc).
The compiler path is very helpful for intellisense configuration but must be set manually as the path is unpredictable:
eg:
${config:idf.toolsPath}/tools/xtensa-esp32-elf/1.22.0-80-g6c4433a-5.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
${config:idf.toolsPath}/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
And it may change if the project changes IDF/compiler version during development.

**Describe the solution you'd like**
Assuming the IDF extension must know where to find the compiler for the current project - it would be great if it could expose it (eg. via **${config:idf.compilerPath}**) for use in our VS Code configuration files.

**Additional context**
I just noticed that the current compilerPath is automatically added to settings.json for IDF examples via:
https://github.com/espressif/vscode-esp-idf-extension/blob/cc22eafe7f3f59bb175c9b0a89fe35dba868c12e/src/examples/ExamplesPanel.ts#L117-L122
However, shouldn't this be _-gcc_ instead of _-gdb_?

brianignacio5, 2021-02-20:
Thanks for reporting this @boarchuz will fix asap, should be gcc there.

By the way, you can use `${command:espIdf.getXtensaGcc}` in tasks.json and launch.json to get the compiler absolute path in tasks or debug configuration. Please a look at #324

closed 2021-02-20 by brianignacio5

boarchuz, 2021-02-20:
Thanks for that @brianignacio5 
I'm just testing it now and it's not working quite as I'd hoped. I pulled in your changes and the command works fine when I create a simple task to `"echo ${command:espIdf.getXtensaGcc}"` however using it (`"compilerPath": "${command:espIdf.getXtensaGcc}"`) in c_cpp_properties.json results in this output under "C/C++ Configuration Warnings":

> [20/02/2021, 8:51:17 pm] Unable to resolve configuration with compilerPath "${command:espIdf.getXtensaGcc}".  Using "/usr/bin/gcc" instead.

Perhaps commands can't be used in this way? Or am I doing something silly? I'll have another try in a couple days. It is still a useful addition for checking and setting the path manually anyway.

brianignacio5, 2021-02-22:
Oh!  you can't use `${command:command_ID}"` in c_cpp_properties.json only on settings.json and tasks.json.

I think at some point Visual Studio Code limited the use of this to only these 2 files. We could add a command to update this file with latest xtensaGCC but some users might not like this behavior.

boarchuz, 2021-02-22:
@brianignacio5 Sorry I didn't realise that, I just assumed it would work.
I actually can't even get it to work via settings.json with `"C_Cpp.default.compilerPath": "${command:espIdf.getXtensaGcc}"`. (It does work with the complete path provided manually.)

What do you think about a command to set a bunch of defaults for the C/C++ extension in settings.json for the current project and a corresponding entry in cpp_properties.json?
eg.
```
    "C_Cpp.default.intelliSenseMode": "gcc-x86",
    "C_Cpp.default.cStandard": "c11",
    "C_Cpp.default.cppStandard": "c++17",
    "C_Cpp.default.compilerPath": "/home/matt/espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc",
    "C_Cpp.default.compileCommands": "${workspaceFolder}/build/compile_commands.json",
    "C_Cpp.default.includePath": [
        "/home/matt/espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/lib/gcc/xtensa-esp32-elf/8.4.0/include-fixed/**",
        "/home/matt/espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/lib/gcc/xtensa-esp32-elf/8.4.0/include/**",
        "${workspaceFolder}/build/config/**",
        "${config:idf.espIdfPath}/components/freertos/include/freertos/queue.h",
        "${config:idf.espIdfPath}/components/freertos/include/freertos/**",
        "${workspaceFolder}/**",
        "${config:idf.espIdfPath}/components/log/**",
        "${config:idf.espIdfPath}/components/**"
    ],
    "C_Cpp.default.browse.path": [
        "/home/matt/espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/lib/gcc/xtensa-esp32-elf/8.4.0/include-fixed",
        "/home/matt/espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/lib/gcc/xtensa-esp32-elf/8.4.0/include",
        "${workspaceFolder}/build/config",
        "${config:idf.espIdfPath}/components/freertos/include/freertos",
        "${workspaceFolder}",
        "${config:idf.espIdfPath}/components/log",
        "${config:idf.espIdfPath}/components"
    ],
    "C_Cpp.default.browse.limitSymbolsToIncludedHeaders": false,
```
And add/edit the entry in cpp_properties.json like this:
```
{
    "name": "ESP-IDF Default",
    "compilerPath": "${default}"
}
```
(in my testing, compilerPath is the only field that must be explicitly included else will silently default to /usr/bin/gcc; all other ${default}s are automatically brought in from settings.json)

The benefit of having them defined as defaults is that we can then add a customised one like this:
```
{
    "name": "ESP-IDF Default",
    "compilerPath": "${default}"
},
{
    "name": "ESP-IDF My Custom",
    "compilerPath": "${default}",
    "includePath: [
        "/custom/path/to/my/external/library/**",
        "${default}"
    ]
}
```

Or, alternately, allow the user to define additional paths and have the default ESP-IDF config include them. eg:
cpp_properties.json
```
{
    "env": {
        "userAdditionalIDFIncludePaths": [
            "/custom/path/to/my/external/library/**",
            "/another/mycomponent/myheader.h" <-- users can edit these
        ],
        "userAdditionalIDFBrowsePaths": [
            "/custom/path/to/my/external/library" <-- and these too
        ],
    },
    "configurations": [
        {
            "name": "ESP-IDF Default",
            "compilerPath": "${default}",
            "includePath": [
                "${env:userAdditionalIDFIncludePaths}",
                "${default}"
            ],
            "browse": {
                "path": [
                    "${env:userAdditionalIDFBrowsePaths}",
                    "${default}"
                ],
            }
        }
        etc
        
```
Would anything like that work (with a warning to not unexpectedly overwrite existing fields in settings.json, of course)?

I have also seen that the C/C++ extension has an interface for a "Configuration Provider", which seems to allow another extension to provide all of this on-the-fly (eg. https://github.com/microsoft/vscode-cmake-tools/blob/develop/src/cpptools.ts). I have never used this and can't find much documentation on it - longer term, would it be feasible for ESP-IDF to become a Configuration Provider?

PR #324 (2021-02-20): Bugfix/get compiler path
Fix #323 

Was using gdb instead of gcc for settings.json default compiler path in ESP-IDF Examples webview. Also use IDF target from configuration settings.

Also add `${command:espIdf.getXtensaGdb}` description in README.

Please take a look @boarchuz