espressif/vscode-esp-idf-extension #327: VSC does not find ADF path [Bug Report]:

derflaryam, 2021-02-21:
using the Version 1 of the IDF extension. 

VSC does not find the ADF includes. see png. 

Compiling a ADF application is working now.

[image]

brianignacio5, 2021-02-22:
What is the content of `${workspaceFolder}/build/compile_commands.json` ? If you are using the default configuration (as shown in C_CPP_CONFIGURATION documentation) please add

```json
"includePath": [
        "${config:idf.espIdfPath}/components/**",
        "${config:idf.espIdfPathWin}/components/**",
         "${config:idf.espAdfPath}/components/**",
        "${config:idf.espAdfPathWin}/components/**",
        "${workspaceFolder}/**"
      ],
```
or use the compile_commands.json configuration which should provide a more accurately resolution of include headers.

Please take a look at [C_CPP_CONFIGURATION](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/C_CPP_CONFIGURATION.md) documentation for more details about include configuration.

derflaryam, 2021-02-22:
[compile_commands.zip](https://github.com/espressif/vscode-esp-idf-extension/files/6024189/compile_commands.zip)

with this configuration the ADF includes aren not found in the VSC editor, but VSC can find the IDF includes ! !

derflaryam, 2021-02-22:
I changed Extension IntelliSense engine to Tag Parser. 

now it works

meltdown03, 2021-02-23:
I had really good luck using the CMake Tools
(ms-vscode.cmake-tools) extension and `Default` engine. It needs a bit of setup up, but only finds symbols that pertain to the project once it is.

brianignacio5, 2021-02-23:
Can you expand a bit how did you configured it @meltdown03 ? I've tried using it before without much success.

brianignacio5, 2021-02-23:
Closing this issue since original issue is fixed.

closed 2021-02-23 by brianignacio5