espressif/vscode-esp-idf-extension #328: [Bug Report]: Cant format code when file is within a esp-idf-extension project

derflaryam, 2021-02-24:
Formatting code when file is within a  esp-idf-extension project does not work. 

I copy the file to a platformio project and then WITHOUT RESTARTING VSC I am able to format it. 

Then content of the test file is very simple and Format style is { BasedOnStyle: Google, IndentWidth: 4, ColumnLimit: 0}
==========================

int _http_stream_event_handle(http_stream_event_msg_t *msg)
{
    if (msg->event_id == HTTP_STREAM_RESOLVE_ALL_TRACKS)
    {
        return ESP_OK;
    }

    if (msg->event_id == HTTP_STREAM_FINISH_TRACK)
    {
        return http_stream_next_track(msg->el);
    }
    if (msg->event_id == HTTP_STREAM_FINISH_PLAYLIST)
    {
        return http_stream_restart(msg->el);
    }
    return ESP_OK;
}
======================================

brianignacio5, 2021-02-24:
Change `"C_Cpp.clang_format_style": "Visual Studio"` in your project settings.json for your desired format style.

derflaryam, 2021-02-24:
As you think you found the solutin, I was trying following your suggestions the change the format to Visual Studio -
1) it does not format according to my wishes
2) I noticed, that the esp-idf-extension fixed the format to Visual Studio 

Pleas read the original more careful - I was trying two different cases with the same test c-code within the same VSC window ! ! !
a)  esp-idf-extension   does not format the code according to the settings in C_Cpp.clang_format_style. 
b) the same file in a PIO Project - the formatting works as desired 

So I think the esp-idf-extension is overwriting the C_Cpp.clang_format_style to the Visual Studio Format. 

In VSC you should be free to format the cade according your wishes.

brianignacio5, 2021-02-25:
**This extension doesn't do formatting of any kind.** Besides of adding `"C_Cpp.clang_format_style": "Visual Studio"` in the settings.json in a new project (when created with ESP-IDF: Show Examples or ESP-IDF: Create ESP-IDF project) we don't handle C/C++ features, that is what the [ms-vscode.cpptools](https://github.com/microsoft/vscode-cpptools) is required for.

My question is, how do you set the format style in PlatformIO ? As far as I understand they also depend on `ms-vscode.cpptools` so there must be a setting you are using in a PIO project but not in an ESP-IDF project.

> I noticed, that the esp-idf-extension fixed the format to Visual Studio

How is it fixed? Can you show where is fixed? The extension is fully open source and is maintained right here in Github (check the PRs) Just because it is set in the new project template ? You can change it at any time.

TL;DR Code formatting is part of the C/C++ language extension.

derflaryam, 2021-02-25:
1)
I uninstalled VSC and the extensions. Deleted 
%USER%\.platformio   
%appdata%\Roaming\
Code
%USER%\.vscode\extensions

2)
Then I installed a new vscode (VSCodeSetup-x64-1.53.2.exe) and extension C/C++ms-vscode.cpptools.
Then I change the C_Cpp: Clang_format_style to my preferred style “{ BasedOnStyle: Google, IndentWidth: 4, ColumnLimit: 0}”
I verified it with my test code and the format changed according the rules.

3)
Then I install a clean version of ESP-IDF extension and created a new IDF Project – TEMPLATE-APP by pressing F1- ESP-IDF: Create project. 

Then I try to format main.c
It will be formatted according Visual Studio although the format is set to another style. 

4) In addition I noticed, that when I create a IDF Project it starts a new instance of VSC. Within this instance (see VSC_afterCreateIDFProject.PNG) there is no Workspace and I can’t format the code according my settings.

BUT when I open the main.c from the IDF Project in the other instance of VSC (see VSC_beforeIDF-Project.PNG) where you can see the Workspace entry – I am able to format the main.c according to my settings.

So the conclusion is, that you can’t work with the VSC instance which is opened by creating a IDF project with the IDF extension.
 

I really hope, that I provided enough information now, that you can fix it.
[image]
[image]

derflaryam, 2021-02-25:
I found something ...

When creating a template project from ESP-IDF then this project generates a .vscode.settings.json file where the formatter is fixed to 
"C_Cpp.clang_format_style": "Visual Studio",  ?????

This prevents to chose a formatting style with the intelisense configuration menu. 

PLEASE REMOVE SUCH FILES FROM TEMPLATES 

so from my point of view it was an error of the extension changing settings from another extension.

brianignacio5, 2021-02-25:
So your solution was [this one](https://github.com/espressif/vscode-esp-idf-extension/issues/328#issuecomment-784951168) as previously explained ( [step by step](https://github.com/espressif/vscode-esp-idf-extension/issues/328#issuecomment-785541393)). 

 Please review [Visual Studio Code settings](https://code.visualstudio.com/docs/getstarted/settings) to learn more about how visual studio code and extensions allow you to customize your IDE experience with settings.json.

Thanks for your feedback, will remove this settings.json from template projects in the next release.

closed 2021-02-25 by brianignacio5

derflaryam, 2021-02-25:
I know the VSC settings well .... 

But no one expects that your example code breaks the other extension behavior.