espressif/vscode-esp-idf-extension #329: Closing Monitor before flashing

seulater, 2021-02-24:
If you open the monitor, build your project, then close the monitor,  and try to flash you will get a "ESP-IDF Monitor closed" popup. 
The only way to recover is to close VSC and reopen it to flash again. 

Also, if you already have the monitor open, and try to flash it does close the monitor automatically now, but it cannot then flash. I think you need a bit of a delay from the close to when you actually try to flash.

brianignacio5, 2021-03-01:
Thanks @seulater  for reporting ! Will take a look

seulater, 2021-03-24:
Just letting you know that this problem is still here. on v 1.0.1 
Did you guys actually connect a board and try it?

brianignacio5, 2021-03-25:
Yes @seulater we did tested, actually was hoping you try #337 to see if this still happens. Another user tested and it seemed ok. I'll add a PR .

seulater, 2021-03-25:
OK, To be honest I don't know how to pull in these fixes. I just wait until VSC updates your extension. 
If you feel the need, i can take a video of what happens.

PR #337 (2021-03-03): Relative links in tasks output, truncated export cmd in terminal
Fix #333 

Fix #266 

Fix #329

Use vscode ShellExecution to fix relative paths link in build/flash task outputs. PTAL @boarchuz 

Remove export cmd and use strict env for Monitor terminal and ESP-IDF Terminal . PTAL @resetnow

Fix closing monitor issues when flash is trigger. PTAL @seulater
PR #358 (2021-03-31): Enhance/kill monitor timeout
Use full path for shell executable. 

Add timeout when killing monitor terminal process.

Fix #356 
Fix #329
closed 2021-03-31 by brianignacio5