espressif/vscode-esp-idf-extension #333: [Bug Report]: Project relative paths in terminal not resolved on Ctrl+Click

boarchuz, 2021-02-26:
**Describe the bug**
Version 1.0.0
Ctrl+Click on a relative path in build output in the terminal window results in a VS Code popup indicating that the file could not be found.
eg. 
Terminal:
```
../main/hello_world.c: In function 'app_main':
../main/hello_world.c:28:5: error: 'qwerty' undeclared (first use in this function); did you mean 'qsort'?
     qwerty;
     ^~~~~~
```
VS Code popup upon Ctrl+Click on any relative path (eg. "../main/hello_world.c:28:5"):
```
../main/hello_world.c:28:5
No matching results
```

Absolute paths work correctly. This only affects relative paths. 

**More info**
This was working in 0.5.1.
git bisect:
7abc85897490e3586fa5329e8a1f4c94680019f2 is the first bad commit

brianignacio5, 2021-03-01:
Thanks for reporting this @boarchuz . I roll back to `shellExecution` since the original problem is gone. PTAL #337

boarchuz, 2021-03-01:
Thanks, @brianignacio5 , that's fixed it

PR #337 (2021-03-03): Relative links in tasks output, truncated export cmd in terminal
Fix #333 

Fix #266 

Fix #329

Use vscode ShellExecution to fix relative paths link in build/flash task outputs. PTAL @boarchuz 

Remove export cmd and use strict env for Monitor terminal and ESP-IDF Terminal . PTAL @resetnow

Fix closing monitor issues when flash is trigger. PTAL @seulater
closed 2021-03-03 by brianignacio5