espressif/vscode-esp-idf-extension #335: [Bug Report]: Add ESP32C3

JohnMacrae, 2021-02-28:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [y] Make sure you have searched for existing bugs and features request before you post an issue.
- [y] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [y] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
ESP32C3 is not selectable as target in View>Set Espressif Device Target

**To Reproduce**
Install Master IDF. Run VSCode. Select View>Set Espressif Device Target
List of available targets would be displayed including ESP32C3 but C3 is missing

**Environment (please complete the following information):**

- Win 10 20H2
- VSCode Version: 1.53.2
- ESP-IDF Version: Master
- Python Version: 3.9

**Extension Configuration settings**
Open Command Palette (F1) and type `Preferences: Open Settings (JSON)` or use `{CURRENT_PROJECT}/.vscode/settings.json` to get the following extension configuration settings value:

- `idf.espIdfPath` (MacOS or Linux) or `idf.espIdfPathWin` (Windows) also called `Path to locate ESP-IDF framework (IDF_PATH)` in `Preferences: Open Settings (UI)`.
- `idf.pythonBinPath` (MacOS or Linux) or `idf.pythonBinPathWin` (Windows) also called `Python absolute binary path used to execute ESP-IDF Python Scripts` in `Preferences: Open Settings (UI)`.
- `idf.openOcdConfigs` also called `List of configuration files inside OpenOCD Scripts directory` in `Preferences: Open Settings (UI)`.
- `idf.customExtraPaths` also called `Paths to be appended to PATH` in `Preferences: Open Settings (UI)`.
- `idf.customExtraVars` also called `Variables to be added to system environment variables` in `Preferences: Open Settings (UI)`.
- `idf.adapterTargetName` also called `Target name for ESP-IDF Debug Adapter` in `Preferences: Open Settings (UI)`.

    "idf.customExtraPaths": "C:\\Users\\john\\.espressif\\tools\\xtensa-esp32-elf\\esp-2020r3-8.4.0\\xtensa-esp32-elf\\bin;C:\\Users\\john\\.espressif\\tools\\xtensa-esp32s2-elf\\esp-2020r3-8.4.0\\xtensa-esp32s2-elf\\bin;C:\\Users\\john\\.espressif\\tools\\xtensa-esp32s3-elf\\esp-2020r3-8.4.0\\xtensa-esp32s3-elf\\bin;C:\\Users\\john\\.espressif\\tools\\riscv32-esp-elf\\1.24.0.123_64eb9ff-8.4.0\\riscv32-esp-elf\\bin;C:\\Users\\john\\.espressif\\tools\\esp32ulp-elf\\2.28.51-esp-20191205\\esp32ulp-elf-binutils\\bin;C:\\Users\\john\\.espressif\\tools\\esp32s2ulp-elf\\2.28.51-esp-20191205\\esp32s2ulp-elf-binutils\\bin;C:\\Users\\john\\.espressif\\tools\\cmake\\3.16.4\\bin;C:\\Users\\john\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20200709\\openocd-esp32\\bin;C:\\Users\\john\\.espressif\\tools\\ninja\\1.10.2;C:\\Users\\john\\.espressif\\tools\\idf-exe\\1.0.1;C:\\Users\\john\\.espressif\\tools\\ccache\\3.7;C:\\Users\\john\\.espressif\\tools\\dfu-util\\0.9\\dfu-util-0.9-win64;C:\\Users\\john\\.espressif\\tools\\idf-python-wheels\\idf4.3_py3.9_2021-01-07",
    "idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"C:\\\\Users\\\\john\\\\.espressif\\\\tools\\\\openocd-esp32\\\\v0.10.0-esp32-20200709/openocd-esp32/share/openocd/scripts\",\"IDF_CCACHE_ENABLE\":\"1\"}"

idf.adapterTargetName - not listed

Anwaarullah, 2021-03-01:
If you'd like to get it temporarily working, while an official update is released, check out [Quick mod to get ESP32C3 on VS Code](https://www.hackster.io/Anwaarullah/getting-started-with-esp32-c3-on-visual-code-esp-idf-plugin-abb695)

brianignacio5, 2021-03-01:
Thanks @JohnMacrae for your report will add it soon!
Also thanks @Anwaarullah  I'm sorry for giving you struggle, will add a **custom** option for those cases where not released IDF_TARGET also.

JohnMacrae, 2021-03-01:
Thank you both!
Kind Regards
John