espressif/vscode-esp-idf-extension #336: [Bug Report]: No commands found after update to 1.0

Ostpopcorn, 2021-03-01:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [ x ] Make sure you have searched for existing bugs and features request before you post an issue.
- [ x ] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [ x ] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
After opening VSC and leting both the extension and VSC update after a few weeks untouched I opened it today and let evertyhing update. After that all actions added by the extension result in a error promt such as: 
´´´
Command 'ESP-IDF: Add vscode configuration folder' resulted in an error (command 'espIdf.createVsCodeFolder' not found)
´´´
and no buttons wore present in the bottom bar. This is the case for all actions reachable by F1 menu.

However the menu on the left is there(docs search, app trace etc.), but without any buttons (No the most frequent user so I left a screenshot ).

**To Reproduce**
Well it happend at the update and reinstalling did not help so I dont know. I have restarted my computer a couple of times as well.
Have not reinstalled VSC and not removed any extension directories, only touched it with tools from inside VSC.

**Expected behavior**
Well, commands should be exectued and buttons should be present.

**Screenshots**
[image]

**Environment (please complete the following information):**

- OS Name & Version: Windows 10 v.1909
- VSCode Version: 1.53.2
- ESP-IDF Version: v4.2 (updated today)
- Python Version: 3.8 

**Logs**
I was poking around inte folders but the update seems to have cleared v0.6 folder and no log file was present in that folder.

Don't know if its any help but here is another screenshot
[image]

brianignacio5, 2021-03-01:
Can you try with this VSIX installer ? https://github.com/espressif/vscode-esp-idf-extension/suites/2129771194/artifacts/43492332

Also you can see the output in vscode Menu **Help** -> **Toggle Developer Tools** and check if any extension related error in the console. There is also a ESP-IDF Output (Menu **View** -> Output)

Ostpopcorn, 2021-03-01:
Installing via VSIX did not help.
Tested first to look in developer tools, and behold an error has emerged! 
[image]

Tested uninstall cpp extension and remove its folders. But when it is installed again the errors persists. 
[image]

Since it is something related to cpp extension and not this extension I'll create an issue there.

closed 2021-03-01 by Ostpopcorn