espressif/vscode-esp-idf-extension #339: [Bug Report]: Mac OS 11.1 Unable to build project with spawn cmake EACCES

poky, 2021-03-06:
- OS Name & Version: Mac OS 11.1
- VSCode Version: 1.55.0-insider
- ESP-IDF Version: 4.2 Release
- Python Version: 3.9

```
> Executing task: espressif.esp-idf-extension: ESP-IDF Compile <

ESP-IDF Custom Terminal
$ cmake
spawn cmake EACCES
```

bittailor, 2021-03-07:
I have the same issue when I try a build form vscode ext.

```
> Executing task: espressif.esp-idf-extension: ESP-IDF Compile <

ESP-IDF Custom Terminal
$ cmake
spawn cmake EACCES 
```
I think it's since the update of espressif.esp-idf-extension to 1.0.0

I have:
```
OS: macOS 11.2.2 
Visual Studio Code Version: 1.54.1
espressif.esp-idf-extension: 1.0.0
ESP-IDF version: 4.2 
```

The build from cmd line works.

eedea, 2021-03-07:
same here:
```
OS: macOS 10.15.3 
Visual Studio Code Version: 1.54.1
espressif.esp-idf-extension: 1.0.0
ESP-IDF version: 4.2 
```

brianignacio5, 2021-03-08:
Can you try running these commands in the terminal and share the results:

```
ls -alh `which cmake`
ls -alh `which ncmake`
```

bittailor, 2021-03-08:
In a normal terminal:
```
>ls -alh `which cmake`
lrwxr-xr-x  1 ...  32B Oct 10 11:19 /usr/local/bin/cmake -> ../Cellar/cmake/3.18.4/bin/cmake
>ls -alh `which ncmake`
ls: found: No such file or directory
ls: ncmake: No such file or directory
ls: not: No such file or directory
>
```

In a terminal after a [get_idf](https://docs.espressif.com/projects/esp-idf/en/v4.2/esp32/get-started/index.html#get-started-set-up-env)

```
>ls -alh `which cmake`
-rwxr-xr-x  1 ...   8.5M Feb  5  2020 /Users/bittailor/.espressif/tools/cmake/3.16.4/CMake.app/Contents/bin/cmake
>ls -alh `which ncmake`
ls: found: No such file or directory
ls: ncmake: No such file or directory
ls: not: No such file or directory
>
```

Then I tried to create a terminal by "ESP-IDF: Open ESP-IDF Terminal" but this fails strangely by printing the export path twice put not complete and the waiting for a final " .

With the version from the vsix installer you provided the build works again and also the  "ESP-IDF: Open ESP-IDF Terminal" works.

poky, 2021-03-08:
Using the vsix installer provided by @brianignacio5 works fine!

closed 2021-03-08 by poky

lukehoersten, 2021-03-08:
@brianignacio5 the vsix worked for me as well. How long does it usually take to get this into the extensions marketplace for vscode?

itavero, 2021-03-08:
The VSIX works fine for me as well, but I'm also curious what is different compared to the version I can get from the marketplace (and when I can expect this fix in the marketplace).

brianignacio5, 2021-03-09:
@lukehoersten @itavero We don't a fixed release time, but for this sprint we should release 1.0.1 this March 12