espressif/vscode-esp-idf-extension #340: [Bug Report]: Linking C executable blink.elf FAILED: blink.elf

TerelleM, 2021-03-07:
**Describe the bug**
I have just installed VSCode and installed the ESP-IDF extension and have worked out a few bugs to get the blink example working. I needed to change the idf.py file and the component_get_requirements.cmake to fix the first two bugs. However, after generating the cmake files and when ninja starts building, it says their is no blink.elf file and no blink.map file.

**To Reproduce**
1) Install ESP-IDF extension
2) Build Blink example

**Expected behavior**
Program to build fine with no issues

**Screenshots**

ESP-IDF Custom Terminal
$ cmake
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
Python requirements from C:\Users\terel\esp\esp-idf\requirements.txt are satisfied.
Loading defaults file C:/Users/terel/OneDrive/Documents/Visual Studio Code Projects/test/blink/sdkconfig.defaults...
-- Components: 
-- Component paths: 
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/terel/OneDrive/Documents/Visual Studio Code Projects/test/blink/build

Terminal will be reused by tasks, press any key to close it.

> Executing task: espressif.esp-idf-extension: ESP-IDF Build <

ESP-IDF Custom Terminal
$ cmake
[1/1] Linking C executable blink.elf
FAILED: blink.elf 
cmd.exe /C "cd . && C:\Users\terel\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -mlongcalls -Wno-frame-address   CMakeFiles/blink.elf.dir/project_elf_src.c.obj  -o blink.elf  -Wl,--cref -Wl,--Map=C:/Users/terel/OneDrive/Documents/Visual Studio Code Projects/test/blink/build/blink.map  -fno-rtti  -fno-lto && cd ."
xtensa-esp32-elf-gcc.exe: error: Studio: No such file or directory
xtensa-esp32-elf-gcc.exe: error: Code: No such file or directory
xtensa-esp32-elf-gcc.exe: error: Projects/test/blink/build/blink.map: No such file or directory
ninja: build stopped: subcommand failed.
The terminal process failed to launch (exit code: 1).

**Environment**

- OS Name & Version: Win 10
- VSCode Version: q.54.1
- ESP-IDF Version: v4.2 (new release)
- Python Version: 3.9

**Logs**

**Cmake Log**
The target system is: Generic -  - 
The host system is: Windows - 10.0.19042 - AMD64
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: C:/Users/terel/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe 
Build flags: -mlongcalls;-Wno-frame-address
Id flags:  

The output was:
0

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.out"

The C compiler identification is GNU, found in "C:/Users/terel/OneDrive/Documents/Visual Studio Code Projects/test/blink/build/CMakeFiles/3.16.4/CompilerIdC/a.out"

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: C:/Users/terel/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe 
Build flags: -mlongcalls;-Wno-frame-address
Id flags:  

The output was:
[... 449 lines cut ...]
# ninja log v5
1	61	6368596071452068	project_elf_src.c	d47dcb6912c70225
62	131	6368596072150237	CMakeFiles/blink.elf.dir/project_elf_src.c.obj	2a87c01098ba0b8d

**Extension Configuration settings**
{
    "idf.espIdfPathWin": "C:\\Users\\terel\\esp\\esp-idf",
    "idf.pythonBinPathWin": "C:/Users/terel/.espressif/python_env/idf4.2_py3.9_env/Scripts/python.exe",
    "idf.customExtraPaths": "C:\\Users\\terel\\.espressif\\tools\\xtensa-esp32-elf\\esp-2020r3-8.4.0\\xtensa-esp32-elf\\bin;C:\\Users\\terel\\.espressif\\tools\\xtensa-esp32s2-elf\\esp-2020r3-8.4.0\\xtensa-esp32s2-elf\\bin;C:\\Users\\terel\\.espressif\\tools\\esp32ulp-elf\\2.28.51-esp-20191205\\esp32ulp-elf-binutils\\bin;C:\\Users\\terel\\.espressif\\tools\\esp32s2ulp-elf\\2.28.51-esp-20191205\\esp32s2ulp-elf-binutils\\bin;C:\\Users\\terel\\.espressif\\tools\\cmake\\3.16.4\\bin;C:\\Users\\terel\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20200709\\openocd-esp32\\bin;C:\\Users\\terel\\.espressif\\tools\\ninja\\1.10.0;C:\\Users\\terel\\.espressif\\tools\\idf-exe\\1.0.1;C:\\Users\\terel\\.espressif\\tools\\ccache\\3.7;C:\\Users\\terel\\.espressif\\tools\\dfu-util\\0.9\\dfu-util-0.9-win64",
    "idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"C:\\\\Users\\\\terel\\\\.espressif\\\\tools\\\\openocd-esp32\\\\v0.10.0-esp32-20200709/openocd-esp32/share/openocd/scripts\",\"IDF_CCACHE_ENABLE\":\"1\"}",
    "cmake.configureOnOpen": true,
    "workbench.editorAssociations": [
        {
            "viewType": "jupyter.notebook.ipynb",
            "filenamePattern": "*.ipynb"
        }
    ],
    "files.autoSave": "afterDelay"
}
- `idf.espIdfPath` (MacOS or Linux) or `idf.espIdfPathWin` (Windows) also called `Path to locate ESP-IDF framework (IDF_PATH)` in `Preferences: Open Settings (UI)`.
- `idf.pythonBinPath` (MacOS or Linux) or `idf.pythonBinPathWin` (Windows) also called `Python absolute binary path used to execute ESP-IDF Python Scripts` in `Preferences: Open Settings (UI)`.
- `idf.openOcdConfigs` also called `List of configuration files inside OpenOCD Scripts directory` in `Preferences: Open Settings (UI)`.
- `idf.customExtraPaths` also called `Paths to be appended to PATH` in `Preferences: Open Settings (UI)`.
- `idf.customExtraVars` also called `Variables to be added to system environment variables` in `Preferences: Open Settings (UI)`.
- `idf.adapterTargetName` also called `Target name for ESP-IDF Debug Adapter` in `Preferences: Open Settings (UI)`.

brianignacio5, 2021-03-08:
I think the real problem is that you are using a path with spaces in it and the xtensa toolchain doesn't handle it well.

xtensa-esp32-elf-gcc.exe: error: Studio: No such file or directory
xtensa-esp32-elf-gcc.exe: error: Code: No such file or directory
xtensa-esp32-elf-gcc.exe: error: Projects/test/blink/build/blink.map: No such file or directory

because `C:/Users/terel/OneDrive/Documents/Visual Studio Code Projects/test/blink/build/blink.map` . Can you try in a folder with no spaces ?

TerelleM, 2021-03-08:
> I think the real problem is that you are using a path with spaces in it and the xtensa toolchain doesn't handle it well.
>
> xtensa-esp32-elf-gcc.exe: error: Studio: No such file or directory [...]

Thanks Brian,

I've changed it to a new folder called VSC_Projects and created a new hello_world example, but the original problem still persists it cannot link the .elf file.

**ESP-IDF Custom Terminal**
$ cmake
[1/2] Building C object CMakeFiles/hello_world.elf.dir/project_elf_src.c.obj
[2/2] Linking C executable hello_world.elf
FAILED: hello_world.elf 
cmd.exe /C "cd . && C:\Users\terel\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -mlongcalls -Wno-frame-address   CMakeFiles/hello_world.elf.dir/project_elf_src.c.obj  -o hello_world.elf  -Wl,--cref -Wl,--Map=C:/Users/terel/OneDrive/Documents/VSC_Projects/test_hello_world/hello_world/build/hello_world.map  -fno-rtti  -fno-lto && cd ."
c:/users/terel/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/terel/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/lib/no-rtti/crt0.o:(.literal+0x0): undefined reference to `main'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
The terminal process failed to launch (exit code: 1).

igrr, 2021-03-08:
@TerelleM The linker line looks very odd, because no component libraries (.a files) are present. Are you building an unmodified blink example? If you have modified anything, could you link to or attach your modifications?

> I needed to change the idf.py file and the component_get_requirements.cmake to fix the first two bugs.

Could you describe the issues you have faced and what you have changed?

TerelleM, 2021-03-09:
Hey @igrr,

I've found the issue, my folder name was the original issue and VS reported multiple errors. After changing my modifications back to the original code and keeping the new folder name, everthing seems to be working fine.

Thanks for the quick feedback, im new to VS so It's much appreciated.

closed 2021-03-09 by TerelleM