espressif/vscode-esp-idf-extension #342: unable to use the latest esp idf vs code extension v 1.0.0

ankit-thealchemist, 2021-03-10:
Hello, 

When ever I installed latest version 1.0.0  of esp idf extension  and start the vs code I found the error in the vs code output window stated as the 
idf_versions.txt has (1) KB
idf_versions.txt progress: 100.00% (0.08 / 0.08) KB
'cmd' is not recognized as an internal or external command,
operable program or batch file.
Command failed: cmd /c echo idf4.3_py3.9_2021-01-07
'cmd' is not recognized as an internal or external command,
operable program or batch file.
Error checking idf-python-wheels version
Error: Command failed: cmd /c echo idf4.3_py3.9_2021-01-07
'cmd' is not recognized as an internal or external command,
operable program or batch file.

while I downgrade to the version 0.5.1 it works fine. I uses the esp-idf master and the python 3.7.

brianignacio5, 2021-03-10:
Can you uninstall your extension, erase `c:\Users\programmer.vscode\extensions\espressif.esp-idf-extension-1.0.0` and install [github master vsix installer](https://github.com/espressif/vscode-esp-idf-extension/suites/2213369212/artifacts/45786627) and try again ? 

We have made some changes to the terminal used by extension tasks and keep improving it, sorry for the issues.

ankit-thealchemist, 2021-03-10:
> Can you uninstall your extension, erase `c:\Users\programmer.vscode\extensions\espressif.esp-idf-extension-1.0.0` and install [github master vsix [...]

sorry to bother!  vs code is unable to see the cmd.exe, thus I added the system32 folder to the path variable. It is working now
! anyway thanks for helping
Closing the ticket!

closed 2021-03-10 by ankit-thealchemist