espressif/vscode-esp-idf-extension #343: Uploading data to SPIFFS[Bug Report]:

derflaryam, 2021-03-10:
Is there a way to upload SPIFFS content to the target as it is already inplemented in VSC Platformio ?

derflaryam, 2021-03-10:
[image]

brianignacio5, 2021-03-10:
We currently don't have this feature in this extension but we are setting up some ground work to implement it.

For now, you could execute `ESP-IDF: Open ESP-IDF Terminal`  to launch a terminal (with env variables loaded) and run
```
parttool.py --port "/dev/ttyUSB1" write_partition --partition-name=factory "factory.bin"
```
to write to partition 'factory' the contents of a file named 'factory.bin'

You can also use `esptool.py` to flash it as well

closed 2021-04-05 by github-actions[bot]