espressif/vscode-esp-idf-extension #346: [Bug Report]: Invalid ESP-IDF

orbitcoms, 2021-03-11:
Note sure if this is the right place to post but I am trying to install ESD-IDF plug in on Windows 10 machine
Installed GIT and Python, Used the Express Install method and select latest esp release and provided python path.

Right at end of installed (after showing ticks for each item being installed) I get "Invalid ESP-IDF" in terminal. The "Problems" Tab shows no issues listed.

I am finding the toolchain setup for ESP-IDF very tedious to get right.

brianignacio5, 2021-03-12:
We are very sorry for this @orbitcoms we made the fix in [github master vsix installer](https://github.com/espressif/vscode-esp-idf-extension/suites/2213369212/artifacts/45786627) but have yet to release to marketplace.

Can you try to uninstall the one from marketplace and use this vsix in the meantime. We should release soon a patch version today though.

orbitcoms, 2021-03-12:
Thank you for getting back to me on this.

Like I say, I am impressed with ESP32, ESP-IDF and The extension for VS-code, only issue is the install not setting up toolchain correctly.

For me, it is good if you run through say express install and it goes smoothly and works at end of process.

Regards
David Huisman

Orbit Communications Pty Ltd

BodyGuard Safety Solutions (A division of Orbit Communications Pty)
People & Plant Separation

100% Australian owned. All product designed and manufactured by Orbit Communications Pty, NSW Australia.

Web Site www.BodyGuardSafety.com.au<https://emea01.safelinks.protection.outlook.com/?url=http%3A%2F%2Fwww.bodyguardsafety.com.au%2F&data=04%7C01%7C%7C248e4971a4d04515f17b08d8df93e891%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637505173654178969%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=%2FDSq4EbrJPo4Co%2FRnd99cn3ZGly%2BsPcKIV1HdG7rqO8%3D&reserved=0>
Phone +61 2 43 554 554
29B Amsterdam Circuit
Wyong, NSW 2259
Australia

***@***.***D7175E.D6552D30]

orbitcoms, 2021-03-12:
Sorry, do I just unzip the vsix and double click it to install or do I point to it inside vscode. (I am new to vscode)

Regards
David Huisman

Orbit Communications Pty Ltd

BodyGuard Safety Solutions (A division of Orbit Communications Pty)
People & Plant Separation

100% Australian owned. All product designed and manufactured by Orbit Communications Pty, NSW Australia.

Web Site www.BodyGuardSafety.com.au<https://emea01.safelinks.protection.outlook.com/?url=http%3A%2F%2Fwww.bodyguardsafety.com.au%2F&data=04%7C01%7C%7C248e4971a4d04515f17b08d8df93e891%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637505173654178969%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=%2FDSq4EbrJPo4Co%2FRnd99cn3ZGly%2BsPcKIV1HdG7rqO8%3D&reserved=0>
Phone +61 2 43 554 554
29B Amsterdam Circuit
Wyong, NSW 2259
Australia

***@***.***D7175F.B6B5F8F0]

orbitcoms, 2021-03-12:
Brian,

I have installed the new extension you supplied and currently it is working through express install.

I left default source from Github, Release version 4.2 Python 3.9

Beside GIT and PYTHON are there any other dependencies that need to be installed prior to the extension?

Also, are there any things I need to do or paths to add after the install, or should this all be managed by the extension?

SUCCESS!! The installed completed and now in my project builds ok. I like the sdk config screen and not having to keep switching between ESP command line and vscode.

One thing a bit annoying still is that edit path setting does not fix the red squigglies

This is screen after building the Play_mp3 example from the ESP-ADF examples folder

***@***.***D71761.E0E80E90]

Regards
David Huisman

Orbit Communications Pty Ltd

BodyGuard Safety Solutions (A division of Orbit Communications Pty)
People & Plant Separation

100% Australian owned. All product designed and manufactured by Orbit Communications Pty, NSW Australia.

Web Site www.BodyGuardSafety.com.au<https://emea01.safelinks.protection.outlook.com/?url=http%3A%2F%2Fwww.bodyguardsafety.com.au%2F&data=04%7C01%7C%7C248e4971a4d04515f17b08d8df93e891%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637505173654178969%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=%2FDSq4EbrJPo4Co%2FRnd99cn3ZGly%2BsPcKIV1HdG7rqO8%3D&reserved=0>
Phone +61 2 43 554 554
29B Amsterdam Circuit
Wyong, NSW 2259
Australia

***@***.***D71761.47D49660]

orbitcoms, 2021-03-12:
Brian,

I followed the same steps for another machine at the office. This time it appeared to install ok but when I select to build the project it gets stick “Building Project” with no activity showing in the terminal like below..

***@***.***D7176F.5F728B70]

Regards
David Huisman

Orbit Communications Pty Ltd

BodyGuard Safety Solutions (A division of Orbit Communications Pty)
People & Plant Separation

100% Australian owned. All product designed and manufactured by Orbit Communications Pty, NSW Australia.

Web Site www.BodyGuardSafety.com.au<http://www.bodyguardsafety.com.au/>
Phone +61 2 43 554 554
29B Amsterdam Circuit
Wyong, NSW 2259
Australia

***@***.***D7176F.5F728B70]

This message and any attached files may contain information that is confidential and/or subject of legal privilege intended only for use by the intended recipient. If you are not the intended recipient or the person responsible for delivering the message to the intended recipient, be advised that you have received this message in error and that any dissemination, copying or use of this message or attachment is strictly forbidden, as is the disclosure of the information therein. If you have received this message in error please notify the sender immediately and delete the message.

orbitcoms, 2021-03-12:
Brian,

I can now confirm both the home and office machine have the same issue

They both compile ok if I manually enter the terminal and type “idf.py build”

But, when I use the extension icon to build, it never builds and forever stays as shown below.

The dialog shows blue progress bar going round and round but not actually performing the build.

It looks like a link between the build icon and idf.py command gets broken

***@***.***D71778.C51CEAC0]

Regards
David Huisman

Orbit Communications Pty Ltd

BodyGuard Safety Solutions (A division of Orbit Communications Pty)
People & Plant Separation

100% Australian owned. All product designed and manufactured by Orbit Communications Pty, NSW Australia.

Web Site www.BodyGuardSafety.com.au<https://emea01.safelinks.protection.outlook.com/?url=http%3A%2F%2Fwww.bodyguardsafety.com.au%2F&data=04%7C01%7C%7C248e4971a4d04515f17b08d8df93e891%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637505173654178969%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=%2FDSq4EbrJPo4Co%2FRnd99cn3ZGly%2BsPcKIV1HdG7rqO8%3D&reserved=0>
Phone +61 2 43 554 554
29B Amsterdam Circuit
Wyong, NSW 2259
Australia

***@***.***D71778.C51CEAC0]

brianignacio5, 2021-03-12:
Seems to me that something went wrong in the setup ? Can you see some error ESP-IDF Output (Menu View -> Output -> ESP-IDF tab) ?

When you build with vscode `ESP-IDF: Build your project` command, what is the build output (CTRL+SHIFT+`) and see what is printed in the build ? 

Try to use the `ESP-IDF: Doctor command` to generate a report of your configuration. It will be copied in your clipboard so can you paste it here.

here is a link to [setup documentation](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/SETUP.md) for more details.

orbitcoms, 2021-03-12:
Brian,

Output is blank.

Here is result from esp-idf: doctor

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19041
System environment variable PATH
 C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\nodejs\;C:\Program Files\dotnet\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\170\Tools\Binn\;C:\Program Files\Git\cmd;C:\Users\david\AppData\Local\Programs\Python\Python39\Scripts\;C:\Users\david\AppData\Local\Programs\Python\Python39\;C:\Users\david\AppData\Local\Programs\Microsoft VS Code\bin
Visual Studio Code version 1.0.0
Visual Studio Code language en
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe
ESP-IDF Extension version 1.54.2
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-IDF Path (idf.espIdfPath) c:\esp\esp-idf
Custom extra paths (idf.customExtraPaths) C:\Users\david\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin;C:\Users\david\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin;C:\Users\david\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\david\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\david\.espressif\tools\cmake\3.16.4\bin;C:\Users\david\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin;C:\Users\david\.espressif\tools\ninja\1.10.0;C:\Users\david\.espressif\tools\idf-exe\1.0.1;C:\Users\david\.espressif\tools\ccache\3.7;C:\Users\david\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"C:\\Users\\david\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20200709/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) C:\Users\david\.espressif\python_env\idf4.2_py3.9_env\Scripts\python.exe
Serial port (idf.port) /dev/ttyUSB1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) ${env:IDF_TOOLS_PATH}
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-IDF Custom extra paths
Access to C:\Users\david\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin: true
Access to C:\Users\david\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin: true
Access to C:\Users\david\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: true
Access to C:\Users\david\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: true
Access to C:\Users\david\.espressif\tools\cmake\3.16.4\bin: true
Access to C:\Users\david\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin: true
Access to C:\Users\david\.espressif\tools\ninja\1.10.0: true
Access to C:\Users\david\.espressif\tools\idf-exe\1.0.1: true
Access to C:\Users\david\.espressif\tools\ccache\3.7: true
Access to C:\Users\david\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) false
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.30.2.windows.1
ESP-IDF version 4.2
Python version 3.9.2
Python's pip version 21.0.1
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
bidict version: 0.21.2
bitstring version: 3.1.7
Brotli version: 1.0.9
certifi version: 2020.12.5
cffi version: 1.14.5
chardet version: 4.0.0
click version: 7.1.2
cryptography version: 3.4.6
ecdsa version: 0.16.1
esp-windows-curses version: 0.1
Flask version: 0.12.5
Flask-Compress version: 1.9.0
Flask-SocketIO version: 2.9.6
future version: 0.18.2
gcovr version: 4.2
gdbgui version: 0.13.2.0
gevent version: 1.5.0
greenlet version: 1.0.0
idna version: 2.10
itsdangerous version: 1.1.0
Jinja2 version: 2.11.3
lxml version: 4.6.2
MarkupSafe version: 1.1.1
pip version: 21.0.1
psutil version: 5.8.0
[... 148 lines cut ...]
            "cppStandard": "c++17",
            "intelliSenseMode": "clang-x64"
        }
    ],
    "version": 4
}
--------------------------------------------------------------------------------------------------------------------------------------------

Regards
David Huisman

Orbit Communications Pty Ltd

BodyGuard Safety Solutions (A division of Orbit Communications Pty)
People & Plant Separation

100% Australian owned. All product designed and manufactured by Orbit Communications Pty, NSW Australia.

Web Site www.BodyGuardSafety.com.au<https://emea01.safelinks.protection.outlook.com/?url=http%3A%2F%2Fwww.bodyguardsafety.com.au%2F&data=04%7C01%7C%7C248e4971a4d04515f17b08d8df93e891%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637505173654178969%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=%2FDSq4EbrJPo4Co%2FRnd99cn3ZGly%2BsPcKIV1HdG7rqO8%3D&reserved=0>
Phone +61 2 43 554 554
29B Amsterdam Circuit
Wyong, NSW 2259
Australia

***@***.***D71784.B65A1D80]

orbitcoms, 2021-03-12:
As mentioned, output window is blank

When I click to the small cylinder icon at bottom to “build” . This appears in terminal window only and no build action

If I go to the “D:\firmware\log\build> prompt and type “idf.py build” then the project builds ok and I see build progress happening in terminal. (second image below)

This is what I see when I click the build icon
***@***.***D71786.D92CFE20]

***@***.***D71786.D92CFE20]

This is a couple of snapshots showing build OK IF I manually type “idf.py build” in terminal
… ***@***.***D71786.E7F468D0]

***@***.***D71786.D92CFE20]

….

***@***.***D71786.D92CFE20]

Regards
David Huisman

Orbit Communications Pty Ltd

BodyGuard Safety Solutions (A division of Orbit Communications Pty)
People & Plant Separation

100% Australian owned. All product designed and manufactured by Orbit Communications Pty, NSW Australia.

Web Site www.BodyGuardSafety.com.au<https://emea01.safelinks.protection.outlook.com/?url=http%3A%2F%2Fwww.bodyguardsafety.com.au%2F&data=04%7C01%7C%7C248e4971a4d04515f17b08d8df93e891%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637505173654178969%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=%2FDSq4EbrJPo4Co%2FRnd99cn3ZGly%2BsPcKIV1HdG7rqO8%3D&reserved=0>
Phone +61 2 43 554 554
29B Amsterdam Circuit
Wyong, NSW 2259
Australia

***@***.***D71786.D92CFE20]

orbitcoms, 2021-03-12:
Brian,

I notice the extension updated itself from v1.0.0 to v1.01

On my office machine the installer installs the toolchain correctly but the idf.py prompt is not available.

It states it cannot find python.exe, even though the virtual environment install was successful for python 3.9

Regards
David Huisman

Orbit Communications Pty Ltd

BodyGuard Safety Solutions (A division of Orbit Communications Pty)
People & Plant Separation

100% Australian owned. All product designed and manufactured by Orbit Communications Pty, NSW Australia.

Web Site www.BodyGuardSafety.com.au<http://www.bodyguardsafety.com.au/>
Phone +61 2 43 554 554
29B Amsterdam Circuit
Wyong, NSW 2259
Australia

***@***.***D717E5.0812ABF0]

This message and any attached files may contain information that is confidential and/or subject of legal privilege intended only for use by the intended recipient. If you are not the intended recipient or the person responsible for delivering the message to the intended recipient, be advised that you have received this message in error and that any dissemination, copying or use of this message or attachment is strictly forbidden, as is the disclosure of the information therein. If you have received this message in error please notify the sender immediately and delete the message.

orbitcoms, 2021-03-12:
Brian,

This is very frustrating.

I have attached screen shots.

The installer reports as successful,
Python reported as requirements met ok
But no idf-py terminal available when I open terminal and it reports python.exe not found

If I click build icon, it shows the idf.py command available to use but the box at right says “Building” and never stops.
If I cancel the box, the idf commands are no longer available.

Extension version is 1.0.1 (I first install the version you sent me (1.0.0) and install but when I load vscode, it automatically updates to 1.0.1, not sure how this is affecting the app?

Regards
David Huisman

Orbit Communications Pty Ltd

BodyGuard Safety Solutions (A division of Orbit Communications Pty)
People & Plant Separation

100% Australian owned. All product designed and manufactured by Orbit Communications Pty, NSW Australia.

Web Site www.BodyGuardSafety.com.au<http://www.bodyguardsafety.com.au/>
Phone +61 2 43 554 554
29B Amsterdam Circuit
Wyong, NSW 2259
Australia

***@***.***D717E6.C3BC1840]

This message and any attached files may contain information that is confidential and/or subject of legal privilege intended only for use by the intended recipient. If you are not the intended recipient or the person responsible for delivering the message to the intended recipient, be advised that you have received this message in error and that any dissemination, copying or use of this message or attachment is strictly forbidden, as is the disclosure of the information therein. If you have received this message in error please notify the sender immediately and delete the message.

orbitcoms, 2021-03-14:
Brian,

I found the problem.

It was the “.vscode” folder

When I remove the .vscode folder from my projects, they all compile ok now using the extension.

Regards
David Huisman

Orbit Communications Pty Ltd

BodyGuard Safety Solutions (A division of Orbit Communications Pty)
People & Plant Separation

100% Australian owned. All product designed and manufactured by Orbit Communications Pty, NSW Australia.

Web Site www.BodyGuardSafety.com.au<http://www.bodyguardsafety.com.au/>
Phone +61 2 43 554 554
29B Amsterdam Circuit
Wyong, NSW 2259
Australia

***@***.***D71972.D7FC6ED0]

This message and any attached files may contain information that is confidential and/or subject of legal privilege intended only for use by the intended recipient. If you are not the intended recipient or the person responsible for delivering the message to the intended recipient, be advised that you have received this message in error and that any dissemination, copying or use of this message or attachment is strictly forbidden, as is the disclosure of the information therein. If you have received this message in error please notify the sender immediately and delete the message.

orbitcoms, 2021-03-14:
Brian,

I have 2 issues.

  1.  I have tried to fix these squiggly lines by including path but cannot get rid of them.

***@***.***D71974.8BF8FB50]

  1.  How do I get an IDF.PY command line terminal?

Regards
David Huisman

Orbit Communications Pty Ltd

BodyGuard Safety Solutions (A division of Orbit Communications Pty)
People & Plant Separation

100% Australian owned. All product designed and manufactured by Orbit Communications Pty, NSW Australia.

Web Site www.BodyGuardSafety.com.au<http://www.bodyguardsafety.com.au/>
Phone +61 2 43 554 554
29B Amsterdam Circuit
Wyong, NSW 2259
Australia

***@***.***D71974.8BF8FB50]

This message and any attached files may contain information that is confidential and/or subject of legal privilege intended only for use by the intended recipient. If you are not the intended recipient or the person responsible for delivering the message to the intended recipient, be advised that you have received this message in error and that any dissemination, copying or use of this message or attachment is strictly forbidden, as is the disclosure of the information therein. If you have received this message in error please notify the sender immediately and delete the message.

orbitcoms, 2021-03-15:
Brian,

I found I need to select the “Monitor” icon at bottom of screen and now have access to idf.py terminal commands ok.

Removed .vscode and recompiled and now projects compile ok.

Last thing remaining is that lots of #include statements have RED squiggly lines even though build succeeds.

Maybe this is inteli-sense  issue?

Is there something I can do to fix?

Thank you.

Regards
David Huisman

Orbit Communications Pty Ltd

BodyGuard Safety Solutions (A division of Orbit Communications Pty)
People & Plant Separation

100% Australian owned. All product designed and manufactured by Orbit Communications Pty, NSW Australia.

Web Site www.BodyGuardSafety.com.au<http://www.bodyguardsafety.com.au/>
Phone +61 2 43 554 554
29B Amsterdam Circuit
Wyong, NSW 2259
Australia

***@***.***D719AA.F4F7BC50]

brianignacio5, 2021-03-15:
Have you read how to configure the C/C++ Extension [here](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/C_CPP_CONFIGURATION.md) ?

orbitcoms, 2021-03-15:
Brian,

I had a look but confused what I need in the c_cpp_properties file for the IDF

The esp-idf extension generated the configuration below and it does not mention esp32 at all, just windows and visual studio?

"configurations": [
        {
            "name": "Win32",
            "includePath": [
                "${workspaceFolder}/**"
            ],
            "defines": [
                "_DEBUG",
                "UNICODE",
                "_UNICODE"
            ],
            "windowsSdkVersion": "10.0.18362.0",
            "compilerPath": "C:/Program Files (x86)/Microsoft Visual Studio/2019/Professional/VC/Tools/MSVC/14.28.29333/bin/Hostx64/x64/cl.exe",
            "cStandard": "c17",
            "cppStandard": "c++17",
            "intelliSenseMode": "windows-msvc-x64"
        }
    ],
    "version": 4
}

Regards
David Huisman

Orbit Communications Pty Ltd

BodyGuard Safety Solutions (A division of Orbit Communications Pty)
People & Plant Separation

100% Australian owned. All product designed and manufactured by Orbit Communications Pty, NSW Australia.

Web Site www.BodyGuardSafety.com.au<http://www.bodyguardsafety.com.au/>
Phone +61 2 43 554 554
29B Amsterdam Circuit
Wyong, NSW 2259
Australia

***@***.***D719CD.AA02A890]

This message and any attached files may contain information that is confidential and/or subject of legal privilege intended only for use by the intended recipient. If you are not the intended recipient or the person responsible for delivering the message to the intended recipient, be advised that you have received this message in error and that any dissemination, copying or use of this message or attachment is strictly forbidden, as is the disclosure of the information therein. If you have received this message in error please notify the sender immediately and delete the message.

orbitcoms, 2021-03-15:
Brian,

Actually, I now created a new project using extension from template and copied the resulting .vscode into my project and the intellisense is working.

Can I ask an unrelated question?

Do you have any detailed tutorial for spiffs?

My application has a 1M spiffs partition set up for logging a data file. Now I want to add html files for web server and need to allocate another 1M, but not sure how to allocate spiffs space for the web html code and to set only the html to program into flash.

Do I make 1 larger partition of 2M that is used for both my log file (uses Create, Read and Write file functions) and also for the html code?

I can find some info on partitions and other on spiffs and nvs but not an example how to put it together

{
    "configurations": [
        {
            "name": "ESP-IDF",
            "compilerPath": "C:\\esp\\tools\\.espressif\\tools\\xtensa-esp32-elf\\esp-2020r3-8.4.0\\xtensa-esp32-elf\\bin\\xtensa-esp32-elf-gcc.exe",
            "cStandard": "c11",
            "cppStandard": "c++17",
            "includePath": [
                "${config:idf.espIdfPath}/components/**",
                "${config:idf.espIdfPathWin}/components/**",
                "${config:idf.espAdfPath}/components/**",
                "${config:idf.espAdfPathWin}/components/**",
                "${workspaceFolder}/**"
            ],
            "browse": {
                "path": [
                    "${config:idf.espIdfPath}/components",
                    "${config:idf.espIdfPathWin}/components",
                    "${config:idf.espAdfPath}/components/**",
                    "${config:idf.espAdfPathWin}/components/**",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": false
            }
        }
    ],
    "version": 4
}

Regards
David Huisman

Orbit Communications Pty Ltd

BodyGuard Safety Solutions (A division of Orbit Communications Pty)
People & Plant Separation

100% Australian owned. All product designed and manufactured by Orbit Communications Pty, NSW Australia.

Web Site www.BodyGuardSafety.com.au<http://www.bodyguardsafety.com.au/>
Phone +61 2 43 554 554
29B Amsterdam Circuit
Wyong, NSW 2259
Australia

***@***.***D719CE.FBB15820]

This message and any attached files may contain information that is confidential and/or subject of legal privilege intended only for use by the intended recipient. If you are not the intended recipient or the person responsible for delivering the message to the intended recipient, be advised that you have received this message in error and that any dissemination, copying or use of this message or attachment is strictly forbidden, as is the disclosure of the information therein. If you have received this message in error please notify the sender immediately and delete the message.

brianignacio5, 2021-03-15:
For space allocation look into [Partition tables](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html) and the [ESP-IDF spiffs example](https://github.com/espressif/esp-idf/tree/master/examples/storage/spiffs) to consume spiffs

brianignacio5, 2021-03-15:
Closing this issue since original problem is fixed. If you need additional help you can ask other users in the [esp32 forum](https://esp32.com/viewforum.php?f=23)

closed 2021-03-15 by brianignacio5

orbitcoms, 2021-03-15:
Thanks Brian,

I have taken a look. Would you mind advising me if my understanding is as below.

1.
You can register multiple vfs by setting
.base_path -> path to where files are going to be located in spiffs partition
.partition_label -> name of spiffs partition to use

2.
You only need ONE init_nvs routine that enables access to the nvs partition for the whole project

3.
You can copy specific data to a spiffs partition at flash time using
…create_image(partition)name folder_location FLASH_IN_PROJECT), and this will copy file structure and files from “folder_location” in vscode folder to “folder_location” in spiffs partition. (I understand this does not create and actual folders containing files, but rather filenames that look like part of a directory hierarchy like “html/index.txt”.

4.
To work with files in a particular partition , fopen(partition_name/filename…)

5.
I assume partition_label and partition)name are the same?

Regards
David Huisman

Orbit Communications Pty Ltd

BodyGuard Safety Solutions (A division of Orbit Communications Pty)
People & Plant Separation

100% Australian owned. All product designed and manufactured by Orbit Communications Pty, NSW Australia.

Web Site www.BodyGuardSafety.com.au<http://www.bodyguardsafety.com.au/>
Phone +61 2 43 554 554
29B Amsterdam Circuit
Wyong, NSW 2259
Australia

***@***.***D71A3B.20EE26B0]

This message and any attached files may contain information that is confidential and/or subject of legal privilege intended only for use by the intended recipient. If you are not the intended recipient or the person responsible for delivering the message to the intended recipient, be advised that you have received this message in error and that any dissemination, copying or use of this message or attachment is strictly forbidden, as is the disclosure of the information therein. If you have received this message in error please notify the sender immediately and delete the message.

orbitcoms, 2021-03-16:
From the information in the spiffs and partition tables, would my steps below be correct to set up the 2 partitions and access data in the 2 folders?

I am a bit confused that the file command would use the same base_path? Is that correct.

***@***.***D71A68.DCC83BF0]

Regards
David Huisman

Orbit Communications Pty Ltd

BodyGuard Safety Solutions (A division of Orbit Communications Pty)
People & Plant Separation

100% Australian owned. All product designed and manufactured by Orbit Communications Pty, NSW Australia.

Web Site www.BodyGuardSafety.com.au<http://www.bodyguardsafety.com.au/>
Phone +61 2 43 554 554
29B Amsterdam Circuit
Wyong, NSW 2259
Australia

***@***.***D71A68.DCC83BF0]

This message and any attached files may contain information that is confidential and/or subject of legal privilege intended only for use by the intended recipient. If you are not the intended recipient or the person responsible for delivering the message to the intended recipient, be advised that you have received this message in error and that any dissemination, copying or use of this message or attachment is strictly forbidden, as is the disclosure of the information therein. If you have received this message in error please notify the sender immediately and delete the message.

snoop911, 2022-11-26:
I'm trying to install ESP-IDF for VS-Code 1.73 using the *RECOMMENDED* option of installing via the Extension.

So I installed the EspressIF Extension, added an empty folder to the workspace (otherwise it fails because of this), it still fails during the (EXPRESS) Install : 
"Invalid ESP-IDF"

Since I'm trying to Install ESP-IDF thru the extension, not sure why it needs a valid ESP-IDF to be selected?

brianignacio5, 2022-11-28:
Could you share the [troubleshooting information](https://github.com/espressif/vscode-esp-idf-extension#troubleshooting) as described in the readme so we can understand why is the esp-idf failing.