espressif/vscode-esp-idf-extension #352: [Bug Report]: ESP-IDF VS Code Extension always resets settings on startup.

StevenInacio, 2021-03-16:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [X] Make sure you have searched for existing bugs and features request before you post an issue.
- [X] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [X] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
Everytime I restart VS Code the esp-idf extension shows the toast, that esp-idf reconfigures itself. Maybe a necessary startup procedure. All good. AFTER THIS, however, it overwrites my custom user settings of the paths of some tools that I updated. Specifically the openocd version that I updated, to flash my esp32 over JTAG. Other path changes I haven't tested but my guess would be that it reverts them as well.

**To Reproduce**
Change a setting in "idf.customExtraPaths" or "idf.customExtraVars" to a newer tool version (e.g. openocd) and restart vs code.

**Expected behavior**
Custom settings should stay intact between vs code sessions, only changing when esp-idf updates them.

**Environment (please complete the following information):**

- OS Name & Version: Window 10.0.19042
- VSCode Version: 1.54.3 (user setup)
- ESP-IDF Version: 4.2
- Python Version: 3.9

**Logs**

<details>
<summary>esp_idf_vsc_ext.log</summary>
<pre><code>
{"message":"Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n\tat c:\\Users\\Steven Inacio\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.1\\dist\\extension.js:1:8461\n\tat ChildProcess.exithandler (child_process.js:303:7)\n\tat ChildProcess.emit (events.js:315:20)\n\tat ChildProcess.EventEmitter.emit (domain.js:483:12)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)","level":"error","timestamp":"2021-03-15T12:56:39.974Z"}
{"message":"Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n\tat c:\\Users\\Steven Inacio\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.1\\dist\\extension.js:1:8461\n\tat ChildProcess.exithandler (child_process.js:303:7)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)","level":"error","timestamp":"2021-03-15T12:56:54.637Z"}
{"message":"ENOENT: no such file or directory, open 'c:\\source\\esp-idf\\tools\\tools\\openocd-esp32\\v0.10.0-esp32-20200709\\openocd-esp32\\share\\openocd\\scripts\\esp-config.json'","stack":"Error: ENOENT: no such file or directory, open 'c:\\source\\esp-idf\\tools\\tools\\openocd-esp32\\v0.10.0-esp32-20200709\\openocd-esp32\\share\\openocd\\scripts\\esp-config.json'","level":"error","timestamp":"2021-03-15T12:58:36.699Z"}
{"user":true,"message":"OpenOCD Exit with non-zero error code 3","stack":"Error: Spawn exit with non-zero3\n\tat ChildProcess.<anonymous> (c:\\Users\\Steven Inacio\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.1\\dist\\extension.js:1:87978)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)","level":"error","timestamp":"2021-03-15T13:06:34.662Z"}
{"message":"non zero exit code 2\r\n\r\nERROR: Invalid pattern is specified in \"path:pattern\".\r\n","stack":"Error: non zero exit code 2\r\n\r\nERROR: Invalid pattern is specified in \"path:pattern\".\r\n\n\tat ChildProcess.<anonymous> (c:\\Users\\Steven Inacio\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.1\\dist\\extension.js:1:5604)\n\tat ChildProcess.emit (events.js:315:20)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:275:12)","level":"error","timestamp":"2021-03-15T13:06:35.151Z"}
{"message":"Cannot access filePath: c:\\source\\esp-idf\\tools\\python_env\\idf4.2_py3.9_env\\Scripts\\python.exe","level":"error","timestamp":"2021-03-15T13:06:35.152Z"}
{"user":true,"message":"OpenOCD Exit with non-zero error code 3","stack":"Error: Spawn exit with non-zero3\n\tat ChildProcess.<anonymous> (c:\\Users\\Steven Inacio\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.1\\dist\\extension.js:1:87978)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)","level":"error","timestamp":"2021-03-15T13:11:03.147Z"}
{"message":"non zero exit code 2\r\n\r\nERROR: Invalid pattern is specified in \"path:pattern\".\r\n","stack":"Error: non zero exit code 2\r\n\r\nERROR: Invalid pattern is specified in \"path:pattern\".\r\n\n\tat ChildProcess.<anonymous> (c:\\Users\\Steven Inacio\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.1\\dist\\extension.js:1:5604)\n\tat ChildProcess.emit (events.js:315:20)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:275:12)","level":"error","timestamp":"2021-03-15T13:11:03.424Z"}
{"message":"Cannot access filePath: c:\\source\\esp-idf\\tools\\python_env\\idf4.2_py3.9_env\\Scripts\\python.exe","level":"error","timestamp":"2021-03-15T13:11:03.424Z"}
{"user":true,"message":"Minimum OpenOCD version v0.10.0-esp32-20201125 is required while you have v0.10.0-esp32-20200709 version installed","stack":"Error: PRECHECK_FAILED\n\tat c:\\Users\\Steven Inacio\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.1\\dist\\extension.js:1:4682\n\tat Array.forEach (<anonymous>)\n\tat Function.perform (c:\\Users\\Steven Inacio\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.1\\dist\\extension.js:1:4635)\n\tat c:\\Users\\Steven Inacio\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.1\\dist\\extension.js:8:294478\n\tat c:\\Users\\Steven Inacio\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.1\\dist\\extension.js:8:263368\n\tat S._executeContributedCommand (c:\\Users\\Steven Inacio\\AppData\\Local\\Programs\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\services\\extensions\\node\\extensionHostProcess.js:87:103774)\n\tat S._doExecuteCommand (c:\\Users\\Steven Inacio\\AppData\\Local\\Programs\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\services\\extensions\\node\\extensionHostProcess.js:87:102930)\n\tat S.executeCommand (c:\\Users\\Steven Inacio\\AppData\\Local\\Programs\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\services\\extensions\\node\\extensionHostProcess.js:87:102836)\n\tat Object.executeCommand (c:\\Users\\Steven Inacio\\AppData\\Local\\Programs\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\services\\extensions\\node\\extensionHostProcess.js:95:22107)\n\tat c:\\Users\\Steven Inacio\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.1\\dist\\extension.js:8:296337\n\tat Generator.next (<anonymous>)\n\tat c:\\Users\\Steven Inacio\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.1\\dist\\extension.js:8:262033\n\tat new Promise (<anonymous>)\n\tat ve (c:\\Users\\Steven Inacio\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.1\\dist\\extension.js:8:261778)\n\tat c:\\Users\\Steven Inacio\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.1\\dist\\extension.js:8:296015\n\tat Function.perform (c:\\Users\\Steven Inacio\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.1\\dist\\extension.js:1:4724)\n\tat c:\\Users\\Steven Inacio\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.1\\dist\\extension.js:8:295998\n\tat c:\\Users\\Steven Inacio\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.1\\dist\\extension.js:8:263368\n\tat S._executeContributedCommand (c:\\Users\\Steven Inacio\\AppData\\Local\\Programs\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\services\\extensions\\node\\extensionHostProcess.js:87:103774)\n\tat S.$executeContributedCommand (c:\\Users\\Steven Inacio\\AppData\\Local\\Programs\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\services\\extensions\\node\\extensionHostProcess.js [... cut ...] binutils\\bin;C:\\.espressif\\tools\\cmake\\3.16.4\\bin;C:\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20201202\\openocd-esp32\\bin;C:\\.espressif\\tools\\ninja\\1.10.0;C:\\.espressif\\tools\\idf-exe\\1.0.1;C:\\.espressif\\tools\\ccache\\3.7;C:\\.espressif\\tools\\dfu-util\\0.9\\dfu-util-0.9-win64",
    "idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"C:\\\\.espressif\\\\tools\\\\openocd-esp32\\\\v0.10.0-esp32-20201202/openocd-esp32/share/openocd/scripts\",\"IDF_CCACHE_ENABLE\":\"1\"}",
}
```

brianignacio5, 2021-03-22:
Thanks for reporting this. Will provide a fix soon, sorry for the inconvenience. :(

brianignacio5, 2021-03-25:
@StevenInacio Can you please try to install this [vsix](https://github.com/espressif/vscode-esp-idf-extension/suites/2341110635/artifacts/49411068) and see if it fixed ?

PR #359 (2021-04-06): remove idf tools version match on current setup
Fix #352 

PTAL @StevenInacio 

We were validating ESP-IDF Tools for the given ESP-IDF and if Tools didn't match with version in ESP-IDF, the extension tries to self configure. This doesn't allow for custom setups such as using the latest openOCD. Now we just make sure that tools provide a version string that is valid to IDF_PATH/tools.json

Add build from source in contributing documentation.
closed 2021-04-06 by brianignacio5