espressif/vscode-esp-idf-extension #355: Step by step documentation for integrating vscode-esp-idf-extension on AWS FreeRTOS ESP32

Yoimer, 2021-03-21:
How can we build, flash and compile AWS ESP32 `FreeRTOS V202012.00` using vscode-esp-idf-extension on Windows? The closest information regarding this, is here https://www.freertos.org/2021/01/using-visual-studio-code-for-freertos-development.html, but unfortunately is not clear enough.

Currently I handle this project from `prompt `  using `ESP-IDFV3.3.2`, and the toolchain installer was `esp-idf-tools-setup-2.4.exe`

In order to develop `AWS ESP32 FreeRTOS V202012.00` projects, we download the zip file from AWS console and place it on `ESP-IDFV3.3.2` root installation. Then modifying `IDF_PATH` on `C:\Users\Yoimer\.espressif\idf_cmd_init.bat ` we can make the `ESP-IDF Command Prompt (cmd.exe)` call FreeRTOS. (please see `idf_cmd_init.txt` attachment)

[idf_cmd_init.txt](https://github.com/espressif/vscode-esp-idf-extension/files/6176854/idf_cmd_init.txt)

Just to make things faster, I uploaded a clear zip file generated from AWS FreeRTOS on this bucket. Please feel free to download it https://s3-us-west-2.amazonaws.com/freertos-v202012.00/freertos-1615096093296.zip.

When executing `ESP-IDF Command Prompt (cmd.exe)`, it shows this

```
Using Python in C:\Users\Yoimer\.espressif\tools\idf-python\3.8.7\
Python 3.8.7
Using Git in C:\git\cmd\
git version 2.30.2.windows.1
Setting IDF_PATH: C:\esp-idf-v3.3.2\FreeRTOS\vendors\espressif\esp-idf

Adding ESP-IDF tools to PATH...
    C:\Users\Yoimer\.espressif\tools\xtensa-esp32-elf\1.22.0-80-g6c4433a5-5.2.0\xtensa-esp32-elf\bin
    C:\Users\Yoimer\.espressif\tools\esp32ulp-elf\2.28.51.20170517\esp32ulp-elf-binutils\bin
    C:\Users\Yoimer\.espressif\tools\cmake\3.13.4\bin
    C:\Users\Yoimer\.espressif\tools\openocd-esp32\v0.10.0-esp32-20190313\openocd-esp32\bin
    C:\Users\Yoimer\.espressif\tools\mconf\v4.6.0.0-idf-20190628\
    C:\Users\Yoimer\.espressif\tools\ninja\1.9.0\
    C:\Users\Yoimer\.espressif\tools\idf-exe\1.0.1\
    C:\Users\Yoimer\.espressif\tools\ccache\3.7\
    C:\Users\Yoimer\.espressif\python_env\idf3.3_py3.8_env\Scripts
    C:\esp-idf-v3.3.2\FreeRTOS\vendors\espressif\esp-idf\tools

Checking if Python packages are up to date...
Python requirements from C:\esp-idf-v3.3.2\FreeRTOS\vendors\espressif\esp-idf\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

C:\esp-idf-v3.3.2>
```
Then we change directory to `FreeRTOS` folder and execute the commands for the respective demos, like this:

For `coreMQTT` demo

`cmake -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 -GNinja -S . -B demos/coreMQTT/build`
`cmake --build  demos/coreMQTT/build`, however this `python %IDF_PATH%/tools/idf.py build -B demos/coreMQTT/build` also works.
`cmake demos/coreMQTT/build --target flash`,  however this `python %IDF_PATH%/tools/idf.py erase_flash flash demos/coreMQTT/build` works also and even includes  `erase_flash` to clean all the flash before flashing again.
`python %IDF_PATH%/tools/idf.py monitor -p COM4 -B demos/coreMQTT/build`, where `COM4` is the COM port assigned to your ESP32.
For the rests of the demos, like `ota` and `wifi_provisioning` is the same procedure.

Is there any document or explanation where people could see how to do all what's explained above from vscode-esp-idf-extension  on AWS FreeRTOS ESP32?.

 I am able to use vscode-esp-idf-extension for building and flashing with any of the demos on `C:\esp-idf-v3.3.2`. I'd like to do the same with AWS FreeRTOS ESP32 V202012.00. 
 
 Thanks a lot for you assistance

brianignacio5, 2021-03-24:
Hey @Yoimer sorry for late reply, as far as I understand you are just using a modified ESP-IDF right?

Base on [Setup JSON Manual Configuration](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/SETUP.md#json-manual-configuration) you just need to update the `idf.espIdfPathWin` value in your Visual Studio Code settings to `C:\esp-idf-v3.3.2\FreeRTOS\vendors\espressif\esp-idf`.

You might also need to update `idf.customExtraPaths` with those values from `export.bat` (those from Adding ESP-IDF tools to PATH) separated with `;` as explained in the previous link. Also `idf.customExtraVars` with the right openOCD version. You can add extra environment variables here if needed.

For Intellisense configuration, take a look at [C/C++ Configuration](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/C_CPP_CONFIGURATION.md)

Yoimer, 2021-03-24:
@brianignacio5 Thanks a lot for replying, very kind from you and your team. Actually I am not using a modified ESP-IDF, I am using a native one. It might be `v3.3.2` or `v3.3` or any of them.

AWS FreeRTOS ESP32 is a customized (modified) version of the FreeRTOS implemented by Expressif, however it used native ESP-IDF to work.

https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html

Let's say I use `ESP-IDF v3.3` and I install the `https://dl.espressif.com/dl/esp-idf-tools-setup-1.2.exe` as mentioned here
https://docs.espressif.com/projects/esp-idf/en/v3.3/get-started-cmake/windows-setup.html.

This installs all the packages required for AWS FreeRTOS ESP32 to work on my Windows path, so my `cmd` I can all the commands needed. So after downloading the zip file generated from AWS, I unzipped it on `C:\`, (it creates a folder called `FreeRTOS`) then I `cd` to `FreeRTOS` folder and execute the next commands from `cmd` for the `ota` demo (you choose whichever demo you want to test) .

For building files creation:
`cmake -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 -GNinja -S . -B demos/ota/build`

For building and flashing to the ESP32 (it detects automatically the `COM` port)
`cmake --build demos/ota/build --target flash` 

So my question is: How could I pass the above commands from `vscode-esp-idf-extension`?

As far as I understand, VS code is capable to configure `CMake` commands.

I want to use VSCODE as an `IDE` for AWS FreeRTOS ESP32 projects. I plan to use an `ESP-PROG` hardware debugger even, for breakpoints etc. 

I know VSCODE is capable to be an IDE for ESP-IDF native examples, and I can do that with AWS FreeRTOS ESP32.

On their blog [freertos-and-vscode](https://www.freertos.org/2021/01/using-visual-studio-code-for-freertos-development.html) they mention this is possible, but it is not that clear.

Would you be kind enough to check this out using `vscode-esp-idf-extension`, please?

Thanks in advance.

Yoimer, 2021-03-30:
@brianignacio5 I made some progress about this. It's almost down. Please, would you be kind enough to check it out here https://github.com/aws/amazon-freertos/issues/3002#issuecomment-810437254?

If you know that last two questions, please let me know any comments.

Thanks in advance.

brianignacio5, 2021-04-06:
Ok I understand now. So basically add CMake arguments right?

#363 implement these changes. You can install this [vsix installer](https://github.com/espressif/vscode-esp-idf-extension/suites/2429721859/artifacts/51893732) to try it. Basically adding `idf.cmakeCompilerArgs` for compilation task arguments and `idf.cmakeBuildArgs` for the build task arguments. Defaults values are:

`idf.cmakeCompilerArgs`: ["-G", "Ninja", ".."] which will run a task `cmake -G Ninja ..`

`idf.cmakeBuildArgs`: ["--build", "."] which will run a task `cmake --build .`

Flash should work just running `ESP-IDF: Flash your project` if build was successful.

Yoimer, 2021-04-06:
@brianignacio5 Thanks for your time and support. That's very kind from you. I will test this and let you know.

brianignacio5, 2021-04-21:
Can we close this @Yoimer ?

Yoimer, 2021-04-21:
> Can we close this @Yoimer ?

@brianignacio5 Hello. Thanks for your support. Yes, please close. Have a great day

PR #363 (2021-04-23): Custom CMake arguments use extra_esptool_args
Add `idf.cmakeCompilerArgs` for compilation task arguments and `idf.cmakeBuildArgs` for the build task arguments.

Use `extra_esptool_args` and encrypted flags from `build/flasher_args.json` to improve extension flash command.

Fix #355 

Fix #373 

PTAL @Yoimer
closed 2021-04-23 by brianignacio5