espressif/vscode-esp-idf-extension #356: [Bug Report]:  Can Compile and Flash but not monitor

orbitcoms, 2021-03-23:
I am running vscode on Windows 10 machine. Python 3.9 and git 2.3 are installed

Once this issue showed up I uninstalled vscode and esp-idf and started again. Installed a clean copy of vscode and installed the C/C++ extension and then the esp-idf extension (version 1.0.1) but problem still exists.

My project compiles and flashes ok but when I attempt to run Monitor, I get error "The terminal process failed to launch: Path to shell executable "powershell.exe" does not exist"

I did not see powershell as pre-requisite for the extension so I assumed if it needs it then it will install it?

0phoff, 2021-03-23:
Not sure if my issue is related, but I think it is, so I will add it here.

I have setup VS Code on my Linux machine (Ubuntu 20.04), so that terminals are opened with a special environment flag.
More specifically, I used this in _settings.json_, so that my _.bashrc_ file knows not to automatically start TMUX:
```json
{
    "terminal.integrated.env.linux": {
      "NO_TMUX": "1"
    },
}
```

This is something special about my system, but the important part is that this works when I start a Terminal from VS Code, and also when running the `build`, `clean` and `flash` commands with the little buttons at the bottom of the app or through the command palette.
However, when using the `monitor` button, it does not obey these rules and starts a regular terminal, which in my case results in a TMUX terminal.

__Running the monitor command by going to `Terminal > Run Task > Monitor` does work!__

---

Note that I do not expect you to completely solve my issue, as I have a highly customized setup, but I think my issue shows that there is a bigger problem with the _"ESP-IDF: Monitor your device"_ command, which might be solved by using the default VS Code terminal and running the monitor command there ?

brianignacio5, 2021-03-24:
@orbitcoms To launch the ESP-IDF terminal, we use the shell executable given in `vscode.env.shell` which is override by `terminal.integrated.shell.windows` in your Visual Studio Code Settings or use the `Terminal: Select Default Shell` command to update the shell. I have tested in Windows 10 that the IDF Monitor command picks changes in `terminal.integrated.shell.windows` 

@0phoff about your case, we use environment variables from `idf.customExtraVars` and other IDF specific settings. You can either set `"NO_TMUX": "1"` globally in your system or in `idf.customExtraVars` for the extension to pick this variable.

orbitcoms, 2021-03-24:
I don't understand what this means.
Are you saying the extension does not set the variables or commands required to run the monitor and we do this manually after installing the extension?
I tried Terminal: Default Shell (I tried all 3 options, Powershell failed, bash and cmd both did nothing)

Where do I set this ide.customextravars? do you have steps what I need to change?

brianignacio5, 2021-03-25:
hey @orbitcoms I guess I found the issue. **The problem is we are just using `powershell.exe` instead of the full path of the shell. Will fix soon.** 

I guess initially I didn't notice because shell paths are usually in the environment PATH, which makes me believe that for some reason `"C:\WINDOWS\System32\WindowsPowerShell\v1.0\` is not in your PATH, which I guess it is by default ?

orbitcoms, 2021-03-25:
Thanks Brian,

I am at office at the moment where the Monitor function does work and the PC does have path to powershell.
I suspect you are right and my home computer does not have path set for powershell.
Does the extension install powershell? If not, what is the best method to install it. (I see online several versions and not sure what is required for ESP-IDF, and assumed it was installed as part of the toolchain when configuring the extension)

brianignacio5, 2021-03-25:
You don't need to install Powershell.exe since it comes with Windows. v1.0 is the old version that comes by default but there are several versions and all should work in my opinion. For the moment, just make sure it is in your system environment variable Path. I will fix the extension so "terminal.integrated.shell.windows": "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe" works without any env variable modification.

orbitcoms, 2021-03-25:
I manually set the path in environment for powershell and now Monitor works ok,

Thanks

Regards
David Huisman

Orbit Communications Pty Ltd

BodyGuard Safety Solutions (A division of Orbit Communications Pty)
People & Plant Separation

100% Australian owned. All product designed and manufactured by Orbit Communications Pty, NSW Australia.

Web Site www.BodyGuardSafety.com.au<https://emea01.safelinks.protection.outlook.com/?url=http%3A%2F%2Fwww.bodyguardsafety.com.au%2F&data=04%7C01%7C%7C248e4971a4d04515f17b08d8df93e891%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637505173654178969%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=%2FDSq4EbrJPo4Co%2FRnd99cn3ZGly%2BsPcKIV1HdG7rqO8%3D&reserved=0>
Phone +61 2 43 554 554
29B Amsterdam Circuit
Wyong, NSW 2259
Australia

***@***.***D721A7.4D5F8680]

PR #358 (2021-03-31): Enhance/kill monitor timeout
Use full path for shell executable. 

Add timeout when killing monitor terminal process.

Fix #356 
Fix #329
closed 2021-03-31 by brianignacio5