espressif/vscode-esp-idf-extension #367: [Bug Report]: Installation of toolchain fails

karimhasebou, 2021-04-09:
A clear and concise description of what the bug is (current behavior).
The installation setup fails and shows that some python requirements are not satisfied  (log 1 )and tells me to run
/Users/karimhasebou/esp/esp-idf/install.sh but that fails as well (log 2)

**Environment :

- OS Name & Version: MacOS X 10.15
- VSCode Version: 
- ESP-IDF Version: 4.2 release
- Python Version: 3.9
 
Logs

1) 

{ error: [Error: non zero exit code 1 The following Python requirements are not satisfied: pyserial>=3.0 future>=0.15.2 cryptography>=2.1.4 pyparsing>=2.0.3,<2.4.0 pyelftools>=0.22 gdbgui==0.13.2.0 reedsolo>=1.5.3,<=1.5.4 bitstring>=3.1.6 ecdsa>=0.16.0 To install the missing packages, please run "/Users/karimhasebou/esp/esp-idf/install.sh" Diagnostic information: IDF_PYTHON_ENV_PATH: /Users/karimhasebou/.espressif/python_env/idf4.2_py3.8_env Python interpreter used: /Users/karimhasebou/.espressif/python_env/idf4.2_py3.8_env/bin/python ESP-IDF v4.2-dirty at ChildProcess.<anonymous> (/Users/karimhasebou/.vscode/extensions/espressif.esp-idf-extension-1.0.2/dist/extension.js:1:5662) at ChildProcess.emit (events.js:315:20) at Process.ChildProcess._handle.onexit (internal/child_process.js:275:12)] }

2)
karimhasebou@Karims-MacBook-Pro ~ % /Users/karimhasebou/esp/esp-idf/install.sh 
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp-2020r3-8.4.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2020r3-8.4.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20200709 (already installed)
Installing Python environment and packages
Creating a new Python environment in /Users/karimhasebou/.espressif/python_env/idf4.2_py2.7_env
Installing virtualenv
/System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python: No module named pip
Traceback (most recent call last):
  File "/Users/karimhasebou/esp/esp-idf/tools/idf_tools.py", line 1492, in <module>
    main(sys.argv[1:])
  File "/Users/karimhasebou/esp/esp-idf/tools/idf_tools.py", line 1488, in main
    action_func(args)
  File "/Users/karimhasebou/esp/esp-idf/tools/idf_tools.py", line 1205, in action_install_python_env
    stdout=sys.stdout, stderr=sys.stderr)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 190, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python', '-m', 'pip', 'install', '--user', 'virtualenv']' returned non-zero exit status 1

karimhasebou, 2021-04-09:
the installation script seems to be running python2 instead of 3, ignoring the presets chosen in the extension

brianignacio5, 2021-04-12:
How are you installing ESP-IDF and tools? With the `ESP_IDF: Configure ESP-IDF extension` command ?

For some reason it is picking `/System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python` as python executable, which doesn't seem to have pip as module. Maybe check your Python install is correct. It could also be that you are running the setup from a directory with a `.vscode/settings.json` that is overriding your user level settings.

**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [ ] Make sure you have searched for existing bugs and features request before you post an issue.
- [ ] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [ ] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
A clear and concise description of what the bug is (current behavior).

**To Reproduce**
Steps to reproduce the behavior (including the full log).

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**

- OS Name & Version: [e.g. MacOS X 10.14.5]
- VSCode Version: [e.g. 1.35.1]
- ESP-IDF Version: [e.g. v3.2.2]
- Python Version: [e.g. 2.7.10]

> **NOTE**: You can use the `ESP-IDF: Doctor command` to generate a report of your configuration.

**Logs**
If applicable, please share the log file which can be obtained from

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

**Extension Configuration settings**
Open Command Palette (F1) and type `Preferences: Open Settings (JSON)` or use `{CURRENT_PROJECT}/.vscode/settings.json` to get the following extension configuration settings value:

- `idf.espIdfPath` (MacOS or Linux) or `idf.espIdfPathWin` (Windows) also called `Path to locate ESP-IDF framework (IDF_PATH)` in `Preferences: Open Settings (UI)`.
- `idf.pythonBinPath` (MacOS or Linux) or `idf.pythonBinPathWin` (Windows) also called `Python absolute binary path used to execute ESP-IDF Python Scripts` in `Preferences: Open Settings (UI)`.
- `idf.openOcdConfigs` also called `List of configuration files inside OpenOCD Scripts directory` in `Preferences: Open Settings (UI)`.
- `idf.customExtraPaths` also called `Paths to be appended to PATH` in `Preferences: Open Settings (UI)`.
- `idf.customExtraVars` also called `Variables to be added to system environment variables` in `Preferences: Open Settings (UI)`.
- `idf.adapterTargetName` also called `Target name for ESP-IDF Debug Adapter` in `Preferences: Open Settings (UI)`.
- `idf.customAdapterTargetName` is used when `idf.adapterTargetName` is set to `custom`.

**Additional context**
Add any other context about the problem here.

brianignacio5, 2021-04-23:
Closing this issue for lack of response from user.

closed 2021-04-23 by brianignacio5