espressif/vscode-esp-idf-extension #368: Add Arduino ESP32 as ESP-IDF Component issue

versamodule, 2021-04-09:
Using v1.0.2. 
When I use Add Arduino ESP32 as ESP-IDF Component, i get the following error. 

[image]

brianignacio5, 2021-04-12:
Can you provide more information about your setup as given in the Issue template?

The Arduino ESP32 is just being cloned using `git clone --recursive ARDUINO_ESP32_REPO`. Maybe there is a connection issue ?

**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [ ] Make sure you have searched for existing bugs and features request before you post an issue.
- [ ] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [ ] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
A clear and concise description of what the bug is (current behavior).

**To Reproduce**
Steps to reproduce the behavior (including the full log).

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**

- OS Name & Version: [e.g. MacOS X 10.14.5]
- VSCode Version: [e.g. 1.35.1]
- ESP-IDF Version: [e.g. v3.2.2]
- Python Version: [e.g. 2.7.10]

> **NOTE**: You can use the `ESP-IDF: Doctor command` to generate a report of your configuration.

**Logs**
If applicable, please share the log file which can be obtained from

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

**Extension Configuration settings**
Open Command Palette (F1) and type `Preferences: Open Settings (JSON)` or use `{CURRENT_PROJECT}/.vscode/settings.json` to get the following extension configuration settings value:

- `idf.espIdfPath` (MacOS or Linux) or `idf.espIdfPathWin` (Windows) also called `Path to locate ESP-IDF framework (IDF_PATH)` in `Preferences: Open Settings (UI)`.
- `idf.pythonBinPath` (MacOS or Linux) or `idf.pythonBinPathWin` (Windows) also called `Python absolute binary path used to execute ESP-IDF Python Scripts` in `Preferences: Open Settings (UI)`.
- `idf.openOcdConfigs` also called `List of configuration files inside OpenOCD Scripts directory` in `Preferences: Open Settings (UI)`.
- `idf.customExtraPaths` also called `Paths to be appended to PATH` in `Preferences: Open Settings (UI)`.
- `idf.customExtraVars` also called `Variables to be added to system environment variables` in `Preferences: Open Settings (UI)`.
- `idf.adapterTargetName` also called `Target name for ESP-IDF Debug Adapter` in `Preferences: Open Settings (UI)`.
- `idf.customAdapterTargetName` is used when `idf.adapterTargetName` is set to `custom`.

**Additional context**
Add any other context about the problem here.

versamodule, 2021-04-12:
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19042 
System environment variable PATH 
 C:\Program Files (x86)\VMware\VMware Player\bin\;C:\Program Files (x86)\AdoptOpenJDK\jdk-8.0.275.01-hotspot\bin;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\IDM Computer Solutions\UltraEdit;C:\Program Files\Git\cmd;C:\nburn\gcc\bin;C:\nburn\gcc\msys\1.0\bin;C:\Program Files\nodejs\;C:\nburn\pcbin;C:\nburn\gcc-m68k\bin;C:\nburn\gcc-m68k\msys\1.0\bin;C:\Android;C:\Users\Jim\AppData\Local\Programs\Python\Python39\Scripts\;C:\Users\Jim\AppData\Local\Programs\Python\Python39\;C:\Users\Jim\AppData\Local\Microsoft\WindowsApps;;C:\Users\Jim\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\Jim\AppData\Roaming\npm 
Visual Studio Code version 1.0.2 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.55.1 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-IDF Path (idf.espIdfPath) C:\Users\Jim\esp\esp-idf
Custom extra paths (idf.customExtraPaths) C:\Users\Jim\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin;C:\Users\Jim\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin;C:\Users\Jim\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\Jim\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\Jim\.espressif\tools\cmake\3.16.4\bin;C:\Users\Jim\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin;C:\Users\Jim\.espressif\tools\ninja\1.10.0;C:\Users\Jim\.espressif\tools\idf-exe\1.0.1;C:\Users\Jim\.espressif\tools\ccache\3.7;C:\Users\Jim\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"C:\\Users\\Jim\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20200709/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) C:\Users\Jim\.espressif\python_env\idf4.2_py3.9_env\Scripts\python.exe
Serial port (idf.port) /dev/ttyUSB1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) 
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-IDF Custom extra paths
Access to C:\Users\Jim\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin: true
Access to C:\Users\Jim\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin: true
Access to C:\Users\Jim\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: true
Access to C:\Users\Jim\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: true
Access to C:\Users\Jim\.espressif\tools\cmake\3.16.4\bin: true
Access to C:\Users\Jim\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin: true
Access to C:\Users\Jim\.espressif\tools\ninja\1.10.0: true
Access to C:\Users\Jim\.espressif\tools\idf-exe\1.0.1: true
Access to C:\Users\Jim\.espressif\tools\ccache\3.7: true
Access to C:\Users\Jim\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) false
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.30.0.windows.2
ESP-IDF version 4.2
Python version 3.9.1
Python's pip version 20.2.3
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
bidict version: 0.21.2
bitstring version: 3.1.7
Brotli version: 1.0.9
certifi version: 2020.12.5
cffi version: 1.14.4
chardet version: 4.0.0
click version: 7.1.2
cryptography version: 3.3.1
ecdsa version: 0.16.1
esp-windows-curses version: 0.1
Flask version: 0.12.5
Flask-Compress version: 1.8.0
Flask-SocketIO version: 2.9.6
future version: 0.18.2
gcovr version: 4.2
gdbgui version: 0.13.2.0
gevent version: 1.5.0
greenlet version: 1.0.0
idna version: 2.10
itsdangerous version: 1.1.0
Jinja2 version: 2.11.3
lxml version: 4.6.2
MarkupSafe version: 1.1.1
pip version: 20.2.3
psutil version: 5.8.0
pycparser version: 2.20
pyelftools version: 0.27
pygdbmi version: 0.9.0.2
Pygments version: 2.7.4
pyparsing version: 2.3.1
pyserial version: 3.5
[... 41 lines cut ...]
            "cStandard": "c11",
            "cppStandard": "c++17",
            "includePath": [
                "${config:idf.espIdfPath}/components/**",
                "${config:idf.espIdfPathWin}/components/**",
                "${config:idf.espAdfPath}/components/**",
                "${config:idf.espAdfPathWin}/components/**",
                "${workspaceFolder}/**"
            ],
            "browse": {
                "path": [
[... cut ...] s",
                    "${config:idf.espIdfPathWin}/components",
                    "${config:idf.espAdfPath}/components/**",
                    "${config:idf.espAdfPathWin}/components/**",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": false
            }
        }
    ],
    "version": 4
}
 
--------------------------------------------------------------------------------------------------------------------------------------------

brianignacio5, 2021-04-13:
In your project directory, there is probably an existing `components/arduino` folder which you can delete first and try again. Code 128 seems to be an error that cloning directory that already exists. You can check in `Output` -> `ESP-IDF` the error message in detail.

We missed to validate this, so we made a fix in [this PR](https://github.com/espressif/vscode-esp-idf-extension/pull/370) and you can try using this [vsix installer](https://github.com/espressif/vscode-esp-idf-extension/suites/2482096455/artifacts/53463720) to see if the notification is shown.

versamodule, 2021-04-13:
"vsix installer" link is a 404

nprasan, 2021-04-13:
Facing the same issue with extension v1.0.2.
`Output ` -> `ESP-IDF` shows the following error message:

```
Cloning into 'arduino'...

fatal: Remote branch idf-release/v4.2 not found in upstream origin

Arduino ESP32 cloning has exit with 128
```

It seems that the idf-release/v4.2 branch (along with a bunch of other idf-release branches) has been deleted from the arduino-esp32 GitHub repository. See [link](https://github.com/espressif/arduino-esp32/issues/5029).

brianignacio5, 2021-04-13:
Yes @nprasan this is fixed on #370 

Maybe go to the PR and get the vsix from the PR artifact @versamodule Link seems to work for me

nprasan, 2021-04-13:
@brianignacio5 The linked vsix solves the issue. Thanks.

brianignacio5, 2021-04-15:
@versamodule Apparently artifacts links are only available if you are logged to Github.

Maybe try [this one](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/53463720.zip) ?

closed 2021-04-23 by brianignacio5