espressif/vscode-esp-idf-extension #371: [Bug Report]: Can not install ESP-ID following video instruction.

lgLindstrom, 2021-04-14:
Hi

I am following this instruction  https://www.youtube.com/watch?v=Lc6ausiKvQM for installing ESP-IDF on VsCode. Despite that the video is older than the latest version of the extension everything goes well until I try to compile the blink example. 
I get the following error:

What to do,, what step(s) are missing?

> Executing task: cmake -G Ninja .. <

```
bash: /d: Is a directory
The terminal process "C:\Program Files\Git\git-cmd.exe --command=usr/bin/bash.exe -l -i /d /c cmake -G Ninja .." terminated with exit code: 126.

Terminal will be reused by tasks, press any key to close it.
```

My configuration based on ESP Doctor:

    `---------------------------------------------- ESP-IDF Extension for Visual Studio Code report --------------------------------------------- 
    OS win32 x64 10.0.19042 
    System environment variable PATH 
    C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Python39\Scripts\;C:\Python39\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files (x86)\Pico Technology\PicoScope6\;C:\Program Files\nodejs\;C:\ProgramData\chocolatey\bin;C:\Program Files\dotnet\;C:\ProgramData\flutter\bin\;C:\ProgramData\flutter\bin\dart;C:\Program Files (x86)\Bitvise SSH Client;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Git\cmd;C:\Program Files\Docker\Docker\resources\bin;C:\ProgramData\DockerDesktop\version-bin;C:\Users\lglin\esp\esp-idf;C:\Users\lglin\.espressif;C:\Users\lglin\AppData\Local\Microsoft\WindowsApps;C:\Users\lglin\AppData\Roaming\npm;C:\Users\lglin\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\lglin\AppData\Local\Programs\Microsoft VS Code Insiders\bin 
    Visual Studio Code version 1.0.2 
    Visual Studio Code language en 
    Visual Studio Code shell C:\Program Files\Git\git-cmd.exe 
    ESP-IDF Extension version 1.55.2 
    ---------------------------------------------------- Extension configuration settings ------------------------------------------------------
    ESP-IDF Path (idf.espIdfPath) C:\Users\lglin\esp\esp-idf
    Custom extra paths (idf.customExtraPaths) C:\Users\lglin\.espressif\tools\xtensa-esp32-elf\1.22.0-97-gc752ad5d-5.2.0\xtensa-esp32-elf\bin;C:\Users\lglin\.espressif\tools\esp32ulp-elf\2.28.51.20170517\esp32ulp-elf-binutils\bin;C:\Users\lglin\.espressif\tools\cmake\3.13.4\bin;C:\Users\lglin\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin;C:\Users\lglin\.espressif\tools\mconf\v4.6.0.0-idf-20190628;C:\Users\lglin\.espressif\tools\ninja\1.9.0;C:\Users\lglin\.espressif\tools\idf-exe\1.0.1;C:\Users\lglin\.espressif\tools\ccache\3.7
    Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"C:\\Users\\lglin\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20200709/openocd-esp32/share/openocd/scripts"}
    Virtual env Python Path (idf.pythonBinPath) C:\Users\lglin\.espressif\python_env\idf3.3_py3.9_env\Scripts\python.exe
    Serial port (idf.port) /dev/ttyUSB1
    OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
    ESP-IDF Tools Path (idf.toolsPath) ${env:IDF_TOOLS_PATH}
    -------------------------------------------------------- Configurations access -------------------------------------------------------------
    Access to ESP-IDF Path (idf.espIdfPath) true
    Access to ESP-IDF Custom extra paths
    Access to C:\Users\lglin\.espressif\tools\xtensa-esp32-elf\1.22.0-97-gc752ad5d-5.2.0\xtensa-esp32-elf\bin: true
    Access to C:\Users\lglin\.espressif\tools\esp32ulp-elf\2.28.51.20170517\esp32ulp-elf-binutils\bin: true
    Access to C:\Users\lglin\.espressif\tools\cmake\3.13.4\bin: true
    Access to C:\Users\lglin\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin: true
    Access to C:\Users\lglin\.espressif\tools\mconf\v4.6.0.0-idf-20190628: true
    Access to C:\Users\lglin\.espressif\tools\ninja\1.9.0: true
    Access to C:\Users\lglin\.espressif\tools\idf-exe\1.0.1: true
    Access to C:\Users\lglin\.espressif\tools\ccache\3.7: true
    Access to Virtual env Python Path (idf.pythonBinPath) true
    Access to CMake in environment PATH undefined
    Access to Ninja in environment PATH undefined
    Access to ESP-IDF Tools Path (idf.toolsPath) false
    ----------------------------------------------------------- Executables Versions -----------------------------------------------------------
    Git version 2.30.1.windows.1
    ESP-IDF version 3.3.5
    Python version 3.9.0
    Python's pip version 20.2.3
    -------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
    certifi version: 2020.12.5
    cffi version: 1.14.5
    chardet version: 4.0.0
    click version: 7.1.2
    cryptography version: 3.4.7
    future version: 0.18.2
    gcovr version: 4.2
    idna version: 2.10
    Jinja2 version: 2.11.3
    lxml version: 4.6.3
    MarkupSafe version: 1.1.1
    pip version: 20.2.3
    psutil version: 5.8.0
    pycparser version: 2.20
    pygdbmi version: 0.9.0.2
[... 99 lines cut ...]
{"message":"Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n    at c:\\Users\\lglin\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.2\\dist\\extension.js:1:8595\n    at ChildProcess.exithandler (child_process.js:303:7)\n    at ChildProcess.emit (events.js:315:20)\n    at ChildProcess.EventEmitter.emit (domain.js:483:12)\n    at maybeClose (internal/child_process.js:1021:16)\n    at Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)","level":"error","timestamp":"2021-04-14T07:42:53.304Z"}
{"message":"Command failed: reg query HKEY_CURRENT_USER\\SOFTWARE\\PYTHON\nERROR: The system was unable to find the specified registry key or value.\r\n","stack":"Error: Command failed: reg query HKEY_CURRENT_USER\\SOFTWARE\\PYTHON\nERROR: The system was unable to find the specified registry key or value.\r\n\n    at ChildProcess.exithandler (child_process.js:312:12)\n    at ChildProcess.emit (events.js:315:20)\n    at ChildProcess.EventEmitter.emit (domain.js:483:12)\n    at maybeClose (internal/child_process.js:1021:16)\n    at Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)","level":"error","timestamp":"2021-04-14T07:44:41.856Z"}
{"message":"Error looking for python in windows","stack":"Error: Command failed: reg query HKEY_CURRENT_USER\\SOFTWARE\\PYTHON\nERROR: The system was unable to find the specified registry key or value.\r\n\n    at ChildProcess.exithandler (child_process.js:312:12)\n    at ChildProcess.emit (events.js:315:20)\n    at ChildProcess.EventEmitter.emit (domain.js:483:12)\n    at maybeClose (internal/child_process.js:1021:16)\n    at Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)","level":"error","timestamp":"2021-04-14T07:44:41.856Z"}
{"message":"Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n    at c:\\Users\\lglin\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.2\\dist\\extension.js:1:8595\n    at ChildProcess.exithandler (child_process.js:303:7)\n    at ChildProcess.emit (events.js:315:20)\n    at ChildProcess.EventEmitter.emit (domain.js:483:12)\n    at maybeClose (internal/child_process.js:1021:16)\n    at Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)","level":"error","timestamp":"2021-04-14T07:44:42.463Z"}
{"message":"ENOENT: no such file or directory, open 'C:\\Users\\lglin\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20200709\\openocd-esp32\\share\\openocd\\scripts\\esp-config.json'","stack":"Error: ENOENT: no such file or directory, open 'C:\\Users\\lglin\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20200709\\openocd-esp32\\share\\openocd\\scripts\\esp-config.json'","level":"error","timestamp": [... cut ...] \Users\\lglin\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20200709\\openocd-esp32\\bin;C:\\Users\\lglin\\.espressif\\tools\\mconf\\v4.6.0.0-idf-20190628;C:\\Users\\lglin\\.espressif\\tools\\ninja\\1.9.0;C:\\Users\\lglin\\.espressif\\tools\\idf-exe\\1.0.1;C:\\Users\\lglin\\.espressif\\tools\\ccache\\3.7",
    "idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"C:\\\\Users\\\\lglin\\\\.espressif\\\\tools\\\\openocd-esp32\\\\v0.10.0-esp32-20200709/openocd-esp32/share/openocd/scripts\"}",
    
}
```

brianignacio5, 2021-04-14:
Can you try to change the shell in visual studio code from `C:\Program Files\Git\git-cmd.exe` to `powershell.exe` with
`terminal.integrated.shell.windows` in your Visual Studio Code Settings or use the `Terminal: Select Default Shell` command.

lgLindstrom, 2021-04-14:
Hi and thanks.
I simply removed that setting and vscode defaulted back to powershell. 

Now it works although I have some concerns. My project contains bash scripts that needs, to be run in a bash shell.
How do i run them now?

brianignacio5, 2021-04-15:
Well I believe most Windows users that need a bash end up using WSL with some distribution like Ubuntu. We do have support for WSL environment in the extension and Visual Studio Code provides an extension to interact with WSL.

See more [here](https://code.visualstudio.com/docs/remote/wsl).

lgLindstrom, 2021-04-15:
Okay,, so I could close this issue then..

closed 2021-04-15 by lgLindstrom