espressif/vscode-esp-idf-extension #375: [Bug Report]: New Project command requires serial port

lgLindstrom, 2021-04-22:
I have not yet managed to install a serial port on my new Laptop. Yet I tried to create a new project assuming that I just would like to write some code.

The command "New Project" gave me the following error ( and stopped )

No serial ports found

I guess this will be solved if I manage to install a serial port but I still think it should be possible to create a project.

I am running winndows 10 on a Laptop with only USB-C connectors. I have a USB-C to Micro-USB cable connected to a DevkitC board.

All my efforts to install FTDI drives have failed miserably. I have tried all install variants from the instructions from Silicon Labs.

Can anyone with a similar Laptop (no com ports) give me some directions?

brianignacio5, 2021-04-23:
Hey @lgLindstrom Sorry for the issue we will make a fix soon !

With regard to your drivers issue, maybe you can try our new [idf-env](https://github.com/espressif/idf-env) to handle drivers install.

closed 2021-04-23 by brianignacio5

sssphil, 2021-12-08:
Hi @brianignacio5, I had the same error and tried `ESP-IDF: Show Examples Projects`, but the error popped up again. Here are the last two errors in my log. I have also tried the idf-env installer, and when I tried to install it again, it said the driver is already installed. 

> {"user":true,"message":"Something went wrong while getting the serial port list","stack":"Error: No serial ports found\n\tat d.<anonymous> (c:\\Users\\s\\.vscode\\extensions\\espressif.esp-idf-extension-1.2.0\\dist\\extension.js:1:280355)\n\tat Generator.next (<anonymous>)\n\tat s (c:\\Users\\s\\.vscode\\extensions\\espressif.esp-idf-extension-1.2.0\\dist\\extension.js:1:278702)\n\tat processTicksAndRejections (internal/process/task_queues.js:93:5)","level":"error","timestamp":"2021-12-08T04:59:38.410Z"}
{"user":true,"message":"Something went wrong while getting the serial port list","stack":"Error: No serial ports found\n\tat d.<anonymous> (c:\\Users\\s\\.vscode\\extensions\\espressif.esp-idf-extension-1.2.0\\dist\\extension.js:1:280355)\n\tat Generator.next (<anonymous>)\n\tat s (c:\\Users\\s\\.vscode\\extensions\\espressif.esp-idf-extension-1.2.0\\dist\\extension.js:1:278702)\n\tat processTicksAndRejections (internal/process/task_queues.js:93:5)","level":"error","timestamp":"2021-12-08T04:59:39.476Z"}

brianignacio5, 2021-12-09:
This should be fixed with #378 

Could you please try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/119078917.zip) to see if it is fixed ?

PR #378 (2021-04-23): Fix/release ci actions
Fix #375  to allow New Project Wizard when no serial ports are found. Add esp32s3 and esp32c3 in default targets.

Package VSIX without dependencies before regular VSIX package and specify VSIX for marketplace release.