espressif/vscode-esp-idf-extension #379: [Bug Report]: MacOS: Every time I build I get an "Something went wrong while trying to build the project, Source: Espressif IDF (Extension) error

bpietsch, 2021-04-25:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [ √ ] Make sure you have searched for existing bugs and features request before you post an issue.
- [ √ ] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [ √ ] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
Every single time I build my project I get an error notification from VSCode with the text "Something went wrong while trying to build the project, Source: Espressif IDE (Extension)". This happens even when the build is successful. I have to manually dismiss this alter each time as they just continue to pile up on my screen.

**To Reproduce**
Just build a project using vs-code on Mac OS (version: 11.2.1 (20D74))

**Expected behavior**
Not to see this alert (which I believe first started occurring in plugin version 1.0.1 and is still occurring in 1.0.2)

**Screenshots**

[image]

**Environment (please complete the following information):**

[Doctor Output.txt](https://github.com/espressif/vscode-esp-idf-extension/files/6370991/Doctor.Output.txt)

**Logs**

[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/6370988/esp_idf_vsc_ext.log)

**Extension Configuration settings**

	"idf.espIdfPath": "${env:IDF_PATH}",

	"idf.pythonBinPath": "python",

	"idf.openOcdConfigs": [
		"interface/ftdi/esp32_devkitj_v1.cfg",
		"board/esp32-wrover.cfg"
	],

	"idf.customExtraPaths": "",

	"idf.customExtraVars": "",

	"idf.adapterTargetName": "esp32",

	"idf.customAdapterTargetName": "",

brianignacio5, 2021-04-25:
Could we take a look at the build output ? There is some error still happening even though build is successful. Exit code 1 ?

Could you give a small example project to reproduce this behavior ?

bpietsch, 2021-04-27:
Hi Brian, thank you for your prompt response. In double checking, it turns out I do not see this in the case of a successful build (I'm in a very experimental stage of development and most of my builds tend to fail at the moment). That said, this dialog appearing in the build failure case is still a recent change and is exceptionally annoying as I have to manually dismiss it and multiples of them will pile up on screen if I don't dismiss them after each build. I don't feel the dialogs add any value (and the wording clearly implies there's a problem with the extension, not with the build itself). Thank you for your time.

brianignacio5, 2021-04-28:
You can disable notifications by enabling the `idf.notificationSilentMode` configuration setting as described in [Extension behavior settings](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/SETTINGS.md#extension-behaviour-settings)

brianignacio5, 2021-04-30:
Closing this issue due to lack of bug and user can disable notifications with extension configuration setting.

Please comment here if you need more help with this.

closed 2021-04-30 by brianignacio5

bpietsch, 2021-05-11:
Thank you for your help, Brian. This is just a hobby project for me, so haven't had much time to follow up. The error dialog does *not* appear on a successful build, but still appears on build failure, despite the extension setting you suggested. It seems this may be expected behavior, but it's rather annoying to have to manually dismiss that error dialog on each build failure. Please reconsider this feature for a future release, or at least add an additional setting to suppress this dialog even on build failures. Thank you for your time!

brianignacio5, 2021-12-02:
Added error notifications silent mode in #582 , you could try with latest vsix in GitHub master or wait for the next release of the extension in the marketplace (after v1.2.0)