espressif/vscode-esp-idf-extension #380: [Bug Report]: The "path" argument must be of type string. Received undefined

thedayu, 2021-04-25:
**Describe the bug**
menuconfig from vscode, report "path" argument must be type of string error.

**To Reproduce**
Install vscode extention, 
Download v4.3 esp-idf
Config extention to use v4.3, EXPRESS or Advanced.
Open a example project, click menuconfig.

-----------
This happened since I update my idf from v3.2 to v4.3.

idf.py works fine.
actually I'm using vscode as Editor only, using command line idf.py to build, flash.

**Expected behavior**
Everything works.

**Screenshots**
[image]

**Environment (please complete the following information):**

- OS Name & Version: [e.g. MacOS X 10.14.5]
- VSCode Version: [1.0.2]
- ESP-IDF Version: [V4.3]
- Python Version: [3.8]
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19041 
System environment variable PATH 
 C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\Git\cmd;D:\Go\bin;C:\Program Files\Docker\Docker\resources\bin;C:\ProgramData\DockerDesktop\version-bin;C:\Users\admin\.cargo\bin;C:\Users\admin\AppData\Local\Programs\Python\Python38\Scripts\;C:\Users\admin\AppData\Local\Programs\Python\Python38\;C:\Users\admin\AppData\Local\Microsoft\WindowsApps;C:\Users\admin\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\admin\go\bin;D:\Go\bin; 
Visual Studio Code version 1.0.2 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\cmd.exe 
ESP-IDF Extension version 1.55.2 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-IDF Path (idf.espIdfPath) ${env:IDF_PATH}
Custom extra paths (idf.customExtraPaths) C:\Users\admin\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin;C:\Users\admin\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin;C:\Users\admin\.espressif\tools\xtensa-esp32s3-elf\esp-2020r3-8.4.0\xtensa-esp32s3-elf\bin;C:\Users\admin\.espressif\tools\riscv32-esp-elf\1.24.0.123_64eb9ff-8.4.0\riscv32-esp-elf\bin;C:\Users\admin\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\admin\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\admin\.espressif\tools\cmake\3.16.4\bin;C:\Users\admin\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin;C:\Users\admin\.espressif\tools\ninja\1.10.2;C:\Users\admin\.espressif\tools\idf-exe\1.0.1;C:\Users\admin\.espressif\tools\ccache\3.7;C:\Users\admin\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64;C:\Users\admin\.espressif\tools\idf-python-wheels\idf4.3_py3.9_2021-01-07
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"C:\\Users\\admin\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20200709/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) C:\Users\admin\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe
Serial port (idf.port) /dev/ttyUSB1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) C:\Users\admin\.espressif
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-IDF Path (idf.espIdfPath) false
Access to ESP-IDF Custom extra paths
Access to C:\Users\admin\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin: true
Access to C:\Users\admin\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin: true
Access to C:\Users\admin\.espressif\tools\xtensa-esp32s3-elf\esp-2020r3-8.4.0\xtensa-esp32s3-elf\bin: true
Access to C:\Users\admin\.espressif\tools\riscv32-esp-elf\1.24.0.123_64eb9ff-8.4.0\riscv32-esp-elf\bin: true
Access to C:\Users\admin\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: true
Access to C:\Users\admin\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: true
Access to C:\Users\admin\.espressif\tools\cmake\3.16.4\bin: true
Access to C:\Users\admin\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin: true
Access to C:\Users\admin\.espressif\tools\ninja\1.10.2: true
Access to C:\Users\admin\.espressif\tools\idf-exe\1.0.1: true
Access to C:\Users\admin\.espressif\tools\ccache\3.7: true
Access to C:\Users\admin\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to C:\Users\admin\.espressif\tools\idf-python-wheels\idf4.3_py3.9_2021-01-07: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.26.0.windows.1
ESP-IDF version undefined
Python version undefined
Python's pip version undefined
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
---------------------------------------------------- Check ESP-IDF python requirements.txt -------------------------------------------------
Check ESP-IDF Python packages undefined
---------------------------------------------------- Check extension requirements.txt ------------------------------------------------------
Check Extension Python packages undefined
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages undefined
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at spawn C:\WINDOWS\system32\cmd.exe ENOENT
--------------------------------------------------------------------------------------------------------------------------------------------

brianignacio5, 2021-04-25:
Basically your ESP-IDF path is wrong. Current value is `${env:IDF_PATH}` which will try to resolve from environment variable IDF_PATH. It is probable that your project has a `.vscode/settings.json` that is overriding your extension setup, because you seems to have `idf.customExtraPaths` correctly.

Is probable why you can't use the extension build flash and terminal commands.

brianignacio5, 2021-04-30:
Closing this issue for lack of follow up from user and the issue seems to be found. 

Please comment here to re-open if this doesn't fix your setup.

closed 2021-04-30 by brianignacio5