espressif/vscode-esp-idf-extension #381: [Bug Report]: close previous uart when flash or monitor

thedayu, 2021-04-25:
**Describe the bug**
when click monitor, terminal will open holding uart port.
after that, if you rebuild proejct, and click flash, the terminal which holds uart port is reused, and the flash command fails.
at this point, only thing to do is re-plug the usb cable, which is troublesome.
(I'm using esp32-devkit). 

**Environment (please complete the following information):**
- OS Name & Version: [win10]
- VSCode Version: [1.0.2]
- ESP-IDF Version: [v3.2]
- Python Version: [3.8]

brianignacio5, 2021-04-25:
Have you experimented with `idf.killMonitorTimeout` or  in Settings (UI) T_imeout (milliseconds) for ESP-IDF monitor process kill_ to configure the time to wait for monitor kill process.

brianignacio5, 2021-04-25:
You can also try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/56186808.zip)

brianignacio5, 2021-04-29:
This should be fixed with #382 . If you any additional issue please comment here to re-open.

closed 2021-04-29 by brianignacio5