espressif/vscode-esp-idf-extension #383: [Bug Report]: Cannot load examples

tomascharad, 2021-04-27:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [ X ] Make sure you have searched for existing bugs and features request before you post an issue.
- [ X ] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [ X ] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
Cannot open examples with the ESP-IDF: Show Example Projects command

**To Reproduce**
Install in VS Code, fresh start

**Expected behavior**
The examples page to open

**Screenshots**
[image]

[image]

**Environment (please complete the following information):**

- OS Name & Version: [MacOS Catalina 10.15.7]
- VSCode Version: [1.55.2]
- ESP-IDF Version: [1.0.2]
- Python Version: [3.9.4]

**Logs**
Everything ok with doctor command:

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS darwin x64 19.6.0 
System environment variable PATH 
 /Users/tomascharad/.rvm/gems/ruby-2.3.1/bin:/Users/tomascharad/.rvm/gems/ruby-2.3.1@global/bin:/Users/tomascharad/.rvm/rubies/ruby-2.3.1/bin:/Users/tomascharad/google-cloud-sdk/bin:/usr/local/opt/nvm/versions/node/v14.16.1/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/usr/local/share/dotnet:/opt/X11/bin:~/.dotnet/tools:/Library/Apple/usr/bin:/Library/Frameworks/Mono.framework/Versions/Current/Commands:/Applications/Wireshark.app/Contents/MacOS:/usr/local/git/bin:/Users/tomascharad/.rvm/bin 
Visual Studio Code version 1.0.2 
Visual Studio Code language en 
Visual Studio Code shell /bin/zsh 
ESP-IDF Extension version 1.55.2 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-IDF Path (idf.espIdfPath) /Users/tomascharad/esp/esp-idf
Custom extra paths (idf.customExtraPaths) /Users/tomascharad/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin:/Users/tomascharad/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin:/Users/tomascharad/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/Users/tomascharad/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/Users/tomascharad/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"/Users/tomascharad/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/share/openocd/scripts"}
Virtual env Python Path (idf.pythonBinPath) /Users/tomascharad/.espressif/python_env/idf4.2_py3.9_env/bin/python
Serial port (idf.port) /dev/ttyUSB1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) ${env:IDF_TOOLS_PATH}
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-IDF Custom extra paths
Access to /Users/tomascharad/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin: true
Access to /Users/tomascharad/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin: true
Access to /Users/tomascharad/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin: true
Access to /Users/tomascharad/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin: true
Access to /Users/tomascharad/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) false
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.10.0
ESP-IDF version 4.2
Python version 3.9.4
Python's pip version 21.0.1
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
bidict version: 0.21.2
bitstring version: 3.1.7
Brotli version: 1.0.9
certifi version: 2020.12.5
cffi version: 1.14.5
chardet version: 4.0.0
click version: 7.1.2
cryptography version: 3.4.7
ecdsa version: 0.16.1
Flask version: 0.12.5
Flask-Compress version: 1.9.0
Flask-SocketIO version: 2.9.6
future version: 0.18.2
gcovr version: 4.2
gdbgui version: 0.13.2.0
gevent version: 1.5.0
greenlet version: 1.0.0
idna version: 2.10
itsdangerous version: 1.1.0
Jinja2 version: 2.11.3
lxml version: 4.6.3
MarkupSafe version: 1.1.1
pip version: 21.0.1
psutil version: 5.8.0
pycparser version: 2.20
pyelftools version: 0.27
pygdbmi version: 0.9.0.2
Pygments version: 2.8.1
pyparsing version: 2.3.1
pyserial version: 3.5
python-engineio version: 4.1.0
python-socketio version: 5.2.1
reedsolo version: 1.5.4
requests version: 2.25.1
setuptools version: 54.2.0
six version: 1.15.0
urllib3 version: 1.26.4
websocket-client version: 0.58.0
Werkzeug version: 0.16.1
wheel version: 0.36.2
xmlrunner version: 1.7.7
---------------------------------------------------- Check ESP-IDF python requirements.txt -------------------------------------------------
Check ESP-IDF Python packages Python requirements from /Users/tomascharad/esp/esp-idf/requirements.txt are satisfied.
---------------------------------------------------- Check extension requirements.txt ------------------------------------------------------
Check Extension Python packages Python requirements from /Users/tomascharad/.vscode/extensions/espressif.esp-idf-extension-1.0.2/requirements.txt are satisfied.
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages Python requirements from /Users/tomascharad/.vscode/extensions/espressif.esp-idf-extension-1.0.2/esp_debug_adapter/requirements.txt are satisfied.
---------------------------------------------------- Visual Studio Code launch.json --------------------------------------------------------
{
  "configurations": [
  {
    "name": "(lldb) Attach",
    "type": "cppdbg",
    "request": "attach",
    "program": "enter program name, for example ${workspaceFolder}/a.out",
    "processId": "${command:pickProcess}",
    "MIMode": "lldb"
  }
  ]
} 
--------------------------------------------------------------------------------------------------------------------------------------------

- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

It doesn't write logs when triggering the command

**Additional context**
I'm a beginner on ESP32

brianignacio5, 2021-04-27:
When you click the ` ESP-IDF: Show Example Projects`, do you see at least one option to select the framework ?

[image]

tomascharad, 2021-04-27:
Hi @brianignacio5 ,

I don't see anything.

[image]

tomascharad, 2021-04-27:
Yeap, I get it now, many thanks.

closed 2021-04-27 by tomascharad

bhavikbhansali, 2021-05-10:
Environment (please complete the following information):

OS Name & Version: [Win10]
VSCode Version: [1.56.0]
ESP-IDF Version: [1.0.3]
Python Version: [3.9.5]

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.18363 
System environment variable PATH 
 C:\Program Files\Common Files\Siemens\Automation\Simatic OAM\bin;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\170\Tools\Binn\;C:\Program Files\dotnet\;C:\Program Files (x86)\STMicroelectronics\STM32 ST-LINK Utility\ST-LINK Utility;C:\Program Files\Git\cmd;E:\esp-idf\tools;C:\Program Files (x86)\dotnet\;C:\Users\bhavi\AppData\Local\Programs\Python\Python39\Scripts\;C:\Users\bhavi\AppData\Local\Programs\Python\Python39\;C:\Users\bhavi\AppData\Local\Microsoft\WindowsApps;C:\Users\bhavi\AppData\Local\GitHubDesktop\bin;E:\Microsoft VS Code\bin 
Visual Studio Code version 1.56.0 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.0.3 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-IDF Path (idf.espIdfPath) e:\esp-idf
Custom extra paths (idf.customExtraPaths) C:\Users\bhavi\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin;C:\Users\bhavi\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin;C:\Users\bhavi\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\bhavi\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\bhavi\.espressif\tools\cmake\3.16.4\bin;C:\Users\bhavi\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin;C:\Users\bhavi\.espressif\tools\ninja\1.10.0;C:\Users\bhavi\.espressif\tools\idf-exe\1.0.1;C:\Users\bhavi\.espressif\tools\ccache\3.7;C:\Users\bhavi\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"C:\\Users\\bhavi\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20200709/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) C:\Users\bhavi\.espressif\python_env\idf4.2_py3.9_env\Scripts\python.exe
Serial port (idf.port) /dev/ttyUSB1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) C:\Users\bhavi\.espressif
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-IDF Custom extra paths
Access to C:\Users\bhavi\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin: true
Access to C:\Users\bhavi\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin: true
Access to C:\Users\bhavi\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: true
Access to C:\Users\bhavi\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: true
Access to C:\Users\bhavi\.espressif\tools\cmake\3.16.4\bin: true
Access to C:\Users\bhavi\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin: true
Access to C:\Users\bhavi\.espressif\tools\ninja\1.10.0: true
Access to C:\Users\bhavi\.espressif\tools\idf-exe\1.0.1: true
Access to C:\Users\bhavi\.espressif\tools\ccache\3.7: true
Access to C:\Users\bhavi\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.21.0.windows.1
ESP-IDF version 4.2.1
Python version 3.9.5
Python's pip version 21.1.1
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
bidict version: 0.21.2
bitstring version: 3.1.7
Brotli version: 1.0.9
certifi version: 2020.12.5
cffi version: 1.14.5
chardet version: 4.0.0
click version: 7.1.2
contextlib2 version: 0.6.0.post1
cryptography version: 3.4.7
ecdsa version: 0.16.1
esp-windows-curses version: 0.1
Flask version: 0.12.5
Flask-Compress version: 1.9.0
Flask-SocketIO version: 2.9.6
future version: 0.18.2
gcovr version: 4.2
gdbgui version: 0.13.2.0
gevent version: 1.5.0
greenlet version: 1.1.0
idf-component-manager version: 0.2.1a1
idna version: 2.10
itsdangerous version: 1.1.0
Jinja2 version: 2.11.3
lxml version: 4.6.3
[... 13 lines cut ...]
reedsolo version: 1.5.4
requests version: 2.25.1
requests-toolbelt version: 0.9.1
schema version: 0.7.4
semantic-version version: 2.8.5
setuptools version: 56.0.0
six version: 1.16.0
tqdm version: 4.60.0
urllib3 version: 1.26.4
websocket-client version: 0.59.0
Werkzeug version: 0.16.1
wheel version: 0.36.2
windows-curses version: 2.2.0
xmlrunner version: 1.7.7
[... cut ...]
Check Debug AdapterPython packages Python requirements from c:\Users\bhavi\.vscode\extensions\espressif.esp-idf-extension-1.0.3\esp_debug_adapter\requirements.txt are satisfied.
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at ENOENT: no such file or directory, open 'd:\ElectroSoul Own Github\ESP_IDF_Example\blink\.vscode\launch.json'
--------------------------------------------------------------------------------------------------------------------------------------------

As per the previous image, I am not getting anything like that.

bhavikbhansali, 2021-05-10:
@brianignacio5 No Nothing, I tried to install many time after deleting the python and everything. But no change. Any idea or way to solve it. ESP-IDF works with windows commands which were installed at a different location but, not working with vs code.

brianignacio5, 2021-05-11:
Can you see if there is any error in the logs:

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

Also, based on your doctor command output you are using a path with spaces, can. you try opening a project in a different folder without spaces ?

giomate, 2021-07-28:
Hi brianignacio5,
i have the same problem ..

my vsc log had a lot of error.  I guess there is something wrong:

{"message":"Command failed: C:\\Users\\giova\\.espressif\\tools\\idf-git\\2.30.1\\cmd\\git.exe --version\nThe system cannot find the path specified.\r\n","stack":"Error: Command failed: C:\\Users\\giova\\.espressif\\tools\\idf-git\\2.30.1\\cmd\\git.exe --version\nThe system cannot find the path specified.\r\n\n    at ChildProcess.exithandler (child_process.js:317:12)\n    at ChildProcess.emit (events.js:315:20)\n    at maybeClose (internal/child_process.js:1048:16)\n    at Socket.<anonymous> (internal/child_process.js:439:11)\n    at Socket.emit (events.js:315:20)\n    at Pipe.<anonymous> (net.js:673:12)","level":"error","timestamp":"2021-07-27T04:45:32.690Z"}
{"user":true,"message":"Git is not found in current environment","stack":"Error: Command failed: C:\\Users\\giova\\.espressif\\tools\\idf-git\\2.30.1\\cmd\\git.exe --version\nThe system cannot find the path specified.\r\n\n    at ChildProcess.exithandler (child_process.js:317:12)\n    at ChildProcess.emit (events.js:315:20)\n    at maybeClose (internal/child_process.js:1048:16)\n    at Socket.<anonymous> (internal/child_process.js:439:11)\n    at Socket.emit (events.js:315:20)\n    at Pipe.<anonymous> (net.js:673:12)","level":"error","timestamp":"2021-07-27T04:45:32.691Z"}
{"user":true,"message":"Git is not found in current environment","stack":"Error: git is not found\n    at Object.<anonymous> (c:\\Users\\giova\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.1\\dist\\extension.js:1:9177)\n    at Generator.next (<anonymous>)\n    at s (c:\\Users\\giova\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.1\\dist\\extension.js:1:2687)","level":"error","timestamp":"2021-07-27T04:45:32.692Z"}
{"message":"Command failed: C:\\Users\\giova\\.espressif\\tools\\idf-git\\2.30.1\\cmd\\git.exe --version\nThe system cannot find the path specified.\r\n","stack":"Error: Command failed: C:\\Users\\giova\\.espressif\\tools\\idf-git\\2.30.1\\cmd\\git.exe --version\nThe system cannot find the path specified.\r\n\n    at ChildProcess.exithandler (child_process.js:317:12)\n    at ChildProcess.emit (events.js:315:20)\n    at maybeClose (internal/child_process.js:1048:16)\n    at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-07-27T04:45:32.722Z"}
{"user":true,"message":"Git is not found in current environment","stack":"Error: Command failed: C:\\Users\\giova\\.espressif\\tools\\idf-git\\2.30.1\\cmd\\git.exe --version\nThe system cannot find the path specified.\r\n\n    at ChildProcess.exithandler (child_process.js:317:12)\n    at ChildProcess.emit (events.js:315:20)\n    at maybeClose (internal/child_process.js:1048:16)\n    at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-07-27T04:45:32.723Z"}
{"user":true,"message":"Git is not found in current environment","stack":"Error: git is not found\n    at Object.<anonymous> (c:\\Users\\giova\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.1\\dist\\extension.js:1:9177)\n    at Generator.next (<anonymous>)\n    at s (c:\\Users\\giova\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.1\\dist\\extension.js:1:2687)\n    at processTicksAndRejections (internal/process/task_queues.js:93:5)","level":"error","timestamp":"2021-07-27T04:45:32.724Z"}
{"message":"Command failed: C:\\Users\\giova\\.espressif\\tools\\idf-git\\2.30.1\\cmd\\git.exe --version\nThe system cannot find the path specified.\r\n","stack":"Error: Command failed: C:\\Users\\giova\\.espressif\\tools\\idf-git\\2.30.1\\cmd\\git.exe --version\nThe system cannot find the path specified.\r\n\n    at ChildProcess.exithandler (child_process.js:317:12)\n    at ChildProcess.emit (events.js:315:20)\n    at maybeClose (internal/child_process.js:1048:16)\n    at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-07-27T04:45:38.507Z"}
{"user":true,"message":"Git is not found in current environment","stack":"Error: Command failed: C:\\Users\\giova\\.espressif\\tools\\idf-git\\2.30.1\\cmd\\git.exe --version\nThe system cannot find the path specified.\r\n\n    at ChildProcess.exithandler (child_process.js:317:12)\n    at ChildProcess.emit (events.js:315:20)\n    at maybeClose (internal/child_process.js:1048:16)\n    at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-07-27T04:45:38.508Z"}
{"message":"Command failed: C:\\Users\\giova\\.espressif\\tools\\idf-git\\2.30.1\\cmd\\git.exe --version\nThe system cannot find the path specified.\r\n","stack":"Error: Command failed: C:\\Users\\giova\\.espressif\\tools\\idf-git\\2.30.1\\cmd\\git.exe --version\nThe system cannot find the path specified.\r\n\n    at ChildProcess.exithandler (child_process.js:317:12)\n    at ChildProcess.emit (events.js:315:20)\n    at maybeClose (internal/child_process.js:1048:16)\n    at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-07-27T04:45:38.555Z"}
{"user":true,"message":"Git is not found in current environment","stack":"Error: Command failed: C:\\Users\\giova\\.espressif\\tools\\idf-git\\2.30.1\\cmd\\git.exe --version\nThe system cannot find the path specified.\r\n\n    at ChildProcess.exithandler (child_process.js:317:12)\n    at ChildProcess.emit (events.js:315:20)\n    at maybeClose (intern [... cut ...] 28T05:32:57.805Z"}
{"message":"ENOENT: no such file or directory, open 'c:\\Espressif\\esp-idf\\examples\\get-started\\.vscode\\launch.json'","stack":"Error: ENOENT: no such file or directory, open 'c:\\Espressif\\esp-idf\\examples\\get-started\\.vscode\\launch.json'","level":"error","timestamp":"2021-07-28T05:32:59.323Z"}
{"user":true,"level":"warn","message":"Extension configuration report has been copied to clipboard with errors","timestamp":"2021-07-28T05:32:59.323Z"}

I open  to any comment

brianignacio5, 2021-07-28:
Hi @giomate ! What is the problem you are having ?

giomate, 2021-07-29:
when i try to load examples, after a while, vscode  notifies "No framework selected"
[image]
the ESP_IDF doctor says, at the end "Latest error at ENOENT: no such file or directory, open 'c:\Espressif\esp-idf\examples\get-started\.vscode\launch.json'" as @bhavikbhansali 

from the vscode log, these are the first errors:
{"message":"HTTP/HTTPS Request error https://dl.espressif.com/dl/idf-python/idf-python-3.8.7-embed-win64.zip","stack":"Error: HTTP/HTTPS Request error https://dl.espressif.com/dl/idf-python/idf-python-3.8.7-embed-win64.zip\n    at g.<anonymous> (c:\\Users\\giova\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.1\\dist\\extension.js:1:78272)\n    at g.emit (events.js:315:20)\n    at ClientRequest.l.forEach.d.<computed> (c:\\Users\\giova\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.1\\dist\\extension.js:1:265921)\n    at ClientRequest.emit (events.js:327:22)\n    at TLSSocket.socketErrorListener (_http_client.js:469:9)\n    at TLSSocket.emit (events.js:315:20)\n    at emitErrorNT (internal/streams/destroy.js:106:8)\n    at emitErrorCloseNT (internal/streams/destroy.js:74:3)\n    at processTicksAndRejections (internal/process/task_queues.js:80:21)","level":"error","timestamp":"2021-07-27T04:48:56.087Z"}
{"message":"HTTP/HTTPS Request error https://dl.espressif.com/dl/idf-python/idf-python-3.8.7-embed-win64.zip","stack":"Error: HTTP/HTTPS Request error https://dl.espressif.com/dl/idf-python/idf-python-3.8.7-embed-win64.zip\n    at g.<anonymous> (c:\\Users\\giova\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.1\\dist\\extension.js:1:78272)\n    at g.emit (events.js:315:20)\n    at ClientRequest.l.forEach.d.<computed> (c:\\Users\\giova\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.1\\dist\\extension.js:1:265921)\n    at ClientRequest.emit (events.js:327:22)\n    at TLSSocket.socketErrorListener (_http_client.js:469:9)\n    at TLSSocket.emit (events.js:315:20)\n    at emitErrorNT (internal/streams/destroy.js:106:8)\n    at emitErrorCloseNT (internal/streams/destroy.js:74:3)\n    at processTicksAndRejections (internal/process/task_queues.js:80:21)","level":"error","timestamp":"2021-07-27T04:49:16.162Z"}
{"message":"HTTP/HTTPS Request error https://dl.espressif.com/dl/idf-python/idf-python-3.8.7-embed-win64.zip","stack":"Error: HTTP/HTTPS Request error https://dl.espressif.com/dl/idf-python/idf-python-3.8.7-embed-win64.zip\n    at g.<anonymous> (c:\\Users\\giova\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.1\\dist\\extension.js:1:78272)\n    at g.emit (events.js:315:20)\n    at ClientRequest.l.forEach.d.<computed> (c:\\Users\\giova\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.1\\dist\\extension.js:1:265921)\n    at ClientRequest.emit (events.js:327:22)\n    at TLSSocket.socketErrorListener (_http_client.js:469:9)\n    at TLSSocket.emit (events.js:315:20)\n    at emitErrorNT (internal/streams/destroy.js:106:8)\n    at emitErrorCloseNT (internal/streams/destroy.js:74:3)\n    at processTicksAndRejections (internal/process/task_queues.js:80:21)","level":"error","timestamp":"2021-07-27T04:50:26.830Z"}
{"user":true,"message":"Zip file error","stack":"Error: Zip file error\n    at c:\\Users\\giova\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.1\\dist\\extension.js:1:177757\n    at c:\\Users\\giova\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.1\\dist\\extension.js:8:185821\n    at c:\\Users\\giova\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.1\\dist\\extension.js:8:184167\n    at c:\\Users\\giova\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.1\\dist\\extension.js:8:185223\n    at c:\\Users\\giova\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.1\\dist\\extension.js:22:23834\n    at FSReqCallback.wrapper [as oncomplete] (fs.js:539:5)","level":"error","timestamp":"2021-07-27T04:50:26.855Z"}
{"user":true,"message":"Zip file error","stack":"Error: Zip file error\n    at c:\\Users\\giova\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.1\\dist\\extension.js:1:177757\n    at c:\\Users\\giova\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.1\\dist\\extension.js:8:185821\n    at c:\\Users\\giova\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.1\\dist\\extension.js:8:184167\n    at c:\\Users\\giova\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.1\\dist\\extension.js:8:185223\n    at c:\\Users\\giova\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.1\\dist\\extension.js:22:23834\n    at FSReqCallback.wrapper [as oncomplete] (fs.js:539:5)","level":"error","timestamp":"2021-07-27T04:51:37.584Z"}
{"user":true,"message":"Zip file error","stack":"Error: Zip file error\n    at c:\\Users\\giova\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.1\\dist\\extension.js:1:177757\n    at c:\\Users\\giova\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.1\\dist\\extension.js:8:185821\n    at c:\\Users\\giova\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.1\\dist\\extension.js:8:184167\n    at c:\\Users\\giova\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.1\\dist\\extension.js:8:185223\n    at c:\\Users\\giova\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.1\\dist\\extension.js:22:23834\n    at FSReqCallback.wrapper [as oncomplete] (fs.js:539:5)","level":"error","timestamp":"2021-07-27T04:51:42.529Z"}

to sum ip, the extension in not working...!

brianignacio5, 2021-07-29:
When you click the ` ESP-IDF: Show Example Projects`, do you see at least one option to select the framework  in the top center of Visual Studio Code?

[image]

giomate, 2021-07-30:
after reinstalling everything , vscode ... extension. seems to work!
now the framework is there
[image]

brianignacio5, 2021-10-19:
When you say it works incorrectly, what is the issue or error? Can you share the log :

Windows: %USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log
Linux & MacOSX: $HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log

We might have it fixed in GitHub master but not yet in marketplace version. Maybe you could test the latest vsix installer from the GitHub artifacts.

mbbender, 2021-11-05:
I'm on MacOSX getting same error. My content in the esp_idf_vsc_ext.log files is:

> {"user":true,"message":"File project_description.json cannot be found.","stack":"Error: File-Not-Found\n\tat l.getComponentsInProject (/Users/michael/.vscode/extensions/espressif.esp-idf-extension-1.2.0/dist/extension.js:8:145535)\n\tat /Users/michael/.vscode/extensions/espressif.esp-idf-extension-1.2.0/dist/extension.js:8:144762\n\tat new Promise (<anonymous>)\n\tat l.getChildren (/Users/michael/.vscode/extensions/espressif.esp-idf-extension-1.2.0/dist/extension.js:8:144719)\n\tat I.fetchChildrenNodes (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:94:63461)\n\tat I.getChildren (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:94:60204)\n\tat D.$getChildren (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:94:56424)\n\tat i._doInvokeHandler (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:98:13802)\n\tat i._invokeHandler (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:98:13486)\n\tat i._receiveRequest (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:98:12148)\n\tat i._receiveOneMessage (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:98:11025)\n\tat /Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:98:8922\n\tat a.fire (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:66:1712)\n\tat l.fire (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:74:15990)\n\tat /Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:113:30514\n\tat a.fire (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:66:1712)\n\tat l.fire (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:74:15990)\n\tat u._receiveMessage (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:74:21260)\n\tat /Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:74:18138\n\tat a.fire (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:66:1712)\n\tat w.acceptChunk (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:74:13351)\n\tat /Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:74:12699\n\tat Socket.v (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:113:13465)\n\tat Socket.emit (events.js:315:20)\n\tat Socket.EventEmitter.emit (domain.js:467:12)\n\tat addChunk (internal/streams/readable.js:309:12)\n\tat readableAddChunk (internal/streams/readable.js:284:9)\n\tat Socket.Readable.push (internal/streams/readable.js:223:10)\n\tat Pipe.onStreamRead (internal/stream_base_commons.js:188:23)","level":"error","timestamp":"2021-11-05T02:10:44.993Z"}
{"message":"CMakeLists.txt not found in project directory /Users/michael/Code/idf/youtube\n","stack":"Error: CMakeLists.txt not found in project directory /Users/michael/Code/idf/youtube\n\n\tat m.printError (/Users/michael/.vscode/extensions/espressif.esp-idf-extension-1.2.0/dist/extension.js:1:28725)\n\tat Socket.<anonymous> (/Users/michael/.vscode/extensions/espressif.esp-idf-extension-1.2.0/dist/extension.js:1:28154)\n\tat Socket.emit (events.js:315:20)\n\tat Socket.EventEmitter.emit (domain.js:467:12)\n\tat addChunk (internal/streams/readable.js:309:12)\n\tat readableAddChunk (internal/streams/readable.js:284:9)\n\tat Socket.Readable.push (internal/streams/readable.js:223:10)\n\tat Pipe.onStreamRead (internal/stream_base_commons.js:188:23)","level":"error","timestamp":"2021-11-05T02:12:12.037Z"}
{"message":"SDK Configuration editor confserver process exited with code: 2","stack":"Error: SDK Configuration editor confserver process exited with code: 2\n\tat m.printError (/Users/michael/.vscode/extensions/espressif.esp-idf-extension-1.2.0/dist/extension.js:1:28725)\n\tat ChildProcess.<anonymous> (/Users/michael/.vscode/extensions/espressif.esp-idf-extension-1.2.0/dist/extension.js:1:28331)\n\tat ChildProcess.emit (events.js:315:20)\n\tat ChildProcess.EventEmitter.emit (domain.js:467:12)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:277:12)","level":"error","timestamp":"2021-11-05T02:12:12.049Z"}
[... cut ...]  reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n\n\tat /Users/michael/.vscode/extensions/espressif.esp-idf-extension-1.2.0/dist/extension.js:1:7042\n\tat ChildProcess.exithandler (child_process.js:308:7)\n\tat ChildProcess.emit (events.js:315:20)\n\tat ChildProcess.EventEmitter.emit (domain.js:467:12)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-11-05T02:21:33.785Z"}

heavyStuff3000, 2022-08-27:
Lol you need to select another option but a 'loading examples windows' with a turning wheel pops up in your face... what are you supposed to think. talk about user inyerface

fa-mas, 2022-10-08:
Hello @brianignacio5, I have a similar problem as tomascharad. When I try to `ESP-IDF: Show Example Projects` I cannot select a framework because I get no option.
[image]
After a while the `ESP-IDF: Loading Examples` just stops and nothing else happens.
I've reinstalled both vsc and esp-idf but keep having this issue.

ESP-IDF Doctor reports following Errors:
```
---------------------------------------------------- Check ESP-IDF python requirements.txt -------------------------------------------------
Check ESP-IDF Python packages Error
---------------------------------------------------- Check extension requirements.txt ------------------------------------------------------
Check Extension Python packages Error
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages Error
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at ENOENT: no such file or directory, open '/home/fabio/Projects/.vscode/launch.json'
--------------------------------------------------------------------------------------------------------------------------------------------
```
As you can see I have no .vscode/launch.json file. Do I need to create one?

brianignacio5, 2022-10-26:
Hey @fa-mas could you share the [troubleshooting information](https://github.com/espressif/vscode-esp-idf-extension#troubleshooting)? Seems to be a setup issue in your doctor command output.

Maybe the log has more information.

fa-mas, 2022-10-26:
Hi Brian, thanks for your reply. The Problem was resolved, sorry for not updating.

PCS137, 2025-03-07:
Why did the issue close? where is the fix? how to fix?

[image]

brianignacio5, 2025-03-07:
Hi @PCS137  I'm not sure what issue are you facing from your screenshot.

Since this issue is probably old maybe is not related to your current issue.

There is a related bug fix in PR #1451 which might be related (based on the no esp-idf frameworks found message in your screenshot)

You can try latest master [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/workflows/ci/master/esp-idf-extension.vsix.zip) by pressing F1 or click menu `View` -> `Command Palette...`, type `Install from VSIX` and then select the downloaded .vsix file to install the extension.

PCS137, 2025-03-07:
Okay Let me see that and try