espressif/vscode-esp-idf-extension #385: Installation failed at setup python env

michael999999, 2021-04-27:
Could not fetch URL https://pypi.org/simple/wheel/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/wheel/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.")) - skipping
Could not fetch URL https://pypi.org/simple/pip/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/pip/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.")) - skipping

brianignacio5, 2021-04-28:
This is not an issue with the extension but with the python setup itself.

Maybe try a solution from https://stackoverflow.com/questions/41328451/ssl-module-in-python-is-not-available-when-installing-package-with-pip3

brianignacio5, 2021-04-30:
Closing this issue since it is not related to the extension. Please let us know if we can help you fix your setup somehow.

closed 2021-04-30 by brianignacio5

ma6254, 2023-01-06:
How can an offline environment need to be connected to the Internet? 
Can you apply vscode http_proxy parameter before executing install.ps1?

brianignacio5, 2023-01-09:
To install python packages to the virtual environment you need to be connect to the internet. The install script (install.ps1) will use offline tools if found, but python packages would most likely be download from the PyPi repository or Espressif python package registry.