espressif/vscode-esp-idf-extension #386: [Bug Report]: Python not found when installing esp-idf vscode extension

krupis, 2021-04-27:
Hello. I am having a lot of issues while installing esp-idf extension for VS code.

I have downloaded Python from the link below:
https://www.python.org/downloads/

And select a default installation path, which in my case is:
C:\Users\Lukas\AppData\Local\Programs\Python\Python39

I then followed the esp-idf extension installation procedure for the vscode. I have selected epxress instalation with the following parameters:
[image]

After the installation of the extension, it returns me an error message:
```

Installing ESP-IDF extension python packages in C:\Users\Lukas\.espressif\python_env\idf4.2_py3.9_env\Scripts\python.exe ...

No Python at 'C:\Users\Lukas\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\python.exe'
Command failed: "C:\Users\Lukas\.espressif\python_env\idf4.2_py3.9_env\Scripts\python.exe" -m pip install --no-warn-script-location  -r "c:\Users\Lukas\.vscode\extensions\espressif.esp-idf-extension-1.0.3\requirements.txt"
No Python at 'C:\Users\Lukas\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\python.exe'
Command failed: "C:\Users\Lukas\.espressif\python_env\idf4.2_py3.9_env\Scripts\python.exe" -m pip install --no-warn-script-location  -r "c:\Users\Lukas\.vscode\extensions\espressif.esp-idf-extension-1.0.3\requirements.txt"
No Python at 'C:\Users\Lukas\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\python.exe'

```

It seems like it is looking for at a completely different location to where it is supposed to look:
'C:\Users\Lukas\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python
I do not understand how did visual studio got this location from. 

From what I know, this python destination is default when the python is downloaded and installed from the windows microsoft store. Perhaps I have done it years ago but since then I have done many things ( deleted and reinstalled python multiple times and etc.. )..

Can someone suggest a ways to debug this issue? Also, can someone explain what  is the point of creating a virtual python environment? I do not really understand what does that mean. Can the extension not use the normal Python that I have installed on my pc without using virtual environment?

Things I have done:

1. Reinstall esp-idf extension
2. Reinstall python
3. Add the python path in my user variables:

[image]

4.  Added a python destination in global variables for the python:

[image]

5. Updaded global paths with the destination i have created in the preivous point (4. )
[image]

Appreciate any help solving this issue

BenDinsdale, 2021-04-28:
I'm having the same issue, I did manage to get it to progress a little further through by running the update pip command.
win key + R, C:\ESP-IDF\.espressif\python_env\idf4.2_py3.9_env\Scripts\python.exe -m pip install --upgrade pip' 
Then reinstalling everything again, the extension then installed successfully, but then the example projects I tried wouldn't compile.

I have had some success starting a fresh install by using V0.5.1 of the extension and V4.1.1 of the IDF, this lets you follow along with the youtube video, this did let me compile a the example projects I have tried.

brianignacio5, 2021-04-28:
As you can see from the first picture, your system is trying to use the Python from WindowsApps which appears before the Python39 directory. Either make sure there is not Python reference in the WindowsApps directory or remove this path from the environment variable.

krupis, 2021-04-28:
Hey. Thank you very much for the response. I have checked my environment variables and deleted the following path:
%USERPROFILE%\AppData\Local\Microsoft\WindowsApps

However, after reinstalling the esp-idf, still the same error persists. 

In my directory:
C:\Users\Lukas\AppData\Local\Microsoft\WindowsApps
I have python.exe and python3.exe for some reason:
[image]

I have tried deleting them, but with no luck:
[image]

I have expected that after I have removed the environment path to windowsapps the esp-idf will no longer look for python there but that is not the case for some reason

krupis, 2021-04-28:
Okay I have also deleted both python.exe files which were inside WindowApps using the command line. 

Now there is nothing related with Python in windowsapps and no directory in my paths. Unfortunately the exact same error still persists.

brianignacio5, 2021-04-29:
After removing the python from Windows Apps, did you try to delete the virtual environment and create it from the start ? A virtual environment still has references to the python system packages used to created the environment.

Also forgot to reply the question of why using a virtual environment. The thing is we require a set of python packages for ESP-IDF and this extension to work properly. This packages can have conflict with other python packages so in order to avoid conflict creating an isolated environment allows us to mitigate these issues.

krupis, 2021-04-30:
Hey. I am not 100% sure what do you mean when you say delete the virtual environment. Everytime I am reinstalling esp-idf extension I follow these steps:

1.
[image]

Uninstall esp-idf from the vscode extension manager.

2. Turn OFF VScode.

3. Delete the folder C:\Users\Lukas\esp contets. 

4. Launch the Vscode again, and proceed with fresh installation of esp-idf extension.

I assume the virtual environment gets deleted when I delete the whole contents of the directory C:\Users\Lukas\esp:
[image]
Is that right?

brianignacio5, 2021-04-30:
The virtual environment should be somewhere in `C:\Users\Lukas\.espressif\python_env\idf4.2_py3.9_env` based on your logs.

krupis, 2021-05-03:
Hello. Sorry for the delayed response, I have been away for the weekend.
I have deleted the python_env folder contents and tried to reinstall the esp-idf extension again. It did fix the issue. I am now able to sucesfully build and flash my ESP32 device without any issues. Thank you very much.

closed 2021-05-06 by brianignacio5