espressif/vscode-esp-idf-extension #387: [Bug Report]: 1.0.3 breaks my project

axelriet, 2021-04-28:
After updating the extension to 1.0.3 my project no longer builds. I got a message similar to this:

Cannot parse CMake requirements line: __component_set_property(___idf_XXXXXXX REQUIRES "cxx;newlib;freertos;heap;log;lwip;soc;esp_rom;esp_common;xtensa;esp32")

This exact issue has been reported elsewhere today (27 April 2021) 

https://esp32.com/viewtopic.php?f=25&p=75564

Reverting to 1.0.2 immediately fixed the problem.

brianignacio5, 2021-04-28:
Can try to re run the extension setup on v1.0.3 to install the `idf-component-manager` package.

oachkatzalschwoaf, 2021-04-28:
had the same issue.
`ESP-IDF: Install ESP-IDF Python Packages` solved it for me.

LegToTheLeggi, 2021-04-28:
> had the same issue.
> `ESP-IDF: Install ESP-IDF Python Packages` solved it for me.

Unfortunately this did not solve the problem for me, but downgrading to 1.0.2 does.

brianignacio5, 2021-04-28:
@LegToTheLeggi @axelriet 

I can't reproduce the issue but I will apply a hot fix to marketplace.

Can you try this [link](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/56935226.zip) and see if it solves the issue ?

closed 2021-04-29 by brianignacio5

brianignacio5, 2021-04-29:
`idf-component-manager` 0.2.1a1 has just been released to fix this issue, please run `ESP-IDF: Install ESP-IDF Python Packages` to get the latest after uninstalling the previous one.

axelriet, 2021-04-29:
> Which ESP-IDF version you use ?
> 
> Can try to re run the extension setup on v1.0.3 to install the `idf-component-manager` package or use the `ESP-IDF: Install ESP-IDF Python Packages`

Thanks for your reactivity. 

I’m using idf 4.2 and python 3.9.4. Im back on extension 1.0.2 now.

As a side note you say you used Python 3.7 but this version is no longer supported since 2020-06-27. Not sure if related but you should try on latest versions of everything. This is now new users always start. 

I saw your fix was to disable some manager but I suspect enabling it will cause the problem to surface again. I’m not sure what the component manager is supposed to do though. I’ll try 1.0.3 again when you have root-caused the issue. 

Thanks

brianignacio5, 2021-04-30:
With the `0.2.1a1` version of idf-component-manager the issue should be fixed. You can test 1.0.3 and update `idf-component-manager` python package in the virtual environment.

PR #389 (2021-04-29): enable component manager as configuration setting
Remove IDF_COMPONENT_MANAGER from build task by default and enable only with boolean configuration setting.

Fix #387 

Fix #388