espressif/vscode-esp-idf-extension #388: [Bug Report]: Something_went_wrong_while_trying_to_build_project

techaura, 2021-04-28:
**Describe the bug**
Upgrading to 1.0.3 made the component dead.
At the same time, idf.py in the project folder does its job correctly.
Rolling back to version 1.0.2 makes the component healthy again.

Windows 10 + VS-Code, ESP IDF preinstalled, Python version 3.8.7 used in system
Something went wrong while trying to build project

**Screenshots**
If applicable, add screenshots to help explain your problem.
http://i.piccy.info/i9/04ff4594c4420976645cbc55f404ccd8/1619605350/24924/1426873/Something_went_wrong_while_trying_to_build_project.png
**Environment (please complete the following information):**

- OS Name & Version: [e.g. MacOS X 10.14.5]
- VSCode Version: [e.g. 1.35.1]
- ESP-IDF Version: [e.g. v3.2.2]
- Python Version: [e.g. 2.7.10]

> **NOTE**: You can use the `ESP-IDF: Doctor command` to generate a report of your configuration.
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19042 
System environment variable PATH 
 C:\dev\Python38\Scripts\;C:\dev\Python38\;C:\Program Files\Common Files\Oracle\Java\javapath;C:\windows\system32;C:\windows;C:\windows\System32\Wbem;C:\windows\System32\WindowsPowerShell\v1.0\;C:\windows\System32\OpenSSH\;C:\dev\Git\Git\cmd;C:\dev\CMake\bin;C:\Program Files\PuTTY\;C:\dev\Python38\Scripts\;C:\dev\Python38\;C:\Users\Slava\AppData\Local\Microsoft\WindowsApps;C:\Users\Slava\AppData\Local\Programs\Microsoft VS Code\bin; 
Visual Studio Code version 1.55.2 
Visual Studio Code language en 
Visual Studio Code shell C:\windows\System32\cmd.exe 
ESP-IDF Extension version 1.0.3 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-IDF Path (idf.espIdfPath) c:\dev\espidf42
Custom extra paths (idf.customExtraPaths) C:\dev\espidftools\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin;C:\dev\espidftools\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin;C:\dev\espidftools\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\dev\espidftools\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\dev\espidftools\.espressif\tools\cmake\3.16.4\bin;C:\dev\espidftools\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin;C:\dev\espidftools\.espressif\tools\ninja\1.10.0;C:\dev\espidftools\.espressif\tools\idf-exe\1.0.1;C:\dev\espidftools\.espressif\tools\ccache\3.7;C:\dev\espidftools\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"C:\\dev\\espidftools\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20200709/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) C:\dev\espidftools\.espressif\python_env\idf4.2_py3.8_env\Scripts\python.exe
Serial port (idf.port) /dev/ttyUSB1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) ${env:IDF_TOOLS_PATH}
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-IDF Custom extra paths
Access to C:\dev\espidftools\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin: true
Access to C:\dev\espidftools\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin: true
Access to C:\dev\espidftools\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: true
Access to C:\dev\espidftools\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: true
Access to C:\dev\espidftools\.espressif\tools\cmake\3.16.4\bin: true
Access to C:\dev\espidftools\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin: true
Access to C:\dev\espidftools\.espressif\tools\ninja\1.10.0: true
Access to C:\dev\espidftools\.espressif\tools\idf-exe\1.0.1: true
Access to C:\dev\espidftools\.espressif\tools\ccache\3.7: true
Access to C:\dev\espidftools\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) false
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.31.1.windows.1
ESP-IDF version 4.2
Python version 3.8.7
Python's pip version 20.3.3
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
bidict version: 0.21.2
bitstring version: 3.1.7
Brotli version: 1.0.9
certifi version: 2020.12.5
cffi version: 1.14.4
chardet version: 4.0.0
click version: 7.1.2
cryptography version: 3.3.1
ecdsa version: 0.16.1
esp-windows-curses version: 0.1
Flask version: 0.12.5
Flask-Compress version: 1.8.0
[... 36 lines cut ...]
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages undefined
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Command failed: C:\dev\espidftools\.espressif\python_env\idf4.2_py3.8_env\Scripts\python.exe c:\dev\espidf42\tools\check_python_dependencies.py -r "c:\Users\Slava\.vscode\extensions\espressif.esp-idf-extension-1.0.3\requirements.txt"

--------------------------------------------------------------------------------------------------------------------------------------------

**Logs**

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
{"user":true,"message":"Something went wrong while trying to build the project","stack":"Error: CMake or Ninja executables not found\n\tat d.<anonymous> (c:\\Users\\Slava\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:1:110335)\n\tat Generator.next (<anonymous>)\n\tat s (c:\\Users\\Slava\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:1:109368)\n\tat processTicksAndRejections (internal/process/task_queues.js:97:5)","level":"error","timestamp":"2021-04-28T09:37:22.729Z"}
{"message":"non zero exit code 1\r\n\r\n����������: �� 㤠���� ���� 䠩�� �� ������� 蠡�����.\r\n","stack":"Error: non zero exit code 1\r\n\r\n����������: �� 㤠���� ���� 䠩�� �� ������� 蠡�����.\r\n\n\tat ChildProcess.<anonymous> (c:\\Users\\Slava\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:1:5604)\n\tat ChildProcess.emit (events.js:315:20)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:275:12)","level":"error","timestamp":"2021-04-28T09:38:09.315Z"}
{"message":"Cannot access filePath: ninja","level":"error","timestamp":"2021-04-28T09:38:09.315Z"}
{"user":true,"message":"Something went wrong while trying to build the project","stack":"Error: CMake or Ninja executables not found\n\tat d.<anonymous> (c:\\Users\\Slava\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:1:110335)\n\tat Generator.next (<anonymous>)\n\tat s (c:\\Users\\Slava\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:1:109368)\n\tat processTicksAndRejections (internal/process/task_queues.js:97:5)","level":"error","timestamp":"2021-04-28T09:38:09.316Z"}
{"message":"non zero exit code 1\r\n\r\n����������: �� 㤠���� ���� 䠩�� �� ������� 蠡�����.\r\n","stack":"Error: non zero exit code 1\r\n\r\n����������: �� 㤠���� ���� 䠩�� �� ������� 蠡�����.\r\n\n\tat ChildProcess.<anonymous> (c:\\Users\\Slava\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:1:5604)\n\tat ChildProcess.emit (events.js:315:20)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:275:12)","level":"error","timestamp":"2021-04-28T09:39:32.147Z"}
{"message":"Cannot access filePath: ninja","level":"error","timestamp":"2021-04-28T09:39:32.148Z"}

**Extension Configuration settings**
Open Command Palette (F1) and type `Preferences: Open Settings (JSON)` or use `{CURRENT_PROJECT}/.vscode/settings.json` to get the following extension configuration settings value:
settings.json
{
  "files.associations": {
    "stdio.h": "c"
  }
}

brianignacio5, 2021-04-28:
This is a duplicate of #387 

Try installing missing python packages with `ESP-IDF: Install ESP-IDF Python Packages` command.

techaura, 2021-04-28:
This does not work
http://i.piccy.info/i9/a971642c516f3ce40f62bd1bc2969764/1619607526/42758/1426873/ESP_IDF_doesnt_work.png

brianignacio5, 2021-04-28:
try deleting the build folder and try again.

techaura, 2021-04-28:
no changes after build deleted
Just fix 1.0.3 it is unusable, I will roll out to 1.0.2
 No one needs to find out in the github branch how to make a worker what does not work. It just has to correctly find the paths that are already there.

brianignacio5, 2021-04-28:
It is hard to fix an issue if we don't know the source of the problem... 

Based on the doctor command output, the last error was given in the extension python requirements.txt.

Now not sure where is the issue, I've tested on Windows 10 with IDF v4.2 and Python 3.7 with the blink example from v4.2 and it is working normally.

techaura, 2021-04-28:
The tips that I applied - killed the pre-installed ESP IDF. Now compilation only works from VSCode version 1.0.2 and does not work from terminal. Also reinstall IDF.
I can only advise everyone not to try to install version 1.0.3 in any case and stay on 1.0.2
//-------------
C:\dev\esp32workspace\get-started\blink>idf.py build
Executing action: all (aliases: build)
Running cmake in directory c:\dev\esp32workspace\get-started\blink\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=1 c:\dev\esp32workspace\get-started\blink"...
-- Found Git: C:/dev/Git/Git/bin/git.exe (found version "2.31.1.windows.1")
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/dev/espidftools/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/dev/espidftools/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/dev/espidftools/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/dev/espidftools/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/dev/espidftools/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
Cannot parse CMake requirements line: __component_set_property(___idf_Terminalengine REQUIRES "i2Cengine;Thermostat;userinit;cxx;newlib;freertos;heap;log;lwip;soc;esp_rom;esp_common;esp_system;xtensa;esp32")

CMake Error at C:/dev/espidf42/tools/cmake/component.cmake:240 (message):
Call Stack (most recent call first):
  C:/dev/espidf42/tools/cmake/build.cmake:405 (__component_get_requirements)
  C:/dev/espidf42/tools/cmake/project.cmake:395 (idf_build_process)
  CMakeLists.txt:7 (project)

-- Configuring incomplete, errors occurred!
See also "C:/dev/esp32workspace/get-started/blink/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

C:\dev\esp32workspace\get-started\blink>

brianignacio5, 2021-04-28:
I can't reproduce the issue but I will apply a hot fix to marketplace.

Can you try this [link](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/56935226.zip) and see if it solves the issue ?

brianignacio5, 2021-04-28:
I can't seem to be able to reproduce this issue on my Windows 10 machine, can you please share some steps to reproduce to check if the fix works ?

Thanks and again sorry for the inconvenience.

closed 2021-04-29 by brianignacio5

brianignacio5, 2021-04-29:
`idf-component-manager` 0.2.1a1 has just been released to fix this issue, please run `ESP-IDF: Install ESP-IDF Python Packages` to get the latest after uninstalling the previous one.

baldhead69, 2021-11-11:
Hi,

I am having the same problem.
I already run "ESP-IDF: Install ESP-IDF Python Packages" and not solve the problem.
I deleted build folder and dont solve too.

I installed via visual studio code using "ESP-IDF VS Code Extension"
idf version: v4.3.1

in esp-idf version "v4.4-dev-2359-g58022f859" the program compiled ok.

baldhead69, 2021-11-11:
How to do this ?

https://github.com/espressif/esp-idf/issues/6964#issuecomment-830153155

brianignacio5, 2021-11-11:
I understand why @baldhead69 . When you run `ESP-IDF: Install ESP-IDF Python Packages` it will install `idf-component-manager` from the extension's requirements.txt which doesn't specify a version. If you open a ESP-IDF terminal with the `ESP-IDF: Open ESP-IDF Terminal` and uninstall the existing package in your virtual environment and run `pip install idf-component-manager>=0.2.99-beta`

baldhead69, 2021-11-11:
Sorry @brianignacio5 

I dont know how to uninstall.
Could you be more specific?

by the way i will delete v4.3.1 and will install v4.4 because of esp32-s3 "full" support.

will I have problems with version 4.4 too?

Thank's.

baldhead69, 2021-11-11:
@brianignacio5 

i dont have problem with v4.4 instalation.

I just found it strange that at the end of the compilation there is no explicit message that the compilation was ok and that it is ready to flash on the chip.

baldhead69, 2021-11-11:
@brianignacio5 

I'm actually having trouble opening menuconfig.
In visual studio code does not load.

In "ESP-IDF Command Prompt (cmd.exe)" this error appears:

Cannot parse CMake requirements line: __component_set_property(___idf_.global_conf REQUIRES "cxx;newlib;freertos;esp_hw_support;heap;log;lwip;soc;hal;esp_rom;esp_common;esp_system;esp32")

baldhead69, 2021-11-11:
@brianignacio5 

If i set "environment variables" in win10, this could cause any problem ?

I am setting "environment variables".

brianignacio5, 2021-11-12:
To uninstall the python package on v4.3, just run the `ESP-IDF: Open ESP-IDF Terminal` and type `pip uninstall idf-component-manager` to get rid of the wrong version and install the new one with `pip install "idf-component-manager>=0.2.99-beta"`. 

This issue is not part of v4.4 since the correct package is listed in the esp-idf requirements.txt used to install python packages and build should work correctly. With regards to command prompt issue, after fixing the component manager version it should be fixed. (Delete the project's build folder after changing to newer version)

baldhead69, 2021-11-12:
@brianignacio5 

I deleted all v4.3.1 related stuf and started a complete installation of version v4.4 from scratch.

I'm actually having trouble opening menuconfig.
In visual studio code menuconfig does not load.

In "ESP-IDF Command Prompt (cmd.exe)" this error appears:
Cannot parse CMake requirements line: __component_set_property(__idf.global_conf REQUIRES "cxx;newlib;freertos;esp_hw_support;heap;log;lwip;soc;hal;esp_rom;esp_common;esp_system;esp32")

PR #389 (2021-04-29): enable component manager as configuration setting
Remove IDF_COMPONENT_MANAGER from build task by default and enable only with boolean configuration setting.

Fix #387 

Fix #388