espressif/vscode-esp-idf-extension #390: [Bug Report]: Issues with ESP-IDF: Select OpenOCD Board Configuration

pearson, 2021-04-29:
With the current v1.0.3 ESP-IDF, most of the configurations created by `ESP-IDF: Select OpenOCD Board Configuration` have errors.

- The `ESP-WROVER-KIT 3.3V` and `ESP-WROVER-KIT 1.8V` options generate `idf.openOcdConfigs` with incorrect directory names.
 `"boards/esp32-wrover-kit-1.8v.cfg"` should be `"board/esp32-wrover-kit-1.8v.cfg"`
- Most of the `ESP-PROG` options generate an `idf.openOcdConfigs` with incorrect config file names. 
`"interface/ftdi/esp32_devkitjv1.cfg"` should be `"interface/ftdi/esp32_devkitj_v1.cfg"`

brianignacio5, 2021-04-29:
Which version of openOCD are you using ?

pearson, 2021-04-29:
I'm currently using the latest `openocd-esp32-win32-0.10.0-esp32-202104011`. 

My apologies then. I'm still quite new to ESP-IDF, and I'm still figuring out where it pulls in the various configurations from. 😃 I'll put the issue into [openocd-esp32](https://github.com/espressif/openocd-esp32/issues).

closed 2021-04-29 by pearson

brianignacio5, 2021-05-10:
We are here to help ! Please let us know anyway we can help you get productive with Espressif devices, will try to point you where or help somehow