espressif/vscode-esp-idf-extension #392: [Bug Report]: "ESP-IDF Build, Flash and Monitor" will not flash through JTag

pearson, 2021-04-29:
I'm not certain if this is a bug or something I'm missing, but it smells like a bug...

If I build a project and then use the "ESP-IDF Flash device" button (or the "ESP-IDF: Flash (with JTag)" command), all is well and I get a JTag flash through a ESP-PROG to a ESP32 device. However, if I use the "ESP-IDF Build, Flash and Monitor" button (or command), the flash process attempts to do a UART flash through a non-existent COM1 port.

This is on Windows 10 with the current ESP-IDF and with openocd-esp32-win32-0.10.0-esp32-20210401. 

The `settings.json` in the project directory was set by "ESP-IDF: Select OpenOCD Board Configuration" and contains:
```yaml
{
"C_Cpp.intelliSenseEngine": "Tag Parser",
"idf.adapterTargetName": "esp32",
"idf.openOcdConfigs": [
	"interface/ftdi/esp32_devkitj_v1.cfg",
	"target/esp32.cfg"
],
"idf.flashType": "JTAG"
}
```

My assumption is that there shouldn't be another setting to tell it "no, really, I want JTag flashing", but perhaps I'm missing something. 😄

pearson, 2021-04-29:
An important point that I missed in my initial post is that I left out the `idf.portWin` parameter from the `settings.json` example intentionally. With a correct `idf.portWin`, "ESP-IDF Build, Flash and Monitor" will flash using UART rather than JTag, even though the `idf.flashType` parameter is set to `"JTAG"`. 

For example:
```yaml
{
"C_Cpp.intelliSenseEngine": "Tag Parser",
"idf.openOcdConfigs": [
	"interface/ftdi/esp32_devkitj_v1.cfg",
	"target/esp32.cfg"
],
"idf.adapterTargetName": "esp32",
"idf.portWin": "COM3",
"idf.flashType": "JTAG"
}
```

Leaving out the `idf.portWin` parameter simply pointed to the fact that "ESP-IDF Build, Flash and Monitor" was always flashing with UART.

brianignacio5, 2021-04-30:
Thanks for reporting, indeed we should fix this !

pearson, 2021-04-30:
> The JTAG flashing feature was added later that `build, flash and monitor`.

Aha! That would make sense.

I do get the feeling that JTAG flashing is not all that common, and that perhaps most things to do with JTAG could use a bit of love. This is off-topic, but a good example seems to be the [logging to host via JTAG feature and application tracing](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/app_trace.html#app-trace-logging-to-host). While it works, there doesn't seem to be any tools to monitor that logging in real time. It appears to require a clumsy combination of a telnet connection to OpenOCD in order to give the command to start a trace file, followed by using a python script to dump the log text from the trace file. 

I like the idea of the lower overhead logging that this offers, so clicking `build, flash and monitor` and having it flash with JTAG, followed by monitoring by JTAG, would be my ideal. Perhaps, sometime down the road, working on a real-time JTAG monitor might be my contribution to the project... 😃

meltdown03, 2021-05-05:
In linux I get this error when trying to flash my Kaluga V1.2 via JTAG:
``` bash
{"user":true,"message":"Failed to flash the device (JTag), please try again [got response: 'invalid command name \"program_esp_bins\"', expecting: '0']","stack":"Error: JTAG_FLASH_FAILED\n\tat /home/neal/.vscode/extensions/espressif.esp-idf-extension-1.0.3/dist/extension.js:8:299172\n\tat Generator.throw (<anonymous>)\n\tat a (/home/neal/.vscode/extensions/espressif.esp-idf-extension-1.0.3/dist/extension.js:8:264626)\n\tat runMicrotasks (<anonymous>)\n\tat processTicksAndRejections (internal/process/task_queues.js:97:5)","level":"error","timestamp":"2021-05-05T03:49:57.197Z"}
```

pearson, 2021-05-05:
@meltdown03 , your problem doesn't seem to be related to this bug report. 

That said, and at the risk of going off-topic but wanting to help...  Did you change your `idf.customExtraPaths` and your `idf.customExtraVars` to match the new OpenOCD 20201202 location?

meltdown03, 2021-05-05:
Yeah I did. I have it set in the main settings.json and the project. I
still get a failed+to+match+version error unless it's set in the main
settings.json. Maybe there is too much going on and I should start a new
issue. I also can't build the extension from source anymore either.

closed 2021-05-08 by brianignacio5

brianignacio5, 2021-05-10:
@meltdown03 maybe you can open an issue with details ?

meltdown03, 2021-05-10:
I didn't have the correct OPENOCD_SCRIPTS variable set. It works now,
thanks.

PR #400 (2021-05-08): refactor build flash cmds
Refactor build and flash to add JTAG/UART flashing on `build, flash and monitor` command.

Fix #392