espressif/vscode-esp-idf-extension #393: Unable to install - Invalid ESP-IDF

JamieEC, 2021-05-01:
I get this error when I load the plugin:
[image]
I then get this error when installing:
[image]
I tried the version from VSC-592, but to no avail.

Pinni3, 2021-05-03:
i have the exact same issue

brianignacio5, 2021-05-07:
Can you provide more information about your setup and steps to reproduce?

You can use the `ESP-IDF: Doctor command` to generate a report of your configuration.

**Logs**
If applicable, please share the log file which can be obtained from

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

creandotec, 2021-05-12:
Hi! Did you solve it? I was having the same problem, and I solved it enabling git to work from third party software

lucatedeschi, 2021-05-13:
Just solved the problem.
I was on a fresh install so some element was not set properly, make sure you have the following software installed:

- Python (downloadable here https://www.python.org/downloads/)
- C++ compiler for VS (downloadable here https://visualstudio.microsoft.com/it/visual-cpp-build-tools/)
- Git for windows (dowloadable here https://gitforwindows.org/)

brianignacio5, 2021-05-14:
We have a [prerequisites](https://github.com/espressif/vscode-esp-idf-extension#prerequisites) sections mentioning this but I guess its not visible right away.

brianignacio5, 2021-05-17:
Closing this issue due to lack of response from original user.

Please comment to re-open the issue if still facing difficulties.

closed 2021-05-17 by brianignacio5