espressif/vscode-esp-idf-extension #394: [Bug Report]: Problem installation ESP-IDF Extension for Visual Studio Code

aledoors, 2021-05-01:
[image]

Hello, i have the above problem when installing Espressif extension into VS Code. I already have intalled the ESP-IDF 4.2 CMD on the root, but VS Code can't find it. In addition i don't know why do i have another problem:

IDF_PYTHON_ENV_PATH: C:\Users\Alejandro\.espressif\python_env\idf4.2_py3.9_env
Python interpreter used: C:\Users\Alejandro\.espressif\python_env\idf4.2_py3.9_env\Scripts\python.exe
Command failed: "C:\Users\Alejandro\.espressif\python_env\idf4.2_py3.9_env\Scripts\python.exe" "C:\Users\Alejandro\esp\esp-idf\tools\check_python_dependencies.py" -r "C:\Users\Alejandro\esp\esp-idf\requirements.txt"

Thanks in advance.

brianignacio5, 2021-05-06:
Can you use the `ESP-IDF: Doctor command` to generate a report of your configuration ?

You have missing python packages, probably some error while installing. You can use an existing ESP-IDF install as described in [setup documentation](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/SETUP.md#setup-wizard) when choosing `Select ESP-IDF version` choose `Find ESP-IDF in your system`

aledoors, 2021-05-06:
Hi @brianignacio5 , thanks for your help

I can use ESP-IDF from CMD environment, but I've never used ESP doctor. In the command line I should only write "Doctor command"?
I tried to use the existing installation i have, but still doesn't work. Start the installation over again.

brianignacio5, 2021-05-07:
Hi @aledoors this is one of the commands that extension contribute to Visual studio Code. Just press F1 (or CTRL Shift P) and type `ESP-IDF Doctor` to search for this command. Information from your setup will be copied to the clipboard and you can paste it here.

aledoors, 2021-05-09:
Hello @brianignacio5, here is the information when i type ESP-IDF Doctor, as you said the problem is when install python packages. Then i tried to install the python packages typing ESP-IDF: Install ESP-IDF Python Packages, but got another issue. I show you below.

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19042 
System environment variable PATH 
 C:\Program Files (x86)\Siemens\Automation\SCADA-RT_V11\WinCC\Bin;C:\Program Files\Common Files\Siemens\Automation\Simatic OAM\bin;C:\Program Files (x86)\Common Files\Siemens\Bin;C:\Program Files (x86)\Common Files\Siemens\CommonArchiving;C:\Program Files (x86)\Common Files\Siemens\ACE\Bin;C:\Program Files (x86)\Rockwell Software\RSCommon;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\NAiS MEWNET;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\110\Tools\Binn\;C:\Program Files (x86)\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files\Microsoft SQL Server\120\DTS\Binn\;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Program Files\MATLAB\R2017b\runtime\win64;C:\Program Files\MATLAB\R2017b\bin;C:\Program Files\Git\cmd;C:\Program Files (x86)\Microsoft SQL Server\Client SDK\ODBC\110\Tools\Binn\;C:\Program Files (x86)\Microsoft SQL Server\120\DTS\Binn\;C:\Program Files (x86)\Siemens\Automation\SCADA-RT_V11\WinCC\Interfaces;C:\Program Files (x86)\Common Files\Siemens\;C:\Program Files (x86)\Common Files\Siemens\Interfaces;C:\Program Files (x86)\Common Files\Siemens\ACE\Interfaces;C:\Users\Alejandro\AppData\Local\Programs\Python\Python39\;C:\Users\Alejandro\AppData\Local\Programs\Python\Python39\Scripts\;C:\Users\Alejandro\AppData\Local\Programs\Python\Python39\;C:\Program Files\Java\jdk-11.0.7\bin;;C:\Users\Alejandro\AppData\Local\Microsoft\WindowsApps;C:\Users\Alejandro\AppData\Local\Programs\Microsoft VS Code\bin 
Visual Studio Code version 1.55.2 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.0.3 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-IDF Path (idf.espIdfPath) C:\Users\Alejandro\esp\esp-idf
Custom extra paths (idf.customExtraPaths) C:\Users\Alejandro\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin;C:\Users\Alejandro\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin;C:\Users\Alejandro\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\Alejandro\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\Alejandro\.espressif\tools\cmake\3.16.4\bin;C:\Users\Alejandro\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin;C:\Users\Alejandro\.espressif\tools\ninja\1.10.0;C:\Users\Alejandro\.espressif\tools\idf-exe\1.0.1;C:\Users\Alejandro\.espressif\tools\ccache\3.7;C:\Users\Alejandro\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"C:\\Users\\Alejandro\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20200709/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) C:\Users\Alejandro\.espressif\python_env\idf4.2_py3.9_env\Scripts\python.exe
Serial port (idf.port) COM3
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) C:\Users\Alejandro\.espressif
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-IDF Custom extra paths
Access to C:\Users\Alejandro\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin: true
Access to C:\Users\Alejandro\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin: true
Access to C:\Users\Alejandro\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: true
Access to C:\Users\Alejandro\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: true
Access to C:\Users\Alejandro\.espressif\tools\cmake\3.16.4\bin: true
Access to C:\Users\Alejandro\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin: true
Access to C:\Users\Alejandro\.espressif\tools\ninja\1.10.0: true
Access to C:\Users\Alejandro\.espressif\tools\idf-exe\1.0.1: true
Access to C:\Users\Alejandro\.espressif\tools\ccache\3.7: true
Access to C:\Users\Alejandro\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.21.0.windows.1
ESP-IDF version 4.2
Python version 3.9.4
Python's pip version 21.1.1
[... cut ...]
  generating cffi module 'build\\temp.win-amd64-3.9\\Release\\gevent.libuv._corecffi.c'
  creating build\temp.win-amd64-3.9
  creating build\temp.win-amd64-3.9\Release
  building 'gevent.libev.corecext' extension
  error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/
  ----------------------------------------
  ERROR: Failed building wheel for gevent
ERROR: Could not build wheels for gevent which use PEP 517 and cannot be installed directly

brianignacio5, 2021-05-10:
Well I would suggest, as your error output shows, to install the [Microsoft C++ Build Tools](https://visualstudio.microsoft.com/visual-cpp-build-tools/) since it is necessary to compile `wheels` for your setup and try executing `ESP-IDF: Install ESP-IDF Python Packages` command again.

aledoors, 2021-05-11:
Hi @brianignacio5, you were correct, I had to reinstall https://visualstudio.microsoft.com/visual-cpp-build-tools/ and the problem was resolved. Thanks you very much for your help.

closed 2021-05-12 by brianignacio5

RudyFiero, 2021-07-13:
Hi Brian
I was able to install the changes to my home machine. There was one error.

----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at ENOENT: no such file or directory, open 'c:\Users\Rudy\Documents\PythonProgramming\.vscode\launch.json'
--------------------------------------------------------------------------------------------------------------------------------------------

I am pretty sure I didn't get this error on my work machine. Did I miss a step on my home machine? Is this something that I can ignore?