espressif/vscode-esp-idf-extension #395: vscode esp-idf

774759517, 2021-05-03:
Issue Type: <b>Bug</b>

尝试多次配置不能成功编译工程

Extension version: 1.0.3
VS Code version: Code 1.55.2 (3c4e3df9e89829dce27b7b5c24508306b151f30d, 2021-04-13T09:35:57.887Z)
OS version: Windows_NT x64 10.0.18363

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i5-9400F CPU @ 2.90GHz (6 x 2904)|
|GPU Status|2d_canvas: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>oop_rasterization: enabled<br>opengl: enabled_on<br>protected_video_decode: unavailable_off<br>rasterization: enabled<br>skia_renderer: enabled_on<br>video_decode: enabled<br>vulkan: disabled_off<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|undefined|
|Memory (System)|15.94GB (8.86GB free)|
|Process Argv|--crash-reporter-id bc8789cf-290d-40bb-866a-bcaad9ec634a|
|Screen Reader|no|
|VM|44%|
</details><details>
<summary>A/B Experiments</summary>

```
vsliv368cf:30146710
vsreu685:30147344
python383:30185418
pythonvspyt602cf:30294773
vspor879:30202332
vspor708:30202333
vspor363:30204092
pythonvspyt639:30291489
pythontb:30283811
pythonvspyt551:30291414
vspre833cf:30267465
pythonptprofiler:30281270
vscnewfilehidden:30294815
vshan820:30294714
vscorecescf:30290706
pythondataviewer:30285071
vscus158:30286553

```

</details>

brianignacio5, 2021-05-06:
Could you provide more information about your setup?

There is a `ESP-IDF: Doctor command` that will copy the information to your clipboard which you can post here.

brianignacio5, 2021-05-10:
Closing this issue from lack of response from the user.

Please comment here to re-open the issue.

closed 2021-05-10 by brianignacio5