espressif/vscode-esp-idf-extension #396: ESP-IDF Menuconfig hang on loading initial values

edouardreg, 2021-05-03:
Since the update to 1.0.3, I have the same issue as #387 issue (which is now closed but I still have this pb).
So I had try all solutions suggested in issue #387 without any succes.

(Windows 10 64 bits, latest vscode, esp-idf master, python 3.9)

But since then also, I can't anymore launch the GUI configuration tool.
I have downgraded to 1.0.2 : projects builds again, but the  GUI configuration tool is still broken ?

I had remove the extension and re-install : no changes.

edouardreg, 2021-05-03:
Here is the result of doctor command :
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19042 
System environment variable PATH 
 C:\Program Files (x86)\Microsoft\Edge\Application;C:\Users\edoua\AppData\Local\Programs\Python\Python39\Scripts\;C:\Users\edoua\AppData\Local\Programs\Python\Python39\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\dotnet\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\170\Tools\Binn\;C:\Program Files\Git\cmd;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\130\Tools\Binn\;C:\Program Files (x86)\Microsoft SQL Server\140\Tools\Binn\;C:\Program Files\Microsoft SQL Server\140\Tools\Binn\;C:\Program Files\Microsoft SQL Server\140\DTS\Binn\;C:\Program Files\PuTTY\;C:\Program Files\nodejs\;C:\ProgramData\chocolatey\bin;C:\Users\edoua\AppData\Local\Microsoft\WindowsApps;C:\Users\edoua\.dotnet\tools;C:\Users\edoua\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\edoua\AppData\Roaming\npm 
Visual Studio Code version 1.55.2 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\cmd.exe 
ESP-IDF Extension version 1.0.3 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-IDF Path (idf.espIdfPath) d:\ESP-IDF\esp-idf
Custom extra paths (idf.customExtraPaths) d:\ESP-IDF\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin;d:\ESP-IDF\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin;d:\ESP-IDF\.espressif\tools\xtensa-esp32s3-elf\esp-2020r3-8.4.0\xtensa-esp32s3-elf\bin;d:\ESP-IDF\.espressif\tools\riscv32-esp-elf\1.24.0.123_64eb9ff-8.4.0\riscv32-esp-elf\bin;d:\ESP-IDF\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;d:\ESP-IDF\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;d:\ESP-IDF\.espressif\tools\cmake\3.16.4\bin;d:\ESP-IDF\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin;d:\ESP-IDF\.espressif\tools\ninja\1.10.0;d:\ESP-IDF\.espressif\tools\idf-exe\1.0.1;d:\ESP-IDF\.espressif\tools\ccache\3.7;d:\ESP-IDF\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"d:\\ESP-IDF\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20200709/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) d:\ESP-IDF\.espressif\python_env\idf4.3_py3.9_env\Scripts\python.exe
Serial port (idf.port) /dev/ttyUSB1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) d:\ESP-IDF\.espressif
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-IDF Custom extra paths
Access to d:\ESP-IDF\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin: true
Access to d:\ESP-IDF\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin: true
Access to d:\ESP-IDF\.espressif\tools\xtensa-esp32s3-elf\esp-2020r3-8.4.0\xtensa-esp32s3-elf\bin: true
Access to d:\ESP-IDF\.espressif\tools\riscv32-esp-elf\1.24.0.123_64eb9ff-8.4.0\riscv32-esp-elf\bin: true
Access to d:\ESP-IDF\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: true
Access to d:\ESP-IDF\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: true
Access to d:\ESP-IDF\.espressif\tools\cmake\3.16.4\bin: true
Access to d:\ESP-IDF\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin: true
Access to d:\ESP-IDF\.espressif\tools\ninja\1.10.0: true
Access to d:\ESP-IDF\.espressif\tools\idf-exe\1.0.1: true
Access to d:\ESP-IDF\.espressif\tools\ccache\3.7: true
Access to d:\ESP-IDF\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.29.2.windows.1
ESP-IDF version 4.3
Python version 3.9.2
Python's pip version 20.2.3
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
bidict version: 0.21.2
bitstring version: 3.1.7
Brotli version: 1.0.9
certifi version: 2020.12.5
cffi version: 1.14.5
chardet version: 4.0.0
click version: 7.1.2
contextlib2 version: 0.6.0.post1
cryptography version: 3.4.6
ecdsa version: 0.16.1
esp-windows-curses version: 0.1
Flask version: 0.12.5
Flask-Compress version: 1.9.0
Flask-SocketIO version: 2.9.6
future version: 0.18.2
gcovr version: 4.2
gdbgui version: 0.13.2.0
gevent version: 1.5.0
greenlet version: 1.0.0
idf-component-manager version: 0.2.1a0
idna version: 2.10
itsdangerous version: 1.1.0
Jinja2 version: 2.11.3
kconfiglib version: 13.7.1
lxml version: 4.6.2
MarkupSafe version: 1.1.1
pip version: 20.2.3
[... cut ...]
Check Debug AdapterPython packages Python requirements from c:\Users\edoua\.vscode\extensions\espressif.esp-idf-extension-1.0.3\esp_debug_adapter\requirements.txt are satisfied.
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at ENOENT: no such file or directory, open 'd:\ESP-IDF\Projets\GazSol_OTA_ESPIDF_MASTER\.vscode\launch.json'
--------------------------------------------------------------------------------------------------------------------------------------------

edouardreg, 2021-05-03:
Some news : after removing vscode and all folders like user/.vscod... appdate/code... it was OK.
So if you enter those trouble try a clean install of vscode ?

closed 2021-05-03 by edouardreg

remyhx, 2022-05-28:
In my case it was after an update of esp-idf and had to change the updated values in launch.json -> "idf.customExtraPaths"

After fullclean by idf.py fullclean I rebuild and the errors were ok.

eschumacher, 2022-07-27:
I hit this when using a project path under Program Files (stupid, I know).  Resolved when using a more appropriate project path.