espressif/vscode-esp-idf-extension #397: Is Using the "Tag Parser" Engine Still Needed?

pearson, 2021-05-03:
The ESP-IDF VS Code Extension's [marketplace page](https://marketplace.visualstudio.com/items?itemName=espressif.esp-idf-extension) states:

> Note: If you want to get code navigation and ESP-IDF function references, the Microsoft C/C++ Extension can be used to resolve header/source links. By default, projects created with ESP-IDF: Create project from extension template or ESP-IDF: Show Examples Projects tries to resolve headers by manually recursing ESP-IDF directory sources with the Tag Parser engine.

This doesn't seem to be the case -- or perhaps it is no longer the case. Without the `"C_Cpp.intelliSenseEngine": "Tag Parser"` line in `settings.json`, but with the `"compileCommands": "${workspaceFolder}/build/compile_commands.json"` line in `c_cpp_properties.json`, ESP-IDF code navigation and ESP-IDF function references work correctly for me. 

Double-checking my VS Code settings, `"C_Cpp.intelliSenseEngine": "Default"`

A downside to using the Tag Parser engine is that inactive code regions (such as inactive preprocessor definitions) will not be dimmed. (I personally like that dimming and that led me to test this out.) Whether there are other pros and cons to using / not using the Tag Parser engine, I do not know.

brianignacio5, 2021-05-07:
But have you consider the scenario before you have a working `compile_commands.json` ? Tag parser is used to recursively explore IDF_PATH components to resolve headers which gives us Intellisense before we can build. 

We need to work on an alternative I agree, like add a configuration provider module for the c/cpp extension.

pearson, 2021-05-07:
> @brianignacio5 
> But have you consider the scenario before you have a working `compile_commands.json` ? Tag parser is used to recursively explore IDF_PATH components to resolve headers which gives us Intellisense before we can build.

I've found that the first build is creating a working `compile_commands.json` without Tag Parser being active. I've tested this a few times with a simple hello world project (attached here as [_BaseProject.zip](https://github.com/espressif/vscode-esp-idf-extension/files/6443708/_BaseProject.zip)) that does not have an initial build directory, and so (obviously) it doesn't have a `compile_commands.json`. 

It seems that an initial build generates `build/compile_commands.json` without Tag Parser being active, and if there's a `compile_commands` line in the `c_cpp_properties.json` (there is in my test project) then everything seems to work correctly after the first build.

I've double-checked in my VS Code settings that `C_Cpp.intelliSenseEngine` is set to `"C_Cpp.intelliSenseEngine": "Default"`. (Although that code dimming is working, so presumably Tag Parser couldn't be active.)

To be fair, I'm fairly new to VS Code, and I'm definitely new to ESP-IDF and the extension, so I could very well be missing something here. With how my system is set up at the moment, though, it seems like Tag Parser is not needed.

brianignacio5, 2021-05-08:
I understand your point but that is assuming that your project can be correctly build from the start. This is (or should be) true for any example for supported frameworks but for any user projects this is not guaranteed. `Tag Parser` is required to resolve headers (when compile_commands.json doesn't exist or is not specified) for example to fix the `portTICK_PERIOD_MS` for example.

pearson, 2021-05-08:
@brianignacio5 ,

Ah, I see. Okay, thanks for the explanation -- it's appreciated!!

closed 2021-05-08 by pearson