espressif/vscode-esp-idf-extension #399: [Feature Request]: Have installer continue with existing ESP-IDF dir

asmblur, 2021-05-06:
It would be nice if the installer would allow you to specify a directory that contains a previous installation that failed rather than having to delete the existing directory and downloaded data.  As it stands, if an issue occurs(like in my case, the "pip" module wasn't installed correctly so install failed) then you have to delete the existing esp-idf directory before the installer will start(restart).  This is very slow and quite a pain when trying to debug the dependencies.

brianignacio5, 2021-05-06:
[image]

You can use an existing ESP-IDF by selecting `Find ESP-IDF in your system` when you `Select ESP-IDF version`

brianignacio5, 2021-05-08:
Closing this issue since the feature already exists.

closed 2021-05-08 by brianignacio5

pearson, 2021-05-12:
@brianignacio5 , it might be useful to replace the "esp-idf directory exists" error with a message stating that the directory (and the existing download) can be reused. I ran into this earlier today when installing on a different PC and from that point in an install it appeared as if I needed to delete the esp-idf directory and start again.