espressif/vscode-esp-idf-extension #403: [Bug Report]: Unable to Start App Trace

jitenshap, 2021-05-08:
Hi. I trying to OpenOCD debug for esp32.
I bought ESP-PROG at mouser, and set up the extension.
By that solution, It became possible to flash by JTAG.
https://github.com/espressif/vscode-esp-idf-extension/issues/289#issuecomment-758372220
However, There are some issue in Start App Trace.
It seems that there are problems in escape for ":".
I understand already problems was reported that.
But still not fixing even version v1.0.3.
Not working of breakpoint feature of App trace.

[doctor.txt](https://github.com/espressif/vscode-esp-idf-extension/files/6445994/doctor.txt)
[openocd.txt](https://github.com/espressif/vscode-esp-idf-extension/files/6445995/openocd.txt)

jitenshap, 2021-05-10:
Thank you for reply.
My esp_idf_vsc_ext.log is here.
[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/6452998/esp_idf_vsc_ext.log)

brianignacio5, 2021-05-19:
@jitenshap Please try this https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/61308228.zip vsix installer to see if it solves your issue.

closed 2021-05-21 by brianignacio5

jitenshap, 2021-05-21:
@brianignacio5 
Thank you for new installer for fix it.
I trying this. But, I received a new error message.
❌ Error: Invalid 'cmd_data->poll_period' arg!
Erro❌ r: Not enough args! Need 2 trace data destinations!
[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/6523654/esp_idf_vsc_ext.log)
[doctor(2).txt](https://github.com/espressif/vscode-esp-idf-extension/files/6523656/doctor.2.txt)
[openocd(2).txt](https://github.com/espressif/vscode-esp-idf-extension/files/6523657/openocd.2.txt)

PR #410 (2021-05-21): fix trace path in app trace manager
Fix #403 

Fix given path for application trace.