espressif/vscode-esp-idf-extension #407: [Bug Report]: pip is not part of the virtual environment created in WSL 2.

pearson, 2021-05-12:
As noted by @brianignacio5 in [a post on esp32.com](https://esp32.com/viewtopic.php?f=40&t=19793), pip is not part of the virtual environment created in WSL 2. It is not clear to me whether this is due of the ESP-IDF extension setup or not.

@brianignacio5 , you had asked the OP of that thread to check if pip was in his `idf4.2_py3.8_env/bin/`. For me it was not. I only saw two symbolic links to the python3 binary:
`lrwxrwxrwx 1 chris chris    7 May 11 17:21 python -> python3`
`lrwxrwxrwx 1 chris chris   16 May 11 17:21 python3 -> /usr/bin/python3`

After following your pip install directions in that post, and after using the ESP-IDF Setup again, the setup completed.

pearson, 2021-05-12:
To follow up on this, the setup completed, but there was an odd problem -- it appears as if some dependencies were not installed.

After using ESP-IDF Setup, an attempt to build the hello world example gave the following messages:
```
The following Python requirements are not satisfied:
pyserial>=3.0
cryptography>=2.1.4
pyparsing>=2.0.3,<2.4.0
pyelftools>=0.22
gdbgui==0.13.2.0
reedsolo>=1.5.3,<=1.5.4
bitstring>=3.1.6
ecdsa>=0.16.0
To install the missing packages, please run "/home/chris/esp/esp-idf/install.sh"
Diagnostic information:
    IDF_PYTHON_ENV_PATH: /home/chris/.espressif/python_env/idf4.2_py3.8_env
    Python interpreter used: /home/chris/.espressif/python_env/idf4.2_py3.8_env/bin/python
```

Looking over the ESP-IDF Setup output, it appears that these python packages were not installed. None of those package names appeared in that output. However, after running the `/home/chris/esp/esp-idf/install.sh` script as the error suggested, the hello world example could be built successfully.

Perhaps the install script used in the ESP-IDF Setup does not have all of the needed packages that included in the `install.sh` script?

brianignacio5, 2021-05-12:
Hi @pearson thanks for your kind feedback.

I want to know a bit about your setup when running the `ESP-IDF: Configure ESP-IDF extension` command. If python package s are not being installed (without any actual error being shown in the logs or output ) it might be because Visual Studio Code is not running in the WSL distribution.

Are you using the [Remote WSL](https://marketplace.visualstudio.com/items?itemName=ms-vscode-remote.remote-wsl) extension and using the `Remote-WSL: Open Folder in WSL...` command to launch a visual studio code window that is running from the WSL ? If you just open the WSL from the visual studio code terminal, the rest of visual studio code is still running in the Windows host machine.

pearson, 2021-05-12:
I'm happy to contribute where I can. 😄 

Just in case it is relevant, I'm using WSL2 on a standard Windows 10 Pro install.

I am running the Remote WSL extension on a Windows-based VS Code. I had initially installed VS Code directly into the WSL distribution, but a warning message described the "proper" way to use VS Code with WSL, so I uninstalled VS Code from WSL and switched to using the Remote WSL extension.

I decided to try this again from scratch, so I installed a new Ubuntu distribution. I intentionally installed nothing before beginning just to see what errors popped up.

Here is my 
[\home\chris\.vscode-server\extensions\espressif.esp-idf-extension-1.0.3\esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/6469256/esp_idf_vsc_ext.log)
 log. It has been edited with breaks to show where I was during the install. The break notes FOLLOW the log output, as I simply typed what had happened and saved the file. There are two more outputs in files below.

Here are the steps taken:
1. Unregistered Ubuntu and reinstalled from Microsoft Store
2. Re-enabled Remote-WSL
3. Installed these extensions to WSL:Ubuntu:
   - C/C++ Extension Pack
      - An error popped up stating "We were unable to install support for joining sessions using a browser link. You may be missing key Linux libraries. Install them now?". I had not installed `build-essentials` and such. The installation of the C/C++ Extension Pack continued after it installed the libraries and reloaded VS Code.
   - Espressif IDF
4. Used the ESP-IDF: Configure ESP-IDF extension
   - CMake and Ninja were missing. These were installed normally with:
       - `sudo apt install cmake`
       - `sudo apt install ninja-build`
5. Used the ESP-IDF: Configure ESP-IDF extension
   - The "Express" option was chosen
   - The install failed as `/usr/bin/python3 -m pip` was invalid
   - Python is not installed by default. It was installed normally with `sudo apt install python3 python3-pip`.
6. The ESP-IDF configure "Install" button was pressed again
   - Install failed as `python3-venv` was missing
   - Note that the install screen still showed the previous error, `"/usr/bin/python3 -m pip" is not valid.`. I had not clicked the close button on that error. Obviously this could be confusing.
7. Clicked "Install" again, got the "esp-idf already exists. Delete it or use another location." error. (This error could use to be changed to state that the existing directory can be used.)
8. Selected "Find ESP-IDF in your system" and clicked "Install" again.
   - Install failed with `/home/chris/.espressif/python_env/idf4.2_py3.8_env/bin/python: No module named pip`
       - Pip is definitely installed in the main system
   - Note that no new red-banner error message showed up in the GUI area. I had closed the previous error before clicking install. Showing an error banner would be useful
   - Pip was installed into the virtual environment with `/home/chris/.espressif/python_env/idf4.2_py3.8_env/bin/python ./get-pip.py`
9. ESP-IDF: Show Example Projects
   - This fails with the message "No framework selected to load examples". No pop-up list appears with the framework choices.
      - This error may have been be due to not having a folder open yet. If a folder is opened, and then the "ESP-IDF: Show Example Projects" command is used, the frameworks pop-up list appears. 
      - However this problem did **not** reoccur after successfully opening an example project, then opening a "New WSL Window", then using the "ESP-IDF: Show Example Projects" command again.
10. ESP-IDF: Show Example Projects (With a folder now opened.)
    - A "hello world" example was chosen.
    - "Build" was pressed.
    - Build messages: (Full build messages are here: [first_build_attempt.txt](https://github.com/espressif/vscode-esp-idf-extension/files/6469211/first_build_attempt.txt))
```
The following Python requirements are not satisfied:
pyserial>=3.0
cryptography>=2.1.4
pyparsing>=2.0.3,<2.4.0
pyelftools>=0.22
gdbgui==0.13.2.0
reedsolo>=1.5.3,<=1.5.4
bitstring>=3.1.6
ecdsa>=0.16.0
To install the missing packages, please run "/home/chris/esp/esp-idf/install.sh"
Diagnostic information:
    IDF_PYTHON_ENV_PATH: /home/chris/.espressif/python_env/idf4.2_py3.8_env
    Python interpreter used: /home/chris/.espressif/python_env/idf4.2_py3.8_env/bin/python
```
11. "ESP-IDF: Install ESP-IDF extension Python Packages" was used.
    -  Log output is here: [using_Install_ESP-IDF_Python_Packages_command.txt](https://github.com/espressif/vscode-esp-idf-extension/files/6469213/using_Install_ESP-IDF_Python_Packages_command.txt). Essentially it installed the missing packages.
    - Note that in my first WSL install, I had used the `/home/chris/esp/esp-idf/install.sh` script as suggested by the build output. They seem to both do the same thing.
12. A build was attempt and it completed successfully!

### Final Thoughts
A few thoughts about the "ESP-IDF: Configure ESP-IDF extension" process that came out of this:
- the [prerequisites page](https://github.com/espressif/vscode-esp-idf-extension#Prerequisites) could use to have a bit more information regarding what python packages are required. (`python3` itself, `python3-pip`, and `python3-venv`)
- More red-banner errors would be useful, as a few errors do not show any red banner errors at all. Showing some errors, but then not others, is inconsistent and doesn't point the user to look at the output window. 
   - I felt this way on my first install of the extension, as all I was looking at was the screen I had clicked "launch" on previously.)

brianignacio5, 2021-05-13:
There is an already existing [WSL Documentation](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/WSL.md) that mentions the install of python3-pip, env and these things but is probably not visible enough for most users, as most documentation is. We could improve this file and/or add the WSL tutorial doc.

So basically:

1) run `apt-get` from the WSL terminal to install prerequisites in WSL documentation.
2) Run the extension `Configure ESP-IDF extension` command
3) Extension should work on WSL 2 now.

pearson, 2021-05-13:
> There is an already existing [WSL Documentation](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/WSL.md) that mentions the install of python3-pip, env and these things but is [...]

Agreed. I did not know that document existed. A clear link in the "install" document would be very helpful there.

> So basically:
> 
> 1. run `apt-get` from the WSL terminal to install prerequisites in WSL documentation.
> 2. Run the extension `Configure ESP-IDF extension` command
> 3. Extension should work on WSL 2 now.

This misses the issue regarding pip not being installed into the virtual environment. I had pip installed before the creation of the python virtual environment and yet I saw the error mentioned in #8 above. A manual install of pip into that virtual environment was required to continue the ESP-IDF extension install.

PR #408 (2021-05-17): add onboarding doc
Add onboarding documentation with links to the rest of documentation

Fix #407 by adding related documentation.
closed 2021-05-17 by brianignacio5