espressif/vscode-esp-idf-extension #409: [Bug Report]: Instalation of tools fails because of old pip with error no such option: --no-warn-script-location

tomasxvavra, 2021-05-14:
**Describe the bug**

I am trying to install this plugin on Linux Mint 19.3 Tricia which ships pip3 9.0.1. Even if pip can update itself - it is not refclelted when venv is created. /home/xxxxx/work/esp/vscode/tools/python_env/idf4.2_py3.6_env -m pip -V reports 9.0.1

**To Reproduce**
Install plugin on Linux distribution with pip not supporting --no-warn-script-location

**Expected behavior**
No errors.

- OS Name & Version: Linux Mint 19.3 Tricia
- VSCode Version: 1.56.2
- ESP-IDF Version: v4.2.1
- Python Version: 3.6.9

**Logs**

Creating a new Python environment in /home/xxxxx/work/esp/vscode/tools/python_env/idf4.2_py3.6_env ...

Collecting wheel
  Using cached https://files.pythonhosted.org/packages/65/63/39d04c74222770ed1589c0eaba06c05891801219272420b40311cd60c880/wheel-0.36.2-py2.py3-none-any.whl
Installing collected packages: wheel
Successfully installed wheel-0.36.2

Usage:   
  /home/xxxxx/work/esp/vscode/tools/python_env/idf4.2_py3.6_env/bin/python -m pip install [options] <requirement specifier> [package-index-options] ...
  /home/xxxxx/work/esp/vscode/tools/python_env/idf4.2_py3.6_env/bin/python -m pip install [options] -r <requirements file> [package-index-options] ...
  /home/xxxxx/work/esp/vscode/tools/python_env/idf4.2_py3.6_env/bin/python -m pip install [options] [-e] <vcs project url> ...
  /home/xxxxx/work/esp/vscode/tools/python_env/idf4.2_py3.6_env/bin/python -m pip install [options] [-e] <local project path> ...
  /home/xxxxx/work/esp/vscode/tools/python_env/idf4.2_py3.6_env/bin/python -m pip install [options] <archive url/path> ...

**no such option: --no-warn-script-location**
Command failed: "/home/xxxxx/work/esp/vscode/tools/python_env/idf4.2_py3.6_env/bin/python" -m pip install --no-warn-script-location -r "/home/xxxxx/work/esp/vscode/esp-idf/requirements.txt"

brianignacio5, 2021-05-17:
I would recommend:

1) Delete the virtual env in ` /home/xxxxx/work/esp/vscode/tools/python_env/idf4.2_py3.6_env`
2) Update the pip linked with the system python being used `/usr/bin/python3 -m pip install pip --upgrade` ?
3) Run the `ESP-IDF: Install ESP-IDF Python Packages ` command to create the virtual env and install packages.

Is there a reason for you not to update pip ?

tomasxvavra, 2021-05-17:
Thank you for reply, this unfortunately does not work.

```
/usr/bin/python3 -m pip install pip --upgrade
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: pip in /home/tomas/.local/lib/python3.6/site-packages (21.1.1)
```

I have tried to update the system pip but I am not able to.

```
sudo -H pip3 install --upgrade pip
[sudo] password for tomas:              
Requirement already satisfied: pip in /usr/local/lib/python3.6/dist-packages (21.1.1)
```

Some additional info: https://stackoverflow.com/questions/51720909/how-to-get-python-m-venv-to-directly-install-latest-pip-version/60217751#60217751

brianignacio5, 2021-05-17:
Maybe upgrade pip directly in the virtual env ?

`/home/xxxxx/work/esp/vscode/tools/python_env/idf4.2_py3.6_env/bin/python -m pip install --upgrade pip` ?

tomasxvavra, 2021-05-17:
This could work, but how do I continue the plugin configuration ater it fails during the pip step? Maybe the plugin itself should do it?

brianignacio5, 2021-05-17:
This is the last step of the extension setup, which means having python packages installed should complete the setup as described in [setup](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/SETUP.md#setup-wizard) documentation.

brianignacio5, 2021-05-19:
Closing this issue since it is related to old pip version and not an issue with the extension.

closed 2021-05-19 by brianignacio5