espressif/vscode-esp-idf-extension #411: [Bug Report]:

huangbenlsl, 2021-05-19:
idf_versions.txt has (1) KB
idf_versions.txt progress: 100.00% (0.08 / 0.08) KB
Installing ESP-IDF extension python packages in C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\Scripts\python.exe ...

Looking in indexes: https://pypi.tuna.tsinghua.edu.cn/simple
ERROR: Exception:
Traceback (most recent call last):
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_internal\cli\base_command.py", line 180, in _main
    status = self.run(options, args)
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_internal\cli\req_command.py", line 204, in wrapper
    return func(self, options, args)
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_internal\commands\install.py", line 319, in run
    reqs, check_supported_wheels=not options.target_dir
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_internal\resolution\resolvelib\resolver.py", line 128, in resolve
    requirements, max_rounds=try_to_avoid_resolution_too_deep
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_vendor\resolvelib\resolvers.py", line 473, in resolve
    state = resolution.resolve(requirements, max_rounds=max_rounds)
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_vendor\resolvelib\resolvers.py", line 341, in resolve
    name, crit = self._merge_into_criterion(r, parent=None)
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_vendor\resolvelib\resolvers.py", line 172, in _merge_into_criterion
    if not criterion.candidates:
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_vendor\resolvelib\structs.py", line 139, in __bool__
    return bool(self._sequence)
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_internal\resolution\resolvelib\found_candidates.py", line 143, in __bool__
    return any(self)
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_internal\resolution\resolvelib\found_candidates.py", line 129, in <genexpr>
    return (c for c in iterator if id(c) not in self._incompatible_ids)
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_internal\resolution\resolvelib\found_candidates.py", line 30, in _iter_built
    for version, func in infos:
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_internal\resolution\resolvelib\factory.py", line 261, in iter_index_candidate_infos
    hashes=hashes,
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_internal\index\package_finder.py", line 879, in find_best_candidate
    candidates = self.find_all_candidates(project_name)
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_internal\index\package_finder.py", line 824, in find_all_candidates
    page_candidates = list(page_candidates_it)
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_internal\index\sources.py", line 134, in page_candidates
    yield from self._candidates_from_page(self._link)
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_internal\index\package_finder.py", line 783, in process_project_url
    html_page = self._link_collector.fetch_page(project_url)
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_internal\index\collector.py", line 512, in fetch_page
    return _get_html_page(location, session=self.session)
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_internal\index\collector.py", line 422, in _get_html_page
    resp = _get_html_response(url, session=session)
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_internal\index\collector.py", line 137, in _get_html_response
    "Cache-Control": "max-age=0",
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_vendor\requests\sessions.py", line 555, in get
    return self.request('GET', url, **kwargs)
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_internal\network\session.py", line 449, in request
    return super().request(method, url, *args, **kwargs)
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_vendor\requests\sessions.py", line 542, in request
    resp = self.send(prep, **send_kwargs)
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_vendor\requests\sessions.py", line 655, in send
    r = adapter.send(request, **kwargs)
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_vendor\cachecontrol\adapter.py", line 53, in send
    resp = super(CacheControlAdapter, self).send(request, **kw)
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_vendor\requests\adapters.py", line 449, in send
    timeout=timeout
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_vendor\urllib3\connectionpool.py", line 696, in urlopen
    self._prepare_proxy(conn)
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_vendor\urllib3\connectionpool.py", line 964, in _prepare_proxy
    conn.connect()
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_vendor\urllib3\connection.py", line 359, in connect
    conn = self._connect_tls_proxy(hostname, conn)
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_vendor\urllib3\connection.py", line 506, in _connect_tls_proxy
    ssl_context=ssl_context,
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_vendor\urllib3\util\ssl_.py", line 432, in ssl_wrap_socket
    ssl_sock = _ssl_wrap_socket_impl(sock, context, tls_in_tls)
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_vendor\urllib3\util\ssl_.py", line 474, in _ssl_wrap_socket_impl
    return ssl_context.wrap_socket(sock)
  File "C:\Users\huangben\AppData\Local\Programs\Python\Python37-32\lib\ssl.py", line 423, in wrap_socket
[... 119 lines cut ...]
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_vendor\requests\sessions.py", line 542, in request
    resp = self.send(prep, **send_kwargs)
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_vendor\requests\sessions.py", line 655, in send
    r = adapter.send(request, **kwargs)
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_vendor\cachecontrol\adapter.py", line 53, in send
    resp = super(CacheControlAdapter, self).send(request, **kw)
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_vendor\requests\adapters.py", line 449, in send
    timeout=timeout
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_vendor\urllib3\connectionpool.py", line 696, in urlopen
    self._prepare_proxy(conn)
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_vendor\urllib3\connectionpool.py", line 964, in _prepare_proxy
    conn.connect()
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_vendor\urllib3\connection.py", line 359, in connect
    conn = self._connect_tls_proxy(hostname, conn)
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_vendor\urllib3\connection.py", line 506, in _connect_tls_proxy
    ssl_context=ssl_context,
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_vendor\urllib3\util\ssl_.py", line 432, in ssl_wrap_socket
    ssl_sock = _ssl_wrap_socket_impl(sock, context, tls_in_tls)
  File "C:\Users\huangben\.espressif\python_env\idf4.2_py3.7_env\lib\site-packages\pip\_vendor\urllib3\util\ssl_.py", line 474, in _ssl_wrap_socket_impl
    return ssl_context.wrap_socket(sock)
  File "C:\Users\huangben\AppData\Local\Programs\Python\Python37-32\lib\ssl.py", line 423, in wrap_socket
    session=session
  File "C:\Users\huangben\AppData\Local\Programs\Python\Python37-32\lib\ssl.py", line 827, in _create
    raise ValueError("check_hostname requires server_hostname")
ValueError: check_hostname requires server_hostname

brianignacio5, 2021-05-19:
Hi @huangbenlsl you have a connection issue from your Python setup and you are using the pypi index from https://pypi.tuna.tsinghua.edu.cn/simple which might not have all required python packages.

Make sure you have configured SSL for Python properly.

brianignacio5, 2021-05-31:
Closing this issue for lack of response from user. Please comment here if you still have any issues.

closed 2021-05-31 by brianignacio5