espressif/vscode-esp-idf-extension #413: A lot of problem with idf and vs code

toosimit, 2021-05-19:
I drained absolutely drained !!! it didn't work !! I download more than 10 GB what ever needed it for installing IDF  but still it has lot problem ! I can't connect to esp32 I cant build project I can't do anything!

this is debugging message 

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19041 
System environment variable PATH 
 c:\windows\system32;C:\Program Files\Git\cmd;C:\Program Files\WireGuard\;C:\Users\HP\AppData\Local\Programs\Python\Python38\Scripts\;C:\Users\HP\AppData\Local\Programs\Python\Python38\;windows\system32;;C:\Users\HP\AppData\Local\Programs\Microsoft VS Code\bin 
Visual Studio Code version 1.56.2 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.0.3 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-IDF Path (idf.espIdfPath) c:\msys64\home\HP\esp\esp-idf
Custom extra paths (idf.customExtraPaths) C:\Users\HP\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin;C:\Users\HP\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin;C:\Users\HP\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\HP\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\HP\.espressif\tools\cmake\3.16.4\bin;C:\Users\HP\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin;C:\Users\HP\.espressif\tools\ninja\1.10.0;C:\Users\HP\.espressif\tools\idf-exe\1.0.1;C:\Users\HP\.espressif\tools\ccache\3.7;C:\Users\HP\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"C:\\Users\\HP\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20200709/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) C:\Users\HP\.espressif\python_env\idf4.2_py3.9_env\Scripts\python.exe
Serial port (idf.port) /dev/ttyUSB1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) ${env:IDF_TOOLS_PATH}
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-IDF Custom extra paths
Access to C:\Users\HP\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin: true
Access to C:\Users\HP\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin: true
Access to C:\Users\HP\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: true
Access to C:\Users\HP\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: true
Access to C:\Users\HP\.espressif\tools\cmake\3.16.4\bin: true
Access to C:\Users\HP\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin: true
Access to C:\Users\HP\.espressif\tools\ninja\1.10.0: true
Access to C:\Users\HP\.espressif\tools\idf-exe\1.0.1: true
Access to C:\Users\HP\.espressif\tools\ccache\3.7: true
Access to C:\Users\HP\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) false
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.31.1.windows.1
ESP-IDF version 4.2.1
Python version 3.9.0
Python's pip version 21.1.1
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
certifi version: 2020.12.5
chardet version: 4.0.0
click version: 8.0.0
colorama version: 0.4.4
contextlib2 version: 0.6.0.post1
future version: 0.18.2
gcovr version: 4.2
idf-component-manager version: 0.2.1a1
idna version: 2.10
Jinja2 version: 3.0.0
lxml version: 4.6.3
MarkupSafe version: 2.0.0
pip version: 21.1.1
psutil version: 5.8.0
pygdbmi version: 0.9.0.2
pywin32 version: 300
PyYAML version: 5.4.1
requests version: 2.25.1
requests-toolbelt version: 0.9.1
schema version: 0.7.4
semantic-version version: 2.8.5
setuptools version: 49.2.1
six version: 1.16.0
tqdm version: 4.60.0
urllib3 version: 1.26.4
websocket-client version: 1.0.0
wheel version: 0.36.2
xmlrunner version: 1.7.7
---------------------------------------------------- Check ESP-IDF python requirements.txt -------------------------------------------------
Check ESP-IDF Python packages undefined
---------------------------------------------------- Check extension requirements.txt ------------------------------------------------------
Check Extension Python packages undefined
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages undefined
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Command failed: C:\Users\HP\.espressif\python_env\idf4.2_py3.9_env\Scripts\python.exe c:\msys64\home\HP\esp\esp-idf\tools\check_python_dependencies.py -r "c:\msys64\home\HP\esp\esp-idf\requirements.txt"

--------------------------------------------------------------------------------------------------------------------------------------------

and this damn message when I click on everything ! F1 F1 F1 F1 F1 F1  I really looking for any advice to solve this problem!!!
[image]

brianignacio5, 2021-05-20:
Please share the log file which can be obtained from

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

It seems python packages are not configured, please run `ESP-IDF: Install ESP-IDF Python Packages` command

toosimit, 2021-05-20:
(internal/child_process.js:277:12)","level":"error","timestamp":"2021-05-19T21:22:49.314Z"}
{"user":true,"message":"Something went wrong while getting the serial port list","level":"error","timestamp":"2021-05-19T21:22:49.314Z"}
{"message":"non zero exit code 1\r\n\r\nPyserial is not installed for C:\\Users\\HP\\.espressif\\python_env\\idf4.2_py3.9_env\\Scripts\\python.exe. Please, check that IDF dependencies are installed.\r\n","stack":"Error: non zero exit code 1\r\n\r\nPyserial is not installed for C:\\Users\\HP\\.espressif\\python_env\\idf4.2_py3.9_env\\Scripts\\python.exe. Please, check that IDF dependencies are installed.\r\n\n\tat ChildProcess.<anonymous> (c:\\Users\\HP\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:1:5604)\n\tat ChildProcess.emit (events.js:315:20)\n\tat ChildProcess.EventEmitter.emit (domain.js:467:12)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:277:12)","level":"error","timestamp":"2021-05-19T21:23:11.889Z"}
{"user":true,"message":"Something went wrong while getting the serial port list","level":"error","timestamp":"2021-05-19T21:23:11.890Z"}
{"user":true,"message":"Build is required before Flashing, .bin file can't be accessed","stack":"Error: BIN_FILE_ACCESS_ERROR\n\tat c:\\Users\\HP\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:8:303123\n\tat Generator.next (<anonymous>)\n\tat c:\\Users\\HP\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:8:264773\n\tat new Promise (<anonymous>)\n\tat ve (c:\\Users\\HP\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:8:264518)\n\tat c:\\Users\\HP\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:8:302099\n\tat Function.perform (c:\\Users\\HP\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:1:4724)\n\tat We (c:\\Users\\HP\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:8:302079)\n\tat c:\\Users\\HP\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:8:265903\n\tat E._executeContributedCommand (c:\\Users\\HP\\AppData\\Local\\Programs\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\services\\extensions\\node\\extensionHostProcess.js:87:107932)\n\tat E.$executeContributedCommand (c:\\Users\\HP\\AppData\\Local\\Programs\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\services\\extensions\\node\\extensionHostProcess.js:87:108370)\n\tat l._doInvokeHandler (c:\\Users\\HP\\AppData\\Local\\Programs\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\services\\extensions\\node\\extensionHostProcess.js:91:12836)\n\tat l._invokeHandler (c:\\Users\\HP\\AppData\\Local\\Programs\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\services\\extensions\\node\\extensionHostProcess.js:91:12520)\n\tat l._receiveRequest (c:\\Users\\HP\\AppData\\Local\\Programs\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\services\\extensions\\node\\extensionHostProcess.js:91:11187)\n\tat l._receiveOneMessage (c:\\Users\\HP\\AppData\\Local\\Programs\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\services\\extensions\\node\\extensionHostProcess.js:91:9973)\n\tat c:\\Users\\HP\\AppData\\Local\\Programs\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\services\\extensions\\node\\extensionHostProcess.js:91:8074\n\tat fire (c:\\Users\\HP\\AppData\\Local\\Programs\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\services\\extensions\\node\\extensionHostProcess.js:57:1836)\n\tat S.fire (c:\\Users\\HP\\AppData\\Local\\Programs\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\services\\extensions\\node\\extensionHostProcess.js:65:15497)\n\tat c:\\Users\\HP\\AppData\\Local\\Programs\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\services\\extensions\\node\\extensionHostProcess.js:106:29764\n\tat fire (c:\\Users\\HP\\AppData\\Local\\Programs\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\services\\extensions\\node\\extensionHostProcess.js:57:1836)\n\tat S.fire (c:\\Users\\HP\\AppData\\Local\\Programs\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\services\\extensions\\node\\extensionHostProcess.js:65:15497)\n\tat t._receiveMessage (c:\\Users\\HP\\AppData\\Local\\Programs\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\services\\extensions\\node\\extensionHostProcess.js:65:20755)\n\tat c:\\Users\\HP\\AppData\\Local\\Programs\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\services\\extensions\\node\\extensionHostProcess.js:65:17641\n\tat fire (c:\\Users\\HP\\AppData\\Local\\Programs\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\services\\extensions\\node\\extensionHostProcess.js:57:1836)\n\tat acceptChunk (c:\\Users\\HP\\AppData\\Local\\Programs\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\services\\extensions\\node\\extensionHostProcess.js:65:12862)\n\tat c:\\Users\\HP\\AppData\\Local\\Programs\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\services\\extensions\\node\\extensionHostProcess.js:65:12210\n\tat Socket.v (c:\\Users\\HP\\AppData\\Local\\Programs\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\services\\extensions\\node\\extensionHostProcess.js:106:13195)\n\tat Socket.emit (events.js:315:20)\n\tat Socket.EventEmitter.emit (domain.js:467:12)\n\tat addChunk (internal/streams/readable.js:309:12)\n\tat readableAddChunk (internal/streams/readable.js:284:9)\n\tat Socket.Readable.push (internal/stream [... cut ...]
{"message":"Error checking xtensa-esp32s2-elf version","stack":"Error: Command failed: xtensa-esp32s2-elf-gcc --version\n'xtensa-esp32s2-elf-gcc' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat ChildProcess.EventEmitter.emit (domain.js:467:12)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-05-19T21:34:22.903Z"}

toosimit, 2021-05-20:
{"message":"Command failed: xtensa-esp32-elf-gcc --version\n'xtensa-esp32-elf-gcc' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n","stack":"Error: Command failed: xtensa-esp32-elf-gcc --version\n'xtensa-esp32-elf-gcc' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat ChildProcess.EventEmitter.emit (domain.js:467:12)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-05-19T21:34:22.904Z"}
{"message":"Error checking xtensa-esp32-elf version","stack":"Error: Command failed: xtensa-esp32-elf-gcc --version\n'xtensa-esp32-elf-gcc' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat ChildProcess.EventEmitter.emit (domain.js:467:12)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-05-19T21:34:22.905Z"}
{"message":"Command failed: esp32s2ulp-elf-as --version\n'esp32s2ulp-elf-as' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n","stack":"Error: Command failed: esp32s2ulp-elf-as --version\n'esp32s2ulp-elf-as' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat ChildProcess.EventEmitter.emit (domain.js:467:12)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-05-19T21:34:22.906Z"}
{"message":"Error checking esp32s2ulp-elf version","stack":"Error: Command failed: esp32s2ulp-elf-as --version\n'esp32s2ulp-elf-as' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat ChildProcess.EventEmitter.emit (domain.js:467:12)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-05-19T21:34:22.907Z"}
{"message":"Command failed: ccache.exe --version\n'ccache.exe' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n","stack":"Error: Command failed: ccache.exe --version\n'ccache.exe' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat ChildProcess.EventEmitter.emit (domain.js:467:12)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-05-19T21:34:22.918Z"}
{"message":"Error checking ccache version","stack":"Error: Command failed: ccache.exe --version\n'ccache.exe' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat ChildProcess.EventEmitter.emit (domain.js:467:12)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-05-19T21:34:22.919Z"}
{"message":"Command failed: dfu-util --version\n'dfu-util' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n","stack":"Error: Command failed: dfu-util --version\n'dfu-util' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat ChildProcess.EventEmitter.emit (domain.js:467:12)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-05-19T21:34:22.931Z"}
{"message":"Error checking dfu-util version","stack":"Error: Command failed: dfu-util --version\n'dfu-util' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat ChildProcess.EventEmitter.emit (domain.js:467:12)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-05-19T21:34:22.932Z"}
{"message":"Command failed: ninja --version\n'ninja' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n","stack":"Error: Command failed: ninja --version\n'ninja' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat ChildProcess.EventEmitter.emit (domain.js:467:12)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js [... cut ...] rocess.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat ChildProcess.EventEmitter.emit (domain.js:467:12)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-05-19T21:36:49.821Z"}
{"message":"Cannot access filePath: cmake","level":"error","timestamp":"2021-05-19T21:37:19.120Z"}
{"message":"Cannot access filePath: ninja","level":"error","timestamp":"2021-05-19T21:37:19.122Z"}

toosimit, 2021-05-20:
{"user":true,"message":"Something went wrong while trying to build the project","stack":"Error: CMake or Ninja executables not found\n\tat d.<anonymous> (c:\\Users\\HP\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:1:110335)\n\tat Generator.next (<anonymous>)\n\tat s (c:\\Users\\HP\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:1:109368)\n\tat runMicrotasks (<anonymous>)\n\tat processTicksAndRejections (internal/process/task_queues.js:93:5)","level":"error","timestamp":"2021-05-19T21:37:19.123Z"}
{"message":"Cannot access filePath: cmake","level":"error","timestamp":"2021-05-19T21:37:38.263Z"}
{"message":"Cannot access filePath: ninja","level":"error","timestamp":"2021-05-19T21:37:38.265Z"}
{"user":true,"message":"Something went wrong while trying to build the project","stack":"Error: CMake or Ninja executables not found\n\tat d.<anonymous> (c:\\Users\\HP\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:1:110335)\n\tat Generator.next (<anonymous>)\n\tat s (c:\\Users\\HP\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:1:109368)\n\tat runMicrotasks (<anonymous>)\n\tat processTicksAndRejections (internal/process/task_queues.js:93:5)","level":"error","timestamp":"2021-05-19T21:37:38.266Z"}
{"message":"Cannot access filePath: xtensa-esp32-elf-gcc","level":"error","timestamp":"2021-05-19T21:44:57.328Z"}
{"message":"non zero exit code 1\r\n\r\nINFO: Could not find files for the given pattern(s).\r\n","stack":"Error: non zero exit code 1\r\n\r\nINFO: Could not find files for the given pattern(s).\r\n\n\tat ChildProcess.<anonymous> (c:\\Users\\HP\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:1:5604)\n\tat ChildProcess.emit (events.js:315:20)\n\tat ChildProcess.EventEmitter.emit (domain.js:467:12)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:277:12)","level":"error","timestamp":"2021-05-19T21:45:00.897Z"}
{"message":"Cannot access filePath: openocd","level":"error","timestamp":"2021-05-19T21:45:00.898Z"}
{"message":"Command failed: wmic os get osarchitecture\n'wmic' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n","stack":"Error: Command failed: wmic os get osarchitecture\n'wmic' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat ChildProcess.EventEmitter.emit (domain.js:467:12)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Socket.<anonymous> (internal/child_process.js:439:11)\n\tat Socket.emit (events.js:315:20)\n\tat Socket.EventEmitter.emit (domain.js:467:12)\n\tat Pipe.<anonymous> (net.js:673:12)","level":"error","timestamp":"2021-05-19T21:45:01.155Z"}
{"message":"Command failed: esp32s2ulp-elf-as --version\n'esp32s2ulp-elf-as' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n","stack":"Error: Command failed: esp32s2ulp-elf-as --version\n'esp32s2ulp-elf-as' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat ChildProcess.EventEmitter.emit (domain.js:467:12)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-05-19T21:45:01.443Z"}
{"message":"Error checking esp32s2ulp-elf version","stack":"Error: Command failed: esp32s2ulp-elf-as --version\n'esp32s2ulp-elf-as' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat ChildProcess.EventEmitter.emit (domain.js:467:12)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-05-19T21:45:01.444Z"}
{"message":"Command failed: esp32ulp-elf-as --version\n'esp32ulp-elf-as' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n","stack":"Error: Command failed: esp32ulp-elf-as --version\n'esp32ulp-elf-as' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat ChildProcess.EventEmitter.emit (domain.js:467:12)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-05-19T21:45:01.445Z"}
{"message":"Error checking esp32ulp-elf version","stack":"Error: Command failed: esp32ulp-elf-as --version\n'esp32ulp-elf-as' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat ChildProcess.EventEmitter.emit (domain.js:467:12)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-05-19T21:45:01.446Z"}
[... cut ...] 2)\n\tat addChunk (internal/streams/readable.js:309:12)\n\tat readableAddChunk (internal/streams/readable.js:284:9)\n\tat Socket.Readable.push (internal/streams/readable.js:223:10)\n\tat Pipe.onStreamRead (internal/stream_base_commons.js:188:23)","level":"error","timestamp":"2021-05-19T21:46:24.382Z"}
{"message":"Cannot access filePath: cmake","level":"error","timestamp":"2021-05-19T21:46:29.717Z"}
{"message":"Cannot access filePath: ninja","level":"error","timestamp":"2021-05-19T21:46:29.719Z"}

toosimit, 2021-05-20:
{"user":true,"message":"Something went wrong while trying to build the project","stack":"Error: CMake or Ninja executables not found\n\tat d.<anonymous> (c:\\Users\\HP\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:1:110335)\n\tat Generator.next (<anonymous>)\n\tat s (c:\\Users\\HP\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:1:109368)\n\tat processTicksAndRejections (internal/process/task_queues.js:93:5)","level":"error","timestamp":"2021-05-19T21:46:29.719Z"}
{"message":"Cannot access filePath: cmake","level":"error","timestamp":"2021-05-19T21:46:31.193Z"}
{"message":"Cannot access filePath: ninja","level":"error","timestamp":"2021-05-19T21:46:31.195Z"}
{"user":true,"message":"Something went wrong while trying to build the project","stack":"Error: CMake or Ninja executables not found\n\tat d.<anonymous> (c:\\Users\\HP\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:1:110335)\n\tat Generator.next (<anonymous>)\n\tat s (c:\\Users\\HP\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:1:109368)\n\tat processTicksAndRejections (internal/process/task_queues.js:93:5)","level":"error","timestamp":"2021-05-19T21:46:31.195Z"}
{"message":"Cannot access filePath: cmake","level":"error","timestamp":"2021-05-19T21:46:31.401Z"}
{"message":"Cannot access filePath: ninja","level":"error","timestamp":"2021-05-19T21:46:31.403Z"}
{"user":true,"message":"Something went wrong while trying to build the project","stack":"Error: CMake or Ninja executables not found\n\tat d.<anonymous> (c:\\Users\\HP\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:1:110335)\n\tat Generator.next (<anonymous>)\n\tat s (c:\\Users\\HP\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:1:109368)\n\tat processTicksAndRejections (internal/process/task_queues.js:93:5)","level":"error","timestamp":"2021-05-19T21:46:31.403Z"}
{"message":"Cannot access filePath: cmake","level":"error","timestamp":"2021-05-19T21:46:31.571Z"}
{"message":"Cannot access filePath: ninja","level":"error","timestamp":"2021-05-19T21:46:31.572Z"}
{"user":true,"message":"Something went wrong while trying to build the project","stack":"Error: CMake or Ninja executables not found\n\tat d.<anonymous> (c:\\Users\\HP\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:1:110335)\n\tat Generator.next (<anonymous>)\n\tat s (c:\\Users\\HP\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:1:109368)\n\tat processTicksAndRejections (internal/process/task_queues.js:93:5)","level":"error","timestamp":"2021-05-19T21:46:31.573Z"}
{"message":"Cannot access filePath: cmake","level":"error","timestamp":"2021-05-19T21:46:31.752Z"}
{"message":"Cannot access filePath: ninja","level":"error","timestamp":"2021-05-19T21:46:31.754Z"}
{"user":true,"message":"Something went wrong while trying to build the project","stack":"Error: CMake or Ninja executables not found\n\tat d.<anonymous> (c:\\Users\\HP\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:1:110335)\n\tat Generator.next (<anonymous>)\n\tat s (c:\\Users\\HP\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:1:109368)\n\tat processTicksAndRejections (internal/process/task_queues.js:93:5)","level":"error","timestamp":"2021-05-19T21:46:31.754Z"}
{"message":"Cannot access filePath: cmake","level":"error","timestamp":"2021-05-19T21:46:31.930Z"}
{"message":"Cannot access filePath: ninja","level":"error","timestamp":"2021-05-19T21:46:31.932Z"}
{"user":true,"message":"Something went wrong while trying to build the project","stack":"Error: CMake or Ninja executables not found\n\tat d.<anonymous> (c:\\Users\\HP\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:1:110335)\n\tat Generator.next (<anonymous>)\n\tat s (c:\\Users\\HP\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:1:109368)\n\tat processTicksAndRejections (internal/process/task_queues.js:93:5)","level":"error","timestamp":"2021-05-19T21:46:31.933Z"}
{"message":"Cannot access filePath: cmake","level":"error","timestamp":"2021-05-19T21:46:32.106Z"}
{"message":"Cannot access filePath: ninja","level":"error","timestamp":"2021-05-19T21:46:32.108Z"}
{"user":true,"message":"Something went wrong while trying to build the project","stack":"Error: CMake or Ninja executables not found\n\tat d.<anonymous> (c:\\Users\\HP\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:1:110335)\n\tat Generator.next (<anonymous>)\n\tat s (c:\\Users\\HP\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:1:109368)\n\tat processTicksAndRejections (internal/process/task_queues.js:93:5)","level":"error","timestamp":"2021-05-19T21:46:32.108Z"}
{"message":"Cannot access filePath: cmake","level":"error","timestamp":"2021-05-19T21:46:32.288Z"}
{"message":"Cannot access filePath: ninja","level":"error","timestamp":"2021-05-19T21:46:32.290Z"}
{"user":true,"message":"Something went wrong while trying to build the project","stack":"Error: CMake or Ninja executables not found\n\tat d.<anonymous> (c:\\Users\\HP\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:1:110335)\n\tat Generator.next (<anonymous>)\n\tat s (c:\\Users\\HP\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:1:109368)\n\tat processTicksAndRejections (i [... cut ...] spsecure.py\", line 28, in <module>\r\n    from cryptography import exceptions\r\nModuleNotFoundError: No module named 'cryptography'\r\n\n\tat ChildProcess.<anonymous> (c:\\Users\\HP\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:1:5604)\n\tat ChildProcess.emit (events.js:315:20)\n\tat ChildProcess.EventEmitter.emit (domain.js:467:12)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:277:12)","level":"error","timestamp":"2021-05-19T22:16:00.371Z"}

brianignacio5, 2021-05-21:
A close look to your logs shows:

`Pyserial is not installed for C:\Users\HP\.espressif\python_env\idf4.2_py3.9_env\Scripts\python.exe`

There is also another error `wmic os get osarchitecture\n'wmic' is not recognized` which might be related to **wbem** not being available, make sure to have `c:\Windows\System32\wbem` in your system environment PATH (which seems to be the norm in most Windows systems) Related [GitHub Issue](https://github.com/apache/cordova-cli/issues/434)

brianignacio5, 2021-05-31:
Closing the issue from lack of response from the user. Please comment here if you still have any issues.

closed 2021-05-31 by brianignacio5