espressif/vscode-esp-idf-extension #414: [Bug Report]: Install procedure fails to show setup wizard

cjheath, 2021-05-21:
**Describe the bug**
Error dialog when trying to activate the `Configure ESP-IDF extension` setup wizard

**To Reproduce**

Using the installation procedure documented here
https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/install.md

Step 6 asks to use View->Command Palette->ESP-IDF: Configure ESP-IDF extension option.
VSCode shows a new tab ESP-IDF Setup with an error dialog in the bottom right corner saying:
```
this.panel.webview.asWebviewUri is not a function
Source: Espressif IDF (Extension)
Report... Cancel

**Expected behavior**

The setup wizard should be displayed.

**Screenshots**

![Screen Shot 2021-05-21 at 5 47 13 pm](https://user-images.githubusercontent.com/1703/119102705-a7826e80-ba5d-11eb-9e7b-44f50a7582be.png)

**Environment (please complete the following information):**

- MacOS X Version 10.14.6
- OS Name & Version: [e.g. MacOS X 10.14.5]
- VSCode Version: 1.34.0
- ESP-IDF Version: 1.0.3
- Python Version: 3.7.7

> **NOTE**: You can use the `ESP-IDF: Doctor command` to generate a report of your configuration.

**Logs**
If applicable, please share the log file which can be obtained from

- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`
[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/6521042/esp_idf_vsc_ext.log)

**Extension Configuration settings**
Open Command Palette (F1) and type `Preferences: Open Settings (JSON)` or use `{CURRENT_PROJECT}/.vscode/settings.json` to get the following extension configuration settings value:

The template in this section does not say how to "get" those settings.

brianignacio5, 2021-05-21:
There seems to be an error with your installed extension. Please delete the folder in `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION` and re-install the extension (here is the link to latest master [installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/62168384.zip))

> The template in this section does not say how to "get" those settings.
> Open Command Palette (F1) and type `Preferences: Open Settings (JSON)` or use `{CURRENT_PROJECT}/.vscode/settings.json` to get the following extension configuration settings value:

cjheath, 2021-05-21:
It was a fresh download in the last two hours. But I found something. VSCode calls itself "VSCodium" and was installed through HomeBrew. I have a ~/.vscode-oss directory not ~/.vscode.

I expect there is a compatibility issue arising from that? I'm just checking now if there is an updated HomeBrew version of the visual-studio-code package.

cjheath, 2021-05-21:
Oh wait, that's wierd. brew says I don't have the visual-studio-code cask installed so it can't upgrade it. It seems I have "vscodium" installed instead. I just upgraded that to 1.56.2, and will see how the IDF extension goes now

cjheath, 2021-05-21:
Ok the wizard appears now. The `brew upgrade vscodium` replaced my `/usr/local/bin/code` by a `/usr/local/bin/codium`, which should avoid future confusion.

closed 2021-05-21 by cjheath

cjheath, 2021-05-22:
A further update. After updating vscodium, I find that it no longer works with PlatformIO. Uninstall and try to reinstall PlatformIO, but it doesn't appear in the available extensions list.

If anyone finds they have the same problem, the solution is to uninstall vscodium and get the proper vscode, perhaps using `brew install visual-studio-code`, then re-install any extensions you were using.