espressif/vscode-esp-idf-extension #418: WSL: Flash command does not evaluate additional flash-images

volk-christian, 2021-05-28:
While running the esp-idf-extension within WSL2 remotely, additional flash images are not added to the flash command.

Build-Output:
```
Project build complete. To flash, run this command:
/home/user/esp-idf-tools/python_env/idf4.1_py3.8_env/bin/python ../../esp-idf/components/esptool_py/esptool/esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x9000 build/ota_data_initial.bin 0x14000 build/menu.mnu 0xa0000 build/application.bin
```

Flash command within VSCode:
```
> Executing task: powershell.exe -Command "python \\\\wsl$\\Ubuntu-20.04\\home\\user\\esp-idf\\components\\esptool_py\\esptool\\esptool.py -p COM4 -b 460800 --before default_reset --after hard_reset --chip esp32 write_flash --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 bootloader\bootloader.bin 0xa0000 application.bin 0x8000 partition_table\partition-table.bin" <
```

As you can see by this simple comparison, the custom image "menu.mnu" at address 0x14000 is not part of the actual executed flash command.

brianignacio5, 2021-05-31:
Can you try with this vsix installer https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/62168384.zip

There are some fixes in GitHub master which are not yet part of the marketplace release.

volk-christian, 2021-06-01:
Hi @brianignacio5,

many thanks for your quick reply. The nightly preview solved the issue.
Looking forward to have it available in the marketplace.

Many thanks and best regards!

closed 2021-06-01 by volk-christian