espressif/vscode-esp-idf-extension #420: [Bug Report]: Extension doesn't use system's Python

tehKaiN, 2021-05-28:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [x] Make sure you have searched for existing bugs and features request before you post an issue.
- [x] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [x] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**

When in Select Python Version step, the wizard doesn't detect my python install. When I pass the path maually, it results with errors due to old pip version.

**To Reproduce**

- I type in cmd:

```
> where python
C:\prg\kompilatory\Python36-32\python.exe - the proper install
C:\Users\tehKa\AppData\Local\Microsoft\WindowsApps\python.exe - Microsoft's placeholer which redirects to python store page
>pip --version
pip 21.1.2 from c:\prg\kompilatory\python36-32\lib\site-packages\pip (python 3.6)
```

- I specify the path manually with the first one. I also set download tools path to `c:\prg\libs\esp`.
- Then the failure in output window appears when installing esp tools:

```
Creating a new Python environment in c:\prg\libs\esp\python_env\idf4.2_py3.6_env ...

Collecting wheel
  Using cached https://files.pythonhosted.org/packages/65/63/39d04c74222770ed1589c0eaba06c05891801219272420b40311cd60c880/wheel-0.36.2-py2.py3-none-any.whl
Installing collected packages: wheel
Successfully installed wheel-0.36.2
You are using pip version 9.0.1, however version 21.1.2 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.

Usage:   
  c:\prg\libs\esp\python_env\idf4.2_py3.6_env\Scripts\python.exe -m pip install [options] <requirement specifier> [package-index-options] ...
  c:\prg\libs\esp\python_env\idf4.2_py3.6_env\Scripts\python.exe -m pip install [options] -r <requirements file> [package-index-options] ...
  c:\prg\libs\esp\python_env\idf4.2_py3.6_env\Scripts\python.exe -m pip install [options] [-e] <vcs project url> ...
  c:\prg\libs\esp\python_env\idf4.2_py3.6_env\Scripts\python.exe -m pip install [options] [-e] <local project path> ...
  c:\prg\libs\esp\python_env\idf4.2_py3.6_env\Scripts\python.exe -m pip install [options] <archive url/path> ...

no such option: --no-warn-script-location
Command failed: "c:\prg\libs\esp\python_env\idf4.2_py3.6_env\Scripts\python.exe" -m pip install --no-warn-script-location -r "c:\prg\libs\esp\esp-idf\requirements.txt"

Usage:   
  c:\prg\libs\esp\python_env\idf4.2_py3.6_env\Scripts\python.exe -m pip install [options] <requirement specifier> [package-index-options] ...
  c:\prg\libs\esp\python_env\idf4.2_py3.6_env\Scripts\python.exe -m pip install [options] -r <requirements file> [package-index-options] ...
  c:\prg\libs\esp\python_env\idf4.2_py3.6_env\Scripts\python.exe -m pip install [options] [-e] <vcs project url> ...
  c:\prg\libs\esp\python_env\idf4.2_py3.6_env\Scripts\python.exe -m pip install [options] [-e] <local project path> ...
  c:\prg\libs\esp\python_env\idf4.2_py3.6_env\Scripts\python.exe -m pip install [options] <archive url/path> ...

no such option: --no-warn-script-location
Command failed: "c:\prg\libs\esp\python_env\idf4.2_py3.6_env\Scripts\python.exe" -m pip install --no-warn-script-location -r "c:\prg\libs\esp\esp-idf\requirements.txt"

Usage:   
  c:\prg\libs\esp\python_env\idf4.2_py3.6_env\Scripts\python.exe -m pip install [options] <requirement specifier> [package-index-options] ...
  c:\prg\libs\esp\python_env\idf4.2_py3.6_env\Scripts\python.exe -m pip install [options] -r <requirements file> [package-index-options] ...
  c:\prg\libs\esp\python_env\idf4.2_py3.6_env\Scripts\python.exe -m pip install [options] [-e] <vcs project url> ...
  c:\prg\libs\esp\python_env\idf4.2_py3.6_env\Scripts\python.exe -m pip install [options] [-e] <local project path> ...
  c:\prg\libs\esp\python_env\idf4.2_py3.6_env\Scripts\python.exe -m pip install [options] <archive url/path> ...

no such option: --no-warn-script-location
```

**Expected behavior**
The extension should use builtin python instead of this one, or at least upgrade its pip by itself.

**Environment (please complete the following information):**
[... 95 lines cut ...]
{"message":"Command failed: ccache.exe --version\n'ccache.exe' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n","stack":"Error: Command failed: ccache.exe --version\n'ccache.exe' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-05-28T15:41:16.289Z"}
{"message":"Error checking ccache version","stack":"Error: Command failed: ccache.exe --version\n'ccache.exe' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-05-28T15:41:16.289Z"}
{"message":"Command failed: idf.py.exe -v\n'idf.py.exe' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n","stack":"Error: Command failed: idf.py.exe -v\n'idf.py.exe' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-05-28T15:41:16.303Z"}
{"message":"Error checking idf-exe version","stack":"Error: Command failed: idf.py.exe -v\n'idf.py.exe' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-05-28T15:41:16.304Z"}
{"message":"Open On-Chip Debugger 0.10.0 (2017-08-21) [https://github.com/sysprogs/openocd]\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger 0.10.0 (2017-08-21) [https://github.com/sysprogs/openocd]\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n\tat c:\\Users\\tehKa\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:1:8461\n\tat ChildProcess.exithandler (child_process.js:308:7)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-05-28T15:41:16.345Z"}
{"message":"You are using pip version 9.0.1, however version 21.1.2 is available.\r\nYou should consider upgrading via the 'python -m pip install --upgrade pip' command.\r\n","stack":"Error: You are using pip version 9.0.1, however version 21.1.2 is available.\r\nYou should consider upgrading via the 'python -m pip install --upgrade pip' command.\r\n\n\tat c:\\Users\\tehKa\\.vscode\\extensions\\espressif.esp-idf-extension-1.0.3\\dist\\extension.js:1:8461\n\tat ChildProcess.exithandler (child_process.js:308:7)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-05-28T15:49:42.648Z"}
{"message":"Command failed: \"c:\\prg\\libs\\esp\\python_env\\idf4.2_py3.6_env\\Scripts\\python.exe\" -m pip install --no-warn-script-location -r \"c:\\prg\\libs\\esp\\esp-idf\\requirements.txt\"\n\r\nUsage:   \r\n  c:\\prg\\libs\\esp\\python_env\\idf4.2_py3.6_env\\Scripts\\python.exe -m pip install [options] <requirement specifier> [package-index-options] ...\r\n  c:\\prg\\libs\\esp\\python_env\\idf4.2_py3.6_env\\Scripts\\python.exe -m pip install [options] -r <requirements file> [package-index-options] ...\r\n  c:\\prg\\libs\\esp\\python_env\\idf4.2_py3.6_env\\Scripts\\python.exe -m pip install [options] [- [... cut ...]
{"message":"spawn C:\\WINDOWS\\system32\\cmd.exe ENOENT","stack":"Error: spawn C:\\WINDOWS\\system32\\cmd.exe ENOENT\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:269:19)\n\tat onErrorNT (internal/child_process.js:465:16)\n\tat processTicksAndRejections (internal/process/task_queues.js:80:21)","level":"error","timestamp":"2021-05-28T16:00:08.329Z"}
{"user":true,"level":"warn","message":"Extension configuration report has been copied to clipboard with errors","timestamp":"2021-05-28T16:00:08.329Z"}
```

**Extension Configuration settings**

Default ones.

brianignacio5, 2021-05-31:
When you do `C:\prg\kompilatory\Python36-32\python.exe -m pip --version` what is the version of pip ? 

latest pip doesn't seem to be part of the same python binary you are using to create the virtual environment.

tehKaiN, 2021-05-31:
I've already had my pip updated to the latest version on that python installation prior to doing work with extension (pip 21.1.2 from c:\prg\kompilatory\python36-32\lib\site-packages\pip (python 3.6)).

I've worked around the problem by executing pip update python installed with sdk (in dir `esp-sdk\python_env\idf4.2_py3.6_env\Scripts`) and rerunning package download commands manually with both of them.

See in my log that pip commands failed with python installed with sdk:

`Command failed: "c:\prg\libs\esp\python_env\idf4.2_py3.6_env\Scripts\python.exe" -m pip install --no-warn-script-location -r "c:\prg\libs\esp\esp-idf\requirements.txt"`

brianignacio5, 2021-05-31:
Well the Python is not detected because it is not part of the **Windows registry** ( I assume just because of python dir location), which is what we use to find python versions in Windows, so that is probably why you have to manually input the python path.

The reason why the python virtual env doesn't seem to be updated is because is already created with the old pip version and the extension setup is trying to use the existing env. You can delete `c:\prg\libs\esp\python_env\idf4.2_py3.6_env` folder and run the setup again or the `Install ESP-IDF Python Packages` command

tehKaiN, 2021-05-31:
The windows registry approach is incomplete, since it is a quite common practice to get development tools (be it gcc, python or other tools) not by installers but zipped versions - this is much cleaner for having different versions in your OS, so I think addition of scanning PATH for existing python.exe seems to be more complete solution.

Still, I'd be fine without that autodetect. The true problem of this issue lies in extension downloading another python even though I've manually specified the python from my OS in the initial step. The problem of wrong pip being used would be nonexisting for me then, because it would only use the already up to date one.

I've solved this on my end manually with similar steps you've suggested, but I thought I'll report this unexpected behavior of the extension so that it can be fixed.

brianignacio5, 2021-06-01:
Yes I understand we actually just do `which python` on other OS but Windows approach is slightly more complicated. With regards to us download Python this is not true yet, what we do is create a python virtual environment using the specified python executable and install python packages in the virtual environment

For our Windows users, we are working on using an embed Python with proper pip version that is part of the setup of ESP-IDF to avoid many of the problems that comes with Python compatibility.

samsam4, 2021-06-08:
I have a very similar problem when tried to configure ESP-IDF extension with the Python version that was installed from ESP-IDF tools installer 2.9beta from [here](https://github.com/espressif/idf-installer/releases/tag/online-2.9-beta-04)
```
creating a new Python environment in d:\ESP4_2_1\ESP-IDF_Tools\python_env\idf4.2_py3.8_env ...
d:\ESP2_9b_tools\.espressif\tools\idf-python\3.8.7\python.exe: No module named venv.__main__; 'venv' is a package and cannot be directly executed
Command failed: "d:\ESP2_9b_tools\.espressif\tools\idf-python\3.8.7\python.exe" -m venv "d:\ESP4_2_1\ESP-IDF_Tools\python_env\idf4.2_py3.8_env"
d:\ESP2_9b_tools\.espressif\tools\idf-python\3.8.7\python.exe: No module named venv.__main__; 'venv' is a package and cannot be directly executed
Command failed: "d:\ESP2_9b_tools\.espressif\tools\idf-python\3.8.7\python.exe" -m venv "d:\ESP4_2_1\ESP-IDF_Tools\python_env\idf4.2_py3.8_env"
d:\ESP2_9b_tools\.espressif\tools\idf-python\3.8.7\python.exe: No module named venv.__main__; 'venv' is a package and cannot be directly executed.
```

Could you hint me, please, how to resolve it?
More details and the log was already posted [here](https://github.com/espressif/idf-installer/issues/25)

Thanks

samsam4, 2021-06-09:
Update:  Solved the problem with "No module named venv.__main__; 'venv' is a package and cannot be directly executed."
Those ESP-IDF installers make 2 different  python.exe files with different size and different location.
I initially used in Config extension the one from tools\idf-python\3.8.7\  and got errors in installation mentioned above. 
Today I tried with the other one that is in tools\.espressif\python_env\idf4.2_py3.8_env\Scripts and installer didn't report those errors. 
Don't see the logic though as  tools\idf-python\3.8.7\python.exe should be same as the original python if downloaded from python site, while the one in \Scripts is some ESP version that calls the first one?

brianignacio5, 2021-06-10:
The thing is that the IDF Installer is using an embed Python and the extension has not yet implemented this flow, we were using venv but ESP-IDF and the installer are using virtualenv, which is installed if not found.

Please take a look at [Using idf embed python and idf embed git](https://github.com/espressif/vscode-esp-idf-extension/pull/416)

brianignacio5, 2021-06-10:
I'm closing this issue since the setup can in fact use any python path you provide. 

In the future we will use an embed python to avoid many user issues such as this one.

closed 2021-06-10 by brianignacio5